BelfrySCAD/BOSL2 #1: hull functions

adrianVmariano, 2019-04-25:
I took a look at the hull functions and things seem to be working as expected generally.  I have the following comments:

* In openscad base language, we have hull(), not hull2d and hull3d.  What is gained by splitting this into two functions instead of supplying simply hull_points() that does both cases in perfect parallel with the base language?
* I think it would be nicer to wrap the "fast" version in as an option, so hull_points(points, fast=true) would activate it instead of having a separate function for it.  
* The colinearity check increases run time for me from 6s to 36s on 5e5 points input.  One could argue that nobody will ever need to run on such a large set of points, I guess, but it is an unnecessary check.  
* If the number of points isn't divisible by 3 the last one is skipped in hull3d_points_fast.  You need to change -3 to -2 in the loop.   I wonder if running the loop to step off by 2's would decrease the run-time penalty for checking colinearity.  (The end effect would be complicated.)
* Have you observed a test case where it the "fast" code prints a warning?  
* Document that the hull functions can handle about 4k points whereas hull_fast can handle unlimited points (up to max array size, I think)
* The documentation doesn't mention the outcome in the colinear case, which for hull*_points is an error()
* Actually it looks like the colinear case is broken.  In 2d the original linde call works and produces the endpoints and the BOSL function produces incorrect output.  In 3d it appears that the original version and the linde version both produce the same incorrect output (a set of four 3-vertex faces).

revarbat, 2019-05-01:
Committed in master branch BOSL2 commit 3a71633f7f429bfe2ae28e48d76f972ba1bd41ae
- Renamed generic 2D/3D function to `hull()`
- Made 2D/3D `hull_points()`.
- Wrapped fast into `hull_points()` as an arg.
- Documented fast vs. slow.
- Reverted collinearity check on fast polyhedron hull.
- Fixed leftover hull point in fast hull.
- Fixed collinear cases that got broken by angular collinearity test and collinear initial plane point selection.

revarbat, 2019-05-01:
If this looks good, I'll backport it to BOSL v1.

adrianVmariano, 2019-05-06:
I've checked the issues I noted earlier.  Looks like the colinear case is working in hull(), hull2d_path() and hull3d_faces().  You might still consider adding a note about this case in the docs, that you'll just get two points out if the input is colinear.  

Looks like the problem of ignoring some points is fixed as well.  

Do you know of a case where hull_points(...,fast=true) generates a warning?  I feel like I used to see warnings and they went away.  I haven't found any input that generates warnings.  

I did notice that hull_points with fast=true has bad behavior in 2d.   With 5500 points the linde version (slow case) finishes in a couple seconds.  The "fast" case runs for a long time (many minutes).  I've haven't waited for completion.  I'm not sure what's going on there.   I'm not sure polyhedron was meant to be called in 2d, so maybe it's doing something wacky?

adrianVmariano, 2019-05-06:
I tried using polygon() instead of polyhedron() in 2d but it's still slow.  I guess the "fast" option should be either ignored or considered an error when the input is 2d.   It's so slow I don't imagine anybody would want to do more than 5500 points and wait for it....  I suppose one could break a large point set into small sets, hull each small set and then hull the union.

revarbat, 2019-05-06:
`hull_points()` with `fast=true` for 2D points *does* use `hull() polygon(points=points)`.  I have no idea why it would be slow.

revarbat, 2019-05-08:
I've seen the degenerate polygon warnings in the last couple weeks, but I can't recall if I've seen them with THIS code.

closed 2019-05-14 by revarbat