BelfrySCAD/BOSL2 #2: Always have diameter and radius

sej7278, 2019-04-26:
Some functions only take a diameter and not a radius or vice versa, in v2 can we take both by default please as its easier than having to look up which ones take which parameters.

revarbat, 2019-04-26:
Easily doable.  I've been doing that already for a lot of modules.  Only case I'm not sure it makes sense for is in edge roundings, where I'm not sure that anyone would want to use diameter over radius.

sej7278, 2019-04-26:
yup agree, i meant shapes and stuff not edges

adrianVmariano, 2019-06-19:
What is the status of this issue?  Is there a list of non-compliant functions?

revarbat, 2019-06-19:
I want to make a pass through all the docs and verify that the documented arguments match the actual arguments taken by the functions/modules.  While doing that, I can find any violations and fix them.

closed 2020-08-27 by revarbat