BelfrySCAD/BOSL2 #4: attachments documentation

adrianVmariano, 2019-04-27:
I'm trying to figure out how to use the new attachments stuff and confused about it.  I have constructed a shape that is an extruded octagon with a rounded cylinder subtracted.  I want to add a dovetail slot off center at one edge of the octagon.  To do this (because of how my dovetail code works) need to subtract a cube and then add a dovetail part.   How do I do this using the attachments?  That is, I need to mark the location on the model where the thing goes (using the anchor function?) and then do I also create anchors for the cube I want to subtract and the dovetail slot part I want to add in?   Then I somehow use orient_and_anchor?

revarbat, 2019-04-27:
This will take a bit of explaining. I’ve been drafted to do yard work, though, so I can’t get into it too much right now. Short version: think of `attach()` as an optional rotate and a non-optional translate to the named (or directional vector) anchor on the parent. Because that’s what it is. Named anchors can be given to orient_and_anchor(), but every primitive and most BOSL shapes support direction vectors. If attach() gives a second anchor, then the anchor and orient options of the children are overridden.

You can still use rotation and translations

revarbat, 2019-04-27:
Oh, and you can use `norot=true` in attach() to get only the translation.

ether42, 2019-05-19:
Is this simple example expected to work? I wanted to use a named anchor from a child as the `to` parameter of `attach` but it seems to crashes.
From the [documentation](https://github.com/revarbat/BOSL2/wiki/attachments.scad#attach) I would have said yes but it can't find my mistake.
I would have expected that `attach` attaches the `"right"` anchor of the parent to the `"left"` anchor of the child.

```scad
include <BOSL2/std.scad>
include <BOSL2/debug.scad>

module c() {
  dimension = 30;

  anchors = [
    anchorpt("left", [-dimension / 2, 0, 0], dir=LEFT),
    anchorpt("right", [dimension / 2, 0, 0], dir=RIGHT),
  ];

  orient_and_anchor([dimension, dimension, dimension], anchors=anchors) {
    cuboid(dimension);
    children();
  }
}

c() {
  show_anchors(std=false);
  attach("left", RIGHT, norot=true) left(10) c();
  // Unknown anchor: left
  // attach("right", "left", norot=true) right(10) c();
}
```

[image]

revarbat, 2019-05-19:
Two things I see so far:
- `c()` needs to accept a `orient` and `align` arguments that are passed to `orient_and_anchor()`
- `orient_and_anchor()` needs a `chain=true` argument passed to it.

revarbat, 2019-05-19:
I warn that I'm still fiddling with `attach()` and `anchor` and trying to figure out the best way for it to work most intuitively.  Things may still change drastically.  For one, i'm debating changing `orient` to take vector args instead of rotation angles.  For another, I may end up reverting from anchor-after-orient to anchor-before-orient.

revarbat, 2019-05-19:
There's a reason BOSL2 is labled pre-alpha.

ether42, 2019-05-20:
My bad for `chain=true`, I forgot it in this example.
I wasn't aware `orient` and `align` had to be passed to `orient_and_anchor` though.

I saw the pre-alpha warning, I just wanted to report the problem (and this issue was perfect to host it) in case you weren't aware, so no worries.

revarbat, 2019-05-30:
## Status update:
- `anchor` points at the face center, edge center, or corner to center at the origin.
- `spin` rotates around the Z axis by a given number of degrees.
- `orient` is now a vector pointing at where the top should be rotated to.
- The order of application is `anchor`, then `spin`, then `orient`.
- `spin` can accept either a scalar number of degrees, or an [X,Y,Z] rotation angle vector.
- Docs for these have been rewritten, and are hopefully clearer.

## To Do:
- [ ] Tags handling for `diff()`, `intersect()` and `hulling()` probably need testing and rejiggering.
- [ ] The `recolor()` function should maybe take tags.
- [ ] Need to create tutorials to show how to use attachments.

closed 2021-05-22 by revarbat