BelfrySCAD/BOSL2 #10: 2D shapes libfile

revarbat, 2019-05-09:
Need to make a `shapes2d.scad` libfile.  Some shapes for it:
- `pie_slice2d()`
- `trapezoid()`
- `regular_ngon()`
- `stroke()`
- `circle_part()` (like `half_of() circle()` with width/length options)
- `pentagon()`
- `hexagon()`
- `octagon()`
- `star()`
- `superformula()` https://en.wikipedia.org/wiki/Superformula

revarbat, 2019-05-09:
Maybe a little `turtle()` turtle-graphics line generator.

adrianVmariano, 2019-05-09:
circle_part should be circle_segment because that is the technically correct name for the shape, whereas "part" is vague.  

Also there's the question of whether to add glued circle
https://openhome.cc/eGossip/OpenSCAD/lib-shape_glued2circles.html

adrianVmariano, 2019-05-10:
Here's star().  I had a bit of trouble explaining it.  The "step" option gives the standard star where the points are "lined up."  Setting ir gives an arbitrary star.   Do you need a new heading for dual function/modules to avoid having to create double documentation for them?  

// Function/Module: star()
//
// Description:      
//   Produce the coordinates of an n-pointed star with specified radius or diameter.  If you specify step
//   it creates a star that could be drawn by connecting n points on a circle with lines where each line
//   moves around the circle the specified number of steps.  Setting steps=2 creates a star with very blunt points;
//   The steps parameter must be smaller than n/2 and its maximum value produces the pointiest star.
//   Other stars can be created by specifying the inner radius or diameter. 
function star(n, r, d, ir, id, step, align="point") =
   let(
     r = get_radius(r=r, d=d),
     count = len(remove_undefs([ir,id,step])),
     stepOK = is_undef(step) || (step>1 && step<n/2)
   )
   assert(count==1, "Must specify exactly one if ir, id, step")
   assert(stepOK, str("Parameter 'step' must be between 2 and ",floor(n/2)," for ",n," point star"))
   let(
     ir = get_radius(r=ir, d=id, dflt=cos(180*step/n)/cos(180*(step-1)/n)),
     echo(ir=ir),
     offset = align=="point" ? 0 : 180/n
   )
   [for(i=[0:2*n-1]) let(theta=180*i/n+offset, radius=(i%2)?ir:r) radius*[cos(theta), sin(theta)]];

revarbat, 2019-05-11:
I coded up the following last night when I couldn't sleep.  Still need to write the documentation, though:
- pie_slice2d()
- trapezoid()
- regular_ngon()
- pentagon()
- hexagon()
- septagon()
- octagon()
- glued_circles()

adrianVmariano, 2019-05-11:
I think heptagon is the more common customary name for the 7 sided polygon.   I'm not sure it really needs to be included as a special case (one can use regular_non) though I guess there's no harm.

revarbat, 2019-05-11:
Yeah, septagon/heptagon seems unnecessary.  I'm also trying hard to think of a use for the superformula shape, though I wrote the code for it last night.

adrianVmariano, 2019-05-12:
Seems like someone could make some nifty mathematical models using superformula, but immediately practical applications don't leap to mind, I'll agree.  One might ask about the use for the great stellated dodecahedron too.  

Seems like heptagon may be unnecessary but you should probably have triangle() to make the equilateral triangle.

revarbat, 2019-05-12:
I am completely not understanding the `step` argument of `star()`.

revarbat, 2019-05-12:
Oh.  Nevermind. Was a bug. You just need to multiply that step equation by `r`.

revarbat, 2019-05-12:
Checked in the majority of these shapes in commit cdaeb136d08ec2d9c8d5790abcac3e184be00d0e

revarbat, 2019-05-29:
Implemented `arc()` in 0836e821f465217a268d234ae3e360f032c5bce4 which provides `circle_segment()` functionality.

closed 2019-05-29 by revarbat