BelfrySCAD/BOSL2 #11: add roundcorners

adrianVmariano, 2019-05-09:
I'm not sure that "curve" and "type" are the best designed interface.  

```
use<BOSL2/std.scad>
use<BOSL2/beziers.scad>

// Function: roundcorners()
//
// Description:
//   Takes a 2d or 3d point list as input (a path or the points of a polygon) and rounds each corner by a specified amount.
//   The rounding at each point can be different and some points can have zero rounding.  The roundcorners() function supports
//   two types of rounding, circular rounding and continuous curvature rounding using 4th order bezier curves.
//   Circular rounding can produce a tactile "bump" where the curvature changes from flat to circular.
//   See https://hackernoon.com/apples-icons-have-that-shape-for-a-very-good-reason-720d4e7c8a14
//
//   You select the type of rounding using the `curve` option, which should be either "smooth" to get continuous curvature 
//   rounding or "circle" to get circular rounding.  Each rounding method has two options for how you specify the amount of rounding, which
//   you select using the `type` argument.  Both rounding methods accept `type="cut"`.  This mode specifies the amount of rounding
//   as the distance from the corner to the curve.  This can be easier to understand than setting a circular radius, which can
//   be unexpectedly extreme when the corner is very sharp.  It also allows a systematic specification of curves that is the same
//   for both "circle" and "smooth".
//
//   The second `type` setting for circular rounding is "radius", which sets a circular rounding radius.  The second `type`
//   setting for smooth rounding is "joint" which specifies the distance away from the corner where the roundover should start.
//   The "smooth" type rounding also has a parameter that specifies how smooth the curvature match is.  This parameter ranges
//   from 0 to 1, with a default of 0.5.  Larger values give a more abrupt transition and smaller ones a more gradual transition.
//   If you set the value much higher than 0.8 the curvature changes abruptly enough that though it is theoretically continuous, it may not
//   be continous in practice.  If you set it very small then the transition is so gradual that the length of the roundover may be extremely long. 
//
//   If you select curves that are too large to fit the function will fail with an error.  It displays a set of scale factors that you can apply to 
//   the (first) smoothing parameter that will reduce the size of the curves so that they will fit on your path.  If the scale factors are larger than
//   one then they indicate how much you can increase the curve sizes before collisions will occur.  
//
//   To specify rounding parameters you can use the `all` option to round every point in a path.
//   Examples:
//   * `curve="circle", type="radius", all=2`: Rounds every point with circular, radius 2 roundover
//   * `curve="smooth", type="cut", all=2`: Rounds every point with continuous curvature rounding with a cut of 2, and a default 0.5 smoothing parameter
//   * `curve="smooth", type="cut", all=[2,.3]`: Rounds every point with continuous curvature rounding with a cut of 2, and a very gentle 0.3 smooth setting
//
//   The path is a list of 2d or 3d points, possibly with an extra coordinate giving smoothing parameters.  It is important to specify 
//   if the path is a closed path or not using the `closed` parameter.  The default is a closed path for making polygons.  
//   Path examples:
//   * [[0,0],[0,1],[1,1],[0,1]]: 2d point list (a square), `all` was given to set rounding
//   * [[0,0,0], [0,1,1], [1,1,2], [0,1,3]]: 3d point list, `all` was given to set rounding
//   * [[0,0,0.2],[0,1,0.1],[1,1,0],[0,1,0.3]]: 2d point list with smoothing parameters different at every corner, `all` not given
//   * [[0,0,0,.2], [0,1,1,.1], [1,1,2,0], [0,1,3,.3]]: 3d point list with smoothing parameters, `all` not given
//   * [[0,0,[.3,.7], [4,0,[.2,.6]], [4,4,0], [0,4,1]]: 3d point list with smoothing parameters for the "smooth" type roundover, `all` not given.  Note the third entry is sometimes a pair giving both smoothing parameters, sometimes it's zero specifying no smoothing, and sometimes a single number, specifying the amount of smoothing but using the default smoothness parameter.  
//
//   The number of segments used for roundovers is determined by $fa, $fs and $fn as usual for circular roundovers.  For continuous curvature roundovers $fs and $fn are used and $fa is ignored.  When doing continuous curvature rounding be sure to use lots of segments or the effect will be hidden by the discretization.  
//
// Arguments:
//   path = list of points defining the path to be rounded.  Can be 2d or 3d, and may have an extra coordinate giving rounding parameters.  If you specify rounding parameters you must do so on every point.  
//   curve = rounding method to use.  Set to "circle" for circular rounding and "smooth" for continuous curvature 4th order bezier rounding
//   type = rounding parameter type.  Set to "cut" to specify the cut back with either "smooth" or "circle" rounding methods.  Set to "radius" with `curve="circle"` to set circular radius rounding.  Set to "joint" with `curve="smooth"` for joint type rounding.  (See above for details on these rounding options.)
//   all = curvature parameter(s).  Set this to the curvature parameter or parameters to apply to all points on the list.  If you set this then all values given in the path are treated as geometrical coordinates.  If you don't set this then the last value of each entry in `path` is treated as a smoothing parameter.
//   closed = if true treat the path as a closed polygon, otherwise treat it as open.  Default: true.
//
// Example: Standard circular roundover with radius the same at every point. Compare results at the different corners.  
//   shape = [[0,0], [10,0], [15,12], [6,6], [6, 12], [-3,7]];
//   polygon(roundcorners(shape, curve="circle", type="radius", all=1));
//   %down(.1)polygon(shape);
// Example: Circular roundover using the "cut" specification, the same at every corner.  
//   shape = [[0,0], [10,0], [15,12], [6,6], [6, 12], [-3,7]];
//   polygon(roundcorners(shape, curve="circle", type="cut", all=1));
//   %down(.1)polygon(shape);
// Example: Circular roundover using the "cut" specification, the same at every corner.  
//   shape = [[0,0], [10,0], [15,12], [6,6], [6, 12], [-3,7]];
//   polygon(roundcorners(shape, curve="circle", type="cut", all=1));
//   %down(.1)polygon(shape);
// Example: Continous curvature roundover using "cut", still the same at every corner.  The default smoothness parameter of 0.5 was too gradual for these roundovers to fit, but 0.7 works.  
[... 110 lines cut ...]
// has the same length then the arc will end at p2.

function circular_arc(center, p1, p2, N) = let(
   angle = pathangle([p1,center,p2]),
   v1 = p1-center,
   v2 = p2-center,
   N = ceil(angle/360) * segs(norm(v1))
   )
   len(center)==2 ?
     let(dir = sign(v1.x*v2.y-v1.y*v2.x),   // z component of cross product
         r=norm(v1))
	 assert(dir != 0, "Colinear inputs don't define a unique arc")
	 [for(i=[0:N-1])
	    let(theta=atan2(v1.y,v1.x)+i*dir*angle/(N-1))
	      r*[cos(theta),sin(theta)]+center] :
     let(axis = cross(v1,v2))
         assert( axis != [0,0,0], "Colinear inputs don't define a unique arc")
         [for(i=[0:N-1])
	    matrix3_rot_by_axis(axis, i*angle/(N-1)) * v1 + center];

function bezier_curve(P,N) =
   [for(i=[0:N-1]) bez_point(P, i/(N-1))];

function pathangle(pts) = vector_angle(pts[0]-pts[1], pts[2]-pts[1]);
```

adrianVmariano, 2019-05-09:
Another thing I'm not sure about is how to deal with the output of the scale factors.  They are printed unconditionally, which could be awkward if the list is long.  But they are a pretty handy aid when trying to figure out how to make your roundovers fit.

adrianVmariano, 2019-05-09:
There should probably be an example of a non-closed path and also an example in 3d.  Not sure how to actually display either one of those, though.

revarbat, 2019-05-14:
Working on integrating this now.

revarbat, 2019-05-14:
Display of 2D paths can be done with `stroke()` or `trace_polyline()`.  For 3D paths, just `trace_polyline()`.

adrianVmariano, 2019-05-14:
I kinda dislike underscores...but for consistency with the general library practice I suppose you'll change the name to round_corners(), and that's probably the right thing to do.  

Another comment that could affect integration.  I wrote circular_arc for this function which draws an arc in 2d or 3d based on a center point and two end points.   It might be nice to generalize this to have a 'longest path" option (to draw the other arc).  But also there's the question of how you draw semicircles.  Maybe this function just can't do that.

I had to write another function, circular_arc_3pts, which can draw semicircles.  I don't know if this should be a separate function, or if the interface should be merged somehow.  At the moment, this function only supports 2d, though, and it wasn't obvious how to generalize it.  I actually wonder if the right answer is to find the plane containing the 3 points, transform to the 2d case, make a recursive call, and then untransform.   (The circular_arc_3pts is located in the path offset and rounded extrude stuff.)

revarbat, 2019-05-15:
Truthfully, I hadn't even thought of changing `roundcorners()` name.

You can easily use `xyz_to_planar()` and `planar_to_xyz()` for going from 3D planar to 2D and back.

revarbat, 2019-05-15:
I would honestly like to see geometry functions to find the center of a circle from 3 points, and then have an `arc()` 2D Function&Module to do the arc path (and/or drawing via `stroke()`?)

adrianVmariano, 2019-05-15:
Well, even though I'm not a fan of underscores, I think you should change the name to round_corners because you always use underscores, so running it together will confuse the users.  

Circle from 3 points works (mathematically) in 3d, so why should it be limited to 2?   The code is actually pretty simple in 2d:

center = line_intersection(normal_segment(p1,p2),normal_segment(p2,p3))

Of course, you'll notice I've cleverly hidden all the complexity elsewhere.  :)  

It does appear that xyz_to_planar will do the job of mapping to 2d for arc construction.  It seems like it is kind of inefficient to have to do all the geometry calculation over again each time for my 1000 point long arc, though.  

I agree that the arc function you have now can be produced by a stroke() call of some sort.   Also the slot module I think.  And note that every time I've needed a lot in my designs I've needed one end to open out (fillet), which your function doesn't do but my stroke does.  I'm in the final debugging phases of my fancy stroke function that does various end treatments.   It's not clear how this will integrate with the stroke module you wrote.  It seems like we want both somehow, because they have different limitations.   (I have struggled in the past to do stuff to the ends of strokes that I had created via difference of polygons.)  

Also I've been using "closed" to refer to a path being closed and you used "close", I guess as a verb to say "close the path".  I like my usage better, of course.  In the case of round_covers the path isn't being changed by the algorithm, so "close=true" doesn't make sense there.  So that's the case for using "closed" instead of "close"---some functions act on paths and just need to know.

revarbat, 2019-05-15:
I've committed `roundcorners.scad`, and the docs are [here](https://github.com/revarbat/BOSL2/wiki/roundcorners.scad).

There's obviously something still broken, based on the results of Examples 3 and 4.

This is a preliminary checkin, as I still plan to extract `circular_arc()`

revarbat, 2019-05-15:
Do we want `bezcorner()` and `circlecorner()` exposed?  If not, we should prefix their names with a `_`.

adrianVmariano, 2019-05-15:
I took a look at the docs and actually the only thing that's broken is that $fs and $fn were badly set.  Arguably the test for this needs to be more aggressive.  It seems that the default is $fs=2 and $fn=0 which gave N=1 and so I think it didn't compute any points.   Maybe change the last computation in bezcorner to read

```
     N = max(3,
               $fn>0 ? $fn :
               $fs>0 ? ceil(bezier_segment_length(P)/$fs) :
                       3),
```

But also, the examples ought to be run with a decently large $fn or small $fs so that things look nice. Like $fs=0.1 or $fn=50.

adrianVmariano, 2019-05-15:
I pondered the question of whether to expose bezcorner and circlecorner and I'm leaning to keeping them hidden.  It seems like users should use functions that call these, not call them directly.   It's also not obvious that round_corners needs its own file.  But I guess we can shuffle things around later.

adrianVmariano, 2019-05-15:
Like I think round_corners() cooperates with rounded_extrude(), which perhaps needs to be renamed to rounded_sweep(), if we're going to use the nomenclaturel of "sweep" for modules that take a path and "extrude" for modules that take geometry.

adrianVmariano, 2019-05-17:
Here's a non-polygon (not closed) example for rounded corners:

```
zigzagx = [-10, 0, 10, 20, 29, 38, 46, 52, 59, 66, 72, 78, 83, 88, 92, 96, 99, 102, 112];
zigzagy = concat([0], flatten(replist([-10,10],8)), [-10,0]);
zig = zip(zigzagx,zigzagy);
echo(zigzagy);
$fs=.1;$fa=1;

polygon(stroke(zig,thickness=1));   // Original shape

fwd(20)            // Smooth all corners with a cut of 4 and curvature parameter 0.6
  polygon(stroke(roundcorners(zig,all=[4,0.6],closed=false, curve="smooth", type="cut"),thickness=1));

fwd(40)            // Smooth all corners with circular arcs and a cut of 4
  polygon(stroke(roundcorners(zig,all=[4,0.6],closed=false, curve="circle", type="cut"),thickness=1));

                   // Smooth all corners with a circular arc and radius 1.5 (close to maximum possible)
fwd(60)            // Note how the different points are cut back by different amounts
  polygon(stroke(roundcorners(zig,all=1.5,closed=false, curve="circle", type="radius"),thickness=1));
```

You'll need to fix the stroke calls, I think. 

And here are some examples for 3d.  I think the module you recommended, trace_polyline() was terrible.  It doesn't do any sort of blending at the corners, so it's not at all good for demonstrating curves.  

```
$fn=36;
// Construct a square spiral path in 3d
square = [[0,0],[1,0],[1,1],[0,1]];
spiral = flatten(replist(concat(square,reverse(square)),5));
z= list_range(40)*.2+[for(i=[0:9]) each [i,i,i,i]];
   // Make rounding parameters, which get larger up the spiral
   // and set the smoothing parameter to 1.  (Could use circular arc instead?)
rvect = zip([for(i=[0:9]) each [i,i,i,i]]/20,replist(1,40));  
rounding = [for(i=rvect) [i]];  // Needed because zip removes a list level
path3d = zip([spiral,z,rounding]);
rpath = roundcorners(path3d, curve="smooth", type="joint",closed=false);
extrude_2dpath_along_3dpath( path2d_regular_ngon(n=36, r=.1), rpath);

$fn=36; 
list1= [[2.88736, 4.03497, 6.37209], [5.68221, 9.37103, 0.783548], [7.80846, 4.39414, 1.84377],
        [0.941085, 5.30548, 4.46753], [1.86054, 9.81574, 6.49753], [6.93818, 7.21163, 5.79453]];
list2= [[1.07907, 4.74091, 6.90039], [8.77585, 4.42248, 6.65185], [5.94714, 9.17137, 6.15642],
        [0.66266, 6.9563, 5.88423], [6.56454, 8.86334, 9.95311], [5.42015, 4.91874, 3.86696]];

extrude_2dpath_along_3dpath(path2d_regular_ngon(n=36,r=.1),roundcorners(list1,closed=false, curve="smooth", type="cut", all=.65));
right(6)     // Could change this to use "circle" once it is fixed for 3d
  extrude_2dpath_along_3dpath(path2d_regular_ngon(n=36,r=.1),roundcorners(list2,closed=false, curve="smooth", type="cut", all=.75));  
```

Something changed in the matrix libraries that broke circular_arc, but the new arc works (once the projection functions are there) so once it is inserted into round_corner that should work.

adrianVmariano, 2019-06-08:
I think if you add those examples to round_corners then we can close this issue.

closed 2019-06-22 by revarbat