BelfrySCAD/BOSL2 #12: is_def vanished

adrianVmariano, 2019-05-10:
So I know it's not necessary, but I got used to it.  Why did you remove it?

revarbat, 2019-05-12:
It went away with all the other backwards compatability is_* type testing functions in BOSL2. I could bring it back, though.  It was the only reversed test.

adrianVmariano, 2019-05-12:
It wasn't really a compatibility function, since it doesn't exist at all in OpenSCAD.  I mean, we can live without it by writing !is_undef, but it seemed convenient and perhaps slightly more clear in the code to write is_def, without the double negative.  I have to admit I'm a little unclear on when it's ok to write foo==undef or foo!=undef vs using is_undef.

adrianVmariano, 2019-05-12:
It looks like you kept is_vector().

revarbat, 2019-05-12:
Re-added in commit b1c7e7129458e4c3687dfae4b4fb5dd4ece32618

revarbat, 2019-05-12:
If a variable was not in the declared argument list, you used to be able to tell if it existed by checking if it was `!=undef`.  Nowadays, a variable set undef is testable, but undeclared variables throw a deprecation warning.  This was mostly annoying regarding special \$variables.

closed 2019-05-12 by revarbat