BelfrySCAD/BOSL2 #13: preliminary pathoffset and rounded_extrude

adrianVmariano, 2019-05-11:
Check this out and let me know what you think.   It seems like something wrong with bezier functions in bosl2.  I had to change the include in roundcorners to the old version or I got errors, but no time to troubleshoot.   

```
use<BOSL2/transforms.scad>
use<BOSL2/std.scad>

// Opposite of bselect:  Sets array[i] if indexset[i] is true, taking values sequentially from valuelist 
function array_set(indexset, valuelist, _i=0, _v=0,array=[]) = 
            _i==len(indexset) ? array : 
            array_set(indexset, valuelist, _i+1, _v+(indexset[_i]?1:0), concat(array, [ indexset[_i]?valuelist[_v]:0]));

// index is a boolean array.  Return a list where array[i] is on the list if index[i] is true
function bselect(array,index) = [for(i=[0:len(array)-1]) if (index[i]) array[i]];

function circular_arc(center, p1, p2, N) = let(
   angle = pathangle([p1,center,p2]),
   v1 = p1-center,
   v2 = p2-center
   )
   len(center)==2 ?
     let(dir = sign(v1.x*v2.y-v1.y*v2.x),   // z component of cross product
         r=norm(v1))
         assert(dir != 0, "Colinear inputs don't define a unique arc")
         [for(i=[0:N-1])
            let(theta=atan2(v1.y,v1.x)+i*dir*angle/(N-1))
              r*[cos(theta),sin(theta)]+center] :
     let(axis = cross(v1,v2))
         assert( axis != [0,0,0], "Colinear inputs don't define a unique arc")
         [for(i=[0:N-1])
            matrix3_rot_by_axis(axis, i*angle/(N-1)) * v1 + center];

function is_def(x) = !is_undef(x);

// Compute offset to the input path.
// The offset is given as delta or r (as is done with the offset command)
// and a positive value shifts to the left of the path direction and a
// negative value to the right.
//
// Specify closed=true if you want a closed curve
//
// maxstep specifies the maximum step size for roundovers.  

// Function: pathoffset()
//
// Description:
//   Takes an input path and returns a path offset by the specified amount.  As with offset(), you can use
//   r to specify rounded offset and delta to specify offset with corners.  Positive offsets shift the path
//   to the left (relative to the direction of the path).
//
//   When offsets shrink the path, segments cross and become invalid.  By default pathoffset checks for this situation. 
//   This check takes O(N^2) time and can be disabled if you don't need it by setting check_valid=false.
//   In some cases the check may mistakenly eliminate segments that should have been included.  If this
//   happens, increase `quality` to 2 or 3.  Run time will increase.  You may be able to save run time
//   in some cases by setting `quality` to 0.
//
//   For construction of polyhedra pathoffset() can also return face lists.  These list faces between the 
//   original path and the offset path where the vertices are ordered with the original path first,
//   starting at `firstface_index` and the offset path vertices appearing afterwords.  The direction
//   of the faces can be flipped using `flip_faces`.  When you request faces the return value
//   is a list: [offset_path, face_list].  
//
// Arguments:
//   path = the path to process.  A list of 2d points.
//   r = offset radius.  Distance to offset.  Will round over corners.
//   delta = offset distance.  Distance to offset with pointed corners.
//   closed = path is a closed curve. Default: False.
//   check_valid = perform segment validity check.  Default: True.
//   quality = validity check quality parameter, a small integer.  Default: 1.
//   return_faces = return face list.  Default: False.
//   firstface_index = starting index for face list.  Default: 0.
[... 418 lines cut ...]

rounded_extrude(rounded_star, height=20, r1=4, r2=1, steps=15, edge1="round", edge2="round");

right(45) rounded_extrude(rounded_star, height=20, r1=-4, r2=1, steps=15, edge1="round", edge2="round");

// Check out the rounded corners on the chamfer
right(90) rounded_extrude(rounded_star, height=20, r1=4, r2=4, steps=15, edge1="teardrop", edge2="chamfer");

left(45)
difference(){
  rounded_extrude((star), height=20, r1=4, r2=1, steps=15, edge1="sharp", edge2="sharp");
  //up(2)rounded_extrude(pathoffset(reverse(rounded_star),r=-.5,closed=true), height=18, r1=1, r2=-1, steps=15,extra2=1);
  up(2)rounded_extrude(pathoffset(reverse(star),r=-2,closed=true), height=18, r1=7, r2=-1, steps=15,extra2=1);
}

*/

// Example 6

/*
path = 2*[for(theta=[0:360]) [theta/75,sin(theta)]];
color("red")stroke(path,thickness=.1);
stroke(pathoffset(path, r=1.75), thickness=.1);
*/
```

revarbat, 2019-05-12:
Well, it looks like my shapes2d.scad checkin implements some of the things in this file, as well as a stroke() module that seems to correspond to stroke2() here.

adrianVmariano, 2019-05-31:
Here's my latest code, with three functions, path_offset(), newstroke() and rounded_sweep().  I think the code is solid for all of these and the main thing we need to figure out is whether the interface is good.  Did I implement the right features?  Can rounded_sweep() be generalized in some useful way?  What it really does is make a polyhedron by connecting together successive outputs from path_offset().   But there's a gotcha in that you need to do the offsets sequentially to be able to track how to link up the polygons.  (So you can't do a bunch of offsets from a base shape, which would be better---you have to do an offset and then offset from that, and then offset from that.)  

```
include<BOSL2/std.scad>
include<BOSL2/beziers.scad>
//include<BOSL2/roundcorners.scad>
include<roundcorners3.scad>    // Need this until the BOSL2 version is updated

// Opposite of bselect:  Sets array[i] if indexset[i] is true, taking values sequentially from valuelist 
function array_set(indexset, valuelist) =
   let ( trueind = search([true], indexset,0)[0])
   concat(
     list_set(trueind, valuelist),           // Fill in all of the values
     replist(0,len(indexset)-max(trueind)-1));  // Add trailing values so length matches indexset
     
// index is a boolean array.  Return a list where array[i] is on the list if index[i] is true
function bselect(array,index) = [for(i=[0:len(array)-1]) if (index[i]) array[i]];

// maxstep specifies the maximum step size for roundovers.  

// Function: path_offset()
//
// Description:
//   Takes an input path and returns a path offset by the specified amount.  As with offset(), you can use
//   r to specify rounded offset and delta to specify offset with corners.  Positive offsets shift the path
//   to the left (relative to the direction of the path).
//
//   When offsets shrink the path, segments cross and become invalid.  By default path_offset checks for this situation. 
//   To test validity the code checks that segments have distance larger than (r or delta) from the input path.  
//   This check takes O(N^2) time and may mistakenly eliminate segments you wanted included in various situations, 
//   so you can disable it if you wish by setting check_valid=false.  Another situation is that the test is not
//   sufficiently thorough and some segments persist that should be eliminated.  In this case, increase `quality` 
//   to 2 or 3.  (This increases the number of samples on the segment that are checked.)  Run time will increase.
//   In some situations you may be able to decrease run time by setting quality to 0, which causes only segment
//   ends to be checked.  
//
//   For construction of polyhedra path_offset() can also return face lists.  These list faces between the 
//   original path and the offset path where the vertices are ordered with the original path first,
//   starting at `firstface_index` and the offset path vertices appearing afterwords.  The direction
//   of the faces can be flipped using `flip_faces`.  When you request faces the return value
//   is a list: [offset_path, face_list].  
//
// Arguments:
//   path = the path to process.  A list of 2d points.
//   r = offset radius.  Distance to offset.  Will round over corners.
//   delta = offset distance.  Distance to offset with pointed corners.
//   closed = path is a closed curve. Default: False.
//   check_valid = perform segment validity check.  Default: True.
//   quality = validity check quality parameter, a small integer.  Default: 1.
//   return_faces = return face list.  Default: False.
//   firstface_index = starting index for face list.  Default: 0.
//   flip_faces = flip face direction.  Default: false
function path_offset(path, r=undef, delta=undef, maxstep=.1, closed=false, check_valid=true, quality=1,
                    return_faces=false, firstface_index=0, flip_faces=false) =
   let(rcount = num_defined([r,delta]))
   assert(rcount==1,"Must define exactly one of 'delta' and 'r'")
   let(
     quality = max(0,round(quality)),
     d = is_def(r)? r : delta,
     shiftsegs = [for(i=[0:len(path)-1]) _shift_segment(select(path,i,i+1), d)],
     // good segments are ones where no point on the segment is less than distance d from any point on the path
     good = check_valid ? _good_segments(path, abs(d), shiftsegs, closed, quality) : replist(true,len(shiftsegs)),
     goodsegs = bselect(shiftsegs, good),
     goodpath = bselect(path,good)
   )
   assert(len(goodsegs)>0,"Offset of path is degenerate")
   let(
        // Extend the shifted segments to their intersection points
     sharpcorners = [for(i=[0:len(goodsegs)-1]) _segment_extension(select(goodsegs,i-1), select(goodsegs,i))],
        // If some segments are parallel then the extended segments are undefined.  This case is not handled
[... 826 lines cut ...]

polygon(newstroke(lpath, thickness=2, rounded=false,start=["roundover",-.3, -20], end=["roundover",-.3, -35]));

// Using "absolute" angles to align with the x and y axes here:
fwd(3)
   polygon(newstroke(apath, thickness=2, rounded=false,start=["roundover",-1, 90, "absolute"], end=["roundover",-.5, 0, "absolute"]));

fwd(10){
   polygon(newstroke(arc, thickness=1, rounded=false, start=["roundover",-.2, -45], end=["roundover",-.2, 20]));

}

// Different radius on the two corners at an end
fwd(15){
    polygon(newstroke(arc, thickness=1, rounded=false, start=["roundover",[-.4,.3], -45], end=["roundover",[.2,.1], 20]));

}
*/

//
// end of demos and examples
//
////////////////////////////////////////////////////////////////////////////////////////

```

adrianVmariano, 2019-06-13:
Here's the latest update.  I think the only change is that I've stripped the functions that got incorporated into the library already, and the examples have been expanded and clarified. 

```
include<BOSL2/std.scad>
include<BOSL2/beziers.scad>
include<BOSL2/roundcorners.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>

// Opposite of bselect:  Sets array[i] if indexset[i] is true, taking values sequentially from valuelist 
function array_set(indexset, valuelist) =
   let ( trueind = search([true], indexset,0)[0])
   concat(
     list_set(trueind, valuelist),           // Fill in all of the values
     replist(0,len(indexset)-max(trueind)-1));  // Add trailing values so length matches indexset
     
// index is a boolean array.  Return a list where array[i] is on the list if index[i] is true
function bselect(array,index) = [for(i=[0:len(array)-1]) if (index[i]) array[i]];

// maxstep specifies the maximum step size for roundovers.  

// Function: path_offset()
//
// Description:
//   Takes an input path and returns a path offset by the specified amount.  As with offset(), you can use
//   r to specify rounded offset and delta to specify offset with corners.  Positive offsets shift the path
//   to the left (relative to the direction of the path).
//
//   When offsets shrink the path, segments cross and become invalid.  By default path_offset checks for this situation. 
//   To test validity the code checks that segments have distance larger than (r or delta) from the input path.  
//   This check takes O(N^2) time and may mistakenly eliminate segments you wanted included in various situations, 
//   so you can disable it if you wish by setting check_valid=false.  Another situation is that the test is not
//   sufficiently thorough and some segments persist that should be eliminated.  In this case, increase `quality` 
//   to 2 or 3.  (This increases the number of samples on the segment that are checked.)  Run time will increase.
//   In some situations you may be able to decrease run time by setting quality to 0, which causes only segment
//   ends to be checked.  
//
//   For construction of polyhedra path_offset() can also return face lists.  These list faces between the 
//   original path and the offset path where the vertices are ordered with the original path first,
//   starting at `firstface_index` and the offset path vertices appearing afterwords.  The direction
//   of the faces can be flipped using `flip_faces`.  When you request faces the return value
//   is a list: [offset_path, face_list].  
//
// Arguments:
//   path = the path to process.  A list of 2d points.
//   r = offset radius.  Distance to offset.  Will round over corners.
//   delta = offset distance.  Distance to offset with pointed corners.
//   closed = path is a closed curve. Default: False.
//   check_valid = perform segment validity check.  Default: True.
//   quality = validity check quality parameter, a small integer.  Default: 1.
//   return_faces = return face list.  Default: False.
//   firstface_index = starting index for face list.  Default: 0.
//   flip_faces = flip face direction.  Default: false
function path_offset(path, r=undef, delta=undef, maxstep=.1, closed=false, check_valid=true, quality=1,
                    return_faces=false, firstface_index=0, flip_faces=false) =
   let(rcount = num_defined([r,delta]))
   assert(rcount==1,"Must define exactly one of 'delta' and 'r'")
   let(
     quality = max(0,round(quality)),
     d = is_def(r)? r : delta,
     shiftsegs = [for(i=[0:len(path)-1]) _shift_segment(select(path,i,i+1), d)],
     // good segments are ones where no point on the segment is less than distance d from any point on the path
     good = check_valid ? _good_segments(path, abs(d), shiftsegs, closed, quality) : replist(true,len(shiftsegs)),
     goodsegs = bselect(shiftsegs, good),
     goodpath = bselect(path,good)
   )
   assert(len(goodsegs)>0,"Offset of path is degenerate")
   let(
        // Extend the shifted segments to their intersection points
     sharpcorners = [for(i=[0:len(goodsegs)-1]) _segment_extension(select(goodsegs,i-1), select(goodsegs,i))],
[... 757 lines cut ...]
$fn=36;

polygon(newstroke(lpath, thickness=2, rounded=false,start=["roundover",-.3, -20], end=["roundover",-.3, -35]));

// Using "absolute" angles to align with the x and y axes here:
fwd(3)
   polygon(newstroke(apath, thickness=2, rounded=false,start=["roundover",-1, 90, "absolute"], end=["roundover",-.5, 0, "absolute"]));

fwd(10){
   polygon(newstroke(arc, thickness=1, rounded=false, start=["roundover",-.2, -45], end=["roundover",-.2, 20]));

}

// Different radius on the two corners at an end
fwd(15){
    polygon(newstroke(arc, thickness=1, rounded=false, start=["roundover",[-.4,.3], -45], end=["roundover",[.2,.1], 20]));

}
*/

//
// end of demos and examples
//
////////////////////////////////////////////////////////////////////////////////////////
```

adrianVmariano, 2019-06-22:
This update includes a chamfer option.  Seems to work.  Hopefully I didn't break anything.   

```
include<BOSL2/std.scad>
include<BOSL2/beziers.scad>
include<BOSL2/roundcorners.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>

//include<roundcorners3.scad>    // Need this until the BOSL2 version is updated

// Opposite of bselect:  Sets array[i] if indexset[i] is true, taking values sequentially from valuelist 
function array_set(indexset, valuelist) =
   let ( trueind = search([true], indexset,0)[0])
   concat(
     list_set(trueind, valuelist),           // Fill in all of the values
     replist(0,len(indexset)-max(trueind)-1));  // Add trailing values so length matches indexset
     
// index is a boolean array.  Return a list where array[i] is on the list if index[i] is true
function bselect(array,index) = [for(i=[0:len(array)-1]) if (index[i]) array[i]];

// maxstep specifies the maximum step size for roundovers.  

// Function: path_offset()
//
// Description:
//   Takes an input path and returns a path offset by the specified amount.  As with offset(), you can use
//   r to specify rounded offset and delta to specify offset with corners.  Positive offsets shift the path
//   to the left (relative to the direction of the path).
//
//   When offsets shrink the path, segments cross and become invalid.  By default path_offset checks for this situation. 
//   To test validity the code checks that segments have distance larger than (r or delta) from the input path.  
//   This check takes O(N^2) time and may mistakenly eliminate segments you wanted included in various situations, 
//   so you can disable it if you wish by setting check_valid=false.  Another situation is that the test is not
//   sufficiently thorough and some segments persist that should be eliminated.  In this case, increase `quality` 
//   to 2 or 3.  (This increases the number of samples on the segment that are checked.)  Run time will increase.
//   In some situations you may be able to decrease run time by setting quality to 0, which causes only segment
//   ends to be checked.  
//
//   For construction of polyhedra path_offset() can also return face lists.  These list faces between the 
//   original path and the offset path where the vertices are ordered with the original path first,
//   starting at `firstface_index` and the offset path vertices appearing afterwords.  The direction
//   of the faces can be flipped using `flip_faces`.  When you request faces the return value
//   is a list: [offset_path, face_list].  
//
// Arguments:
//   path = the path to process.  A list of 2d points.
//   r = offset radius.  Distance to offset.  Will round over corners.
//   delta = offset distance.  Distance to offset with pointed corners.
//   chamfer = chamfer corners when you specify `delta`.  Default: false  
//   closed = path is a closed curve. Default: False.
//   check_valid = perform segment validity check.  Default: True.
//   quality = validity check quality parameter, a small integer.  Default: 1.
//   return_faces = return face list.  Default: False.
//   firstface_index = starting index for face list.  Default: 0.
//   flip_faces = flip face direction.  Default: false
function path_offset(path, r=undef, delta=undef, chamfer=false, maxstep=.1, closed=false, check_valid=true, quality=1,
                    return_faces=false, firstface_index=0, flip_faces=false) =
   let(rcount = num_defined([r,delta]))
   assert(rcount==1,"Must define exactly one of 'delta' and 'r'")
   let(
     chamfer = is_def(r) ? false : chamfer,
     quality = max(0,round(quality)),
     d = is_def(r)? r : delta,
     qqr=echo(chamfer=chamfer, delta=delta, r=r, d=d),     
     shiftsegs = [for(i=[0:len(path)-1]) _shift_segment(select(path,i,i+1), d)],
     // good segments are ones where no point on the segment is less than distance d from any point on the path
     good = check_valid ? _good_segments(path, abs(d), shiftsegs, closed, quality) : replist(true,len(shiftsegs)),
     goodsegs = bselect(shiftsegs, good),
     goodpath = bselect(path,good)
[... 776 lines cut ...]
$fn=36;

polygon(newstroke(lpath, thickness=2, rounded=false,start=["roundover",-.3, -20], end=["roundover",-.3, -35]));

// Using "absolute" angles to align with the x and y axes here:
fwd(3)
   polygon(newstroke(apath, thickness=2, rounded=false,start=["roundover",-1, 90, "absolute"], end=["roundover",-.5, 0, "absolute"]));

fwd(10){
   polygon(newstroke(arc, thickness=1, rounded=false, start=["roundover",-.2, -45], end=["roundover",-.2, 20]));

}

// Different radius on the two corners at an end
fwd(15){
    polygon(newstroke(arc, thickness=1, rounded=false, start=["roundover",[-.4,.3], -45], end=["roundover",[.2,.1], 20]));

}

*/

//
// end of demos and examples
//
////////////////////////////////////////////////////////////////////////////////////////

adrianVmariano, 2019-06-23:
I realized two things.   One is that I forgot to remove all the debugging echo() calls from the code.  The second is that perhaps rounded_sweep needs to take alignment arguments, though I'm not sure if it's going to be clear what the anchor points should be.

revarbat, 2019-08-04:
With the addition of `offset_stroke()`, should this be closed?

closed 2019-08-04 by adrianVmariano