BelfrySCAD/BOSL2 #14: replace any_defined with num_defined, remove remove_undefs, add all_defined

adrianVmariano, 2019-05-11:
It seems like remove_undefs is a weird operation, not something I could think of a reason to use.  You use it to define first_defined.  Does any other application exist? I think num_undefs is more useful (and can be used to achieve the result of any_defined with a simple check).   I've found it nice to count defined entries for parameter checking and having num_defined is nice for that, and better than len(remove_undefs()).  So here's a tail recursive version of first_defined:

    function first_defined(v,ind=0) = ind<len(v) && is_undef(v[ind]) ? first_defined(v,ind+1) : (ind==len(v) ? undef : v[ind]);

Not sure that's necessary, but it can handle long lists and may as well be robust when possible.  (I can't think of a use case other than the parameter selection one in the code, which obviously doesn't involve long lists.)    

It might also be useful to have all_defined() for checking validity of a calculation, for example.  
```
function all_defined(v) = is_list(v) ? _all_defined_list(v) : !is_undef(v);
function _all_defined_list(v,ind=0) = ind<len(v) && all_defined(v[ind]) ? _all_defined_list(v,ind+1) : (ind==len(v) ? true : false);
```
I needed something like this in pathoffset, though not so sophisticated.  (I didn't need to check inside list members.)  But it seems like that's the right way to define it, so [[3,4],[5,6],[undef,undef],[5,6]] tests as being bad.   And this one needs to be tail recursive to handle potential long inputs.  I found when I tested speed with _makefaces_recursive in my path offset code that making it tail recursive didn't affect speed but did prevent errors on long inputs.

revarbat, 2019-05-14:
Implemented:
```
function num_defined(v) = sum([for (x = v) is_undef(x)? 0 : 1]);
function any_defined(v) = num_defined(v)>0;
function all_defined(v) = num_defined(v)==len(v);
```
Keeping it all in list comprehensions.
**Not** removing `remove_undefs()` though I can be talked into renaming it to `strip_undefs()`.

adrianVmariano, 2019-05-14:
I don't think there's any harm in having remove_undefs().  I mean, I guess there's the question of when extra clutter in the library becomes a problem.  But I can't think of a use case for it---it seems useless.  I like strip_undefs() better but I don't think it makes much difference and I don't think I like it enough better to change it unless there are other "strip" functions.   I think I did see count_true and so that's inconsistent with num_defined, so probably one of them should change.  

I inclined to think that all_defined should recurse down all levels so that it tells you whether the argument is entirely valid, completely free from undefs.  It should not return true on [[undef]], or on [[1,2],[undef,undef],[3,4]].  Those do not look "all" defined to me.  

I suppose this does raise a question about consistency with the other functions.   What should any_defined return for [[undef]] or [[undef,undef],[undef,undef]].  Is "true" the right answer?

revarbat, 2019-05-14:
Hmm.  `count_true()` to `num_true()`?

revarbat, 2019-05-14:
Use case for `remove_undefs()`: Generating a list of data, where some of it can be invalid, and should be ignored.  Just return `undef` for those, then remove them once the list is done.  Mind you, there are other ways of doing this, especially with list comprehensions.

revarbat, 2019-05-14:
Not sure if list comprehensions or recursion would be faster for `first_defined()` on average, given the short lists for general usage.

revarbat, 2019-05-14:
Another use case for `remove_undefs()` you generated a list of points, and some were undefined.  You want to plot the rest.  `place_copies(remove_undefs(pts)) sphere();`

adrianVmariano, 2019-05-14:
Trouble is that I see two ways you do this.  In one case you have a set of points that looks like [[3,4,7],[4,5,9],[3,6,undef],[4,5,7]].  And your function doesn't handle that case.  The other case that seems likely is that you have a list of locations like [[3,4],[5,6],[3,4]] and then values [3,undef,7] and it is no use to remove undefs from the one list if you don't also do it from the other.   Heck, even if your data looks like [[3,4],[5,6],[undef,undef],[5,6]] it still doesn't work.  

Removing undefs seems limited because it doesn't track position or associated information.  

Of course...with the discussion being all in the abstract, who knows what sort of case might arise.

adrianVmariano, 2019-05-14:
Oh, for first_defined if it's used on 4 element lists for parameter selection it obviously makes no difference how you implement it.  The question is if someone calls it on a 400k long list, then what.  Will it happen?  Who knows.  It's a library...someone might do it for some reason, so I'm inclined to think it should be written to be efficient in that case, and presumably recursion is the way to go there, since the first_defined could be element 15 out of 400k.

adrianVmariano, 2019-05-14:
I suppose another possibility with things like any_defined and all_defined would be to have a recursive=true option.

revarbat, 2019-06-24:
Keeping this open, as recursion may be useful.

revarbat, 2019-07-11:
Added `recursive=` options in 58857b439a210418eb098357da71c38bf2339f7f

closed 2019-07-11 by revarbat