BelfrySCAD/BOSL2 #16: matrix3_mult?  Isn't this just a product (of anything)?

adrianVmariano, 2019-05-12:
Isn't this just using the built-in * operator to multiply a list of items?  

I think it should be called "prod" (matlab name for same thing, like "sum"), or maybe product() if you think prod() is too short and the optional parameter m deleted, since 
product(list)*m is perfectly fine.  And in fact I'm not sure if the existing code does that or m*product(list), so the more explicit notation is better.  (I suppose what I really mean is don't expose the accumulator that you need for tail recursion as part of the interface.)   

You can do the product "backwards" and it will be fine for everything else, since matrices are the only non-commutative object that can be multiplied.

revarbat, 2019-05-12:
The accumulator is deliberately exposed in this case, for continuing calculations from previous matrix calculations.  You *could* just put that matrix first in the list, I suppose.  Currently putting it as `m` avoids a matrix mult op against `ident()`, but that could be eliminated with better code.

Mostly this function exists so that I don't have to remember if I need to pre or post multiply to apply a transform matrix.

adrianVmariano, 2019-05-12:
But you can continue calculations by just multiplying by the previous thing.  You don't need to introduce ident.  In other words,  

matrix3_mult(matrix_list, m=A) is the same as matrix3_mult(matrix_list)*A and to me it's hard to see why the former would be preferred.  Of course, there might also be situations (recently discussed on the forum) where you prefer to apply the transformations in the other order with transposed matrices.  

I guess I'm thinking if this is an internal helper function, then hide it.  And if it's meant for users, it should be recast as a generic "product of elements in a list" function rather than something special for matrices.  We have sum() after all, so product() doesn't seem unreasonable.

revarbat, 2019-05-12:
I don't want to have to remember if I need to pre or post multiply, because between times I have to code it, I keep forgetting which it is.  This has happened repeatedly.  However, `m` isn't needed to be exposed, since it's the same as `matrix3_mult(concat([m],matrices))` or `matrix3_mult([m,a,b,c])`.  At least it is since a5065c1595ec07837d049336c6ad40934d817bac

revarbat, 2019-05-12:
Really, `matrix3_mult()` and `matrix4_mult()` aren't meant to be generic matrix multipliers, since you can do that with `*` if you know what you are doing.  They are meant to add 2D (matrix3) and 3D (matrix4) transformation operations together in a straightforward order. Maybe if I renamed all the `matrix3_` prefixes to `affine2d_` and `matrix4_` to `affine3d_` and rename `matrices.scad` to `affine.scad` this will stop confusing.

revarbat, 2019-05-12:
Perhaps `affine[23]d_add()` instead of mult.

revarbat, 2019-05-12:
Looks like `matrix_transpose()` is the only generic matrix function in matrices.scad

revarbat, 2019-05-12:
I do like the idea of `product()` like `sum()`

revarbat, 2019-05-12:
Hmm.  Rename and move `transpose()` to arrays.scad.

adrianVmariano, 2019-05-12:
I was going to suggest renaming matrix_transpose() to transpose().  

I was also wondering if there's a generic need for determinants or not.  I've needed det2() a bunch.  But not yet ever det3().

revarbat, 2019-05-12:
There probably is, but I cannot actually recall what a determinant is, or is used for.  All my matrix usage has been with affine transformations.  A determinant function can be put in math.scad, though.

revarbat, 2019-05-12:
I'm about to check in transpose in arrays.scad.  It's generalized for non-matrices.

revarbat, 2019-05-12:
`transpose()` Committed in 78e3cd3c270841c4e13cc0826fef605baebac3fb

adrianVmariano, 2019-05-12:
What do you mean by non-matrices?  It can handle lists where each entry has a different length?

Determinant is a pain to calculate and it's not clear that there's any reason to supply a general version.   The det2() function is like a cross product of 2d vectors and seems to arise all the time.   When I wrote circular_arc_3pts some people were proposing a solution involving larger matrix determinants, but I ended up avoiding that and using something much simpler, involving the intersection of perpendicular bisectors.  (And my code for line intersection relies on det2.)

adrianVmariano, 2019-05-12:
Hmm.  Actually testing by intersecting is tricky because it can intersect both if the stroke curves around.

revarbat, 2019-05-14:
`matrices.scad` renamed to `affine.scad`.  `matrix3_*` renamed to `affine2d_*`.  `matrix4_*` renamed to `affine3d_*`.  `affine[23]d_mult()` renamed to `affine[23]d_chain()`
Final commit for this: 9a3e2b71f24b86c8739ae00426e679d94f3805b7

closed 2019-05-14 by revarbat