BelfrySCAD/BOSL2 #17: arc

adrianVmariano, 2019-05-16:
```
function arc(N,center,r,d,start,angle,points) =
    // First try for 2d arc specified by angles
    is_def(center) && is_def(start) ?
       let(parm_ok = is_undef(points) && 
                     (is_vector(start) && len(start)==2 && is_undef(angle) ||
                      is_num(start) && is_num(angle)))
       assert(parm_ok,"Invalid parameters")
       let(
       angle = is_def(angle) ? angle : start[1]-start[0],
       start = is_vector(start) ? start[0] : start,
       radius = get_radius(r=r,d=d),
       N = max(3,N))
       [for(i=[0:N-1]) let(theta = start + i*angle/(N-1)) r*[cos(theta),sin(theta)]+center] :
    assert(is_list(points),"Invalid parameters")
    // Arc is 3d, so transform points to 2d and make a recursive call, then remap back to 3d
    len(points[0])==3 ?
      let(
        thirdpoint = is_def(center) ? center : points[2],
        center2d = is_def(center) ? xyz_to_planar(center,thirdpoint,points[0],points[1]) : undef,
        points2d = new_xyz_to_planar(points,thirdpoint,points[0],points[1])
      )
      new_planar_to_xyz(arc(N,center=center2d,points=points2d),thirdpoint,points[0],points[1]) :
    // Arc defined by center plus two points, will have radius defined by center and points[0]
    // and extent defined by direction of point[1] from the center
    is_def(center) ?
      let(
         angle = pathangle([points[0],center,points[1]]),
         v1 = points[0]-center,
         v2 = points[1]-center,
         dir = sign(det2([v1,v2])),   // z component of cross product
         r=norm(v1)
       )
       assert(dir!=0,"Collinear inputs don't define a unique arc")
       arc(N,center,r,start=atan2(v1.y,v1.x),angle=dir*angle) :
     // Final case is arc passing through three points, starting at point[0] and ending at point[3]
     let(col = collinear(points[0],points[1],points[2],1e-3))
     assert(!col, "Collinear inputs do not define an arc")
     let(
        center = line_intersection(normal_segment(points[0],points[1]),normal_segment(points[1],points[2])),
             // select order to be counterclockwise
        dir=det2([points[1]-points[0],points[2]-points[1]])>0,
        points = dir ? select(points,[0,2]) : select(points,[2,0]),  
        r = norm(points[0]-center),
        theta_start = atan2(points[0].y-center.y, points[0].x-center.x),
        theta_end =  atan2(points[1].y-center.y, points[1].x-center.x),
        angle = posmod(theta_end-theta_start,360),
        arcpts = arc(N,center,r,start=theta_start,angle=angle)
     )
     dir ? arcpts:reverse(arcpts);

```

Dependencies are in the pathoffset and rounded extrude code.   It seems like I work too hard to figure out the angles, but I also found it remarkably hard to get that robust, so that my arc goes the right way.  I tested on a bunch of random triples and didn't notice any problems.

adrianVmariano, 2019-05-17:
Revised version.  I posted the dependencies in another issue.  I added wedge=true to get the center point which replaces pie_slice.  I guess pie=true would be another choice.  One quirk is that with the 3 points mode the center can end up at the end.  Otherwise it's always the first point.  

```
function pathangle(pts) = vector_angle(pts[0]-pts[1], pts[2]-pts[1]);

// arc(N,r,angle)            2d arc from angle zero up to specified angle
// arc(N,r,[angle0,angle1]   2d arc from angle0 up to angle1
// arc(N,r,start=angle0, angle=delta_angle    2d arc from angle0 up to angle0+delta_angle
// arc(N,width=w,thickness=t)  2d arc starting and ending on x-axis with width and thickness given
// arc(N,center=c, points=[p0,p1]  shortest 2d or 3d arc with specified center starting at p0 and 
//                                    ending on the vector defined by [c,p1]
// arc(N,points=[p0,p1,p2]   2d or 3d arc starting at p0, ending at p2 and passing through p1
// arc(N,....,wedge=true)    include center point in output to produce a wedge shape
function arc(N,r,angle,d,cp,points, width, thickness,start,wedge=false) = 
    // First try for 2d arc specified by angles
    is_def(width) && is_def(thickness) ?
       arc(N,points=[[width/2,0], [0,thickness], [-width/2,0]],wedge=wedge) :
    is_def(angle) ?
       let(
            parmok=is_undef(points) && is_undef(width) && is_undef(thickness) &&
                 ((is_vector(angle) && len(angle)==2 && is_undef(start)) || is_num(angle)))
       assert(parmok,"Invalid parameters in arc")
       let(
            cp = is_def(cp) ? cp : [0,0], 
            start = is_def(start)? start : is_vector(angle) ? angle[0] : 0,
            angle = is_vector(angle) ? angle[1]-angle[0] : angle,
            radius = get_radius(r=r,d=d),
            N = max(3,N),
            arcpoints = [for(i=[0:N-1]) let(theta = start + i*angle/(N-1)) r*[cos(theta),sin(theta)]+cp],
            extra = wedge ? [cp] : []
       )
       concat(extra,arcpoints) :
    assert(is_list(points),"Invalid parameters")
    // Arc is 3d, so transform points to 2d and make a recursive call, then remap back to 3d
    len(points[0])==3 ?
      let(
        thirdpoint = is_def(cp) ? cp : points[2],
        center2d = is_def(cp) ? xyz_to_planar(cp,thirdpoint,points[0],points[1]) : undef,
        points2d = new_xyz_to_planar(points,thirdpoint,points[0],points[1])
      )
      new_planar_to_xyz(arc(N,cp=center2d,points=points2d,wedge=wedge),thirdpoint,points[0],points[1]) :
    // Arc defined by center plus two points, will have radius defined by center and points[0]
    // and extent defined by direction of point[1] from the center
    is_def(cp) ?
      let(
         angle = pathangle([points[0],cp,points[1]]),
         v1 = points[0]-cp,
         v2 = points[1]-cp,
         dir = sign(det2([v1,v2])),   // z component of cross product
         r=norm(v1)
       )
       assert(dir!=0,"Collinear inputs don't define a unique arc")
       arc(N,cp=cp,r=r,start=atan2(v1.y,v1.x),angle=dir*angle,wedge=wedge) :
     // Final case is arc passing through three points, starting at point[0] and ending at point[3]
     let(col = collinear(points[0],points[1],points[2],1e-3))
     assert(!col, "Collinear inputs do not define an arc")
     let(
        cp = line_intersection(normal_segment(points[0],points[1]),normal_segment(points[1],points[2])),
             // select order to be counterclockwise
        dir=det2([points[1]-points[0],points[2]-points[1]])>0,
        points = dir ? select(points,[0,2]) : select(points,[2,0]),  
        r = norm(points[0]-cp),
        theta_start = atan2(points[0].y-cp.y, points[0].x-cp.x),
        theta_end =  atan2(points[1].y-cp.y, points[1].x-cp.x),
        angle = posmod(theta_end-theta_start,360),
        arcpts = arc(N,cp=cp,r=r,start=theta_start,angle=angle,wedge=wedge)
     )
     dir ? arcpts:reverse(arcpts);
```

revarbat, 2019-05-27:
If I'm reading this correctly, then #18 is a pre-req for this?

adrianVmariano, 2019-05-27:
Yes, actually #18 and #20 are both required.

revarbat, 2019-05-29:
```
function normal_segment(p1,p2) =
    let(center = (p1+p2)/2)
    [center, center + norm(p1-p2)/2 * line_normal(p1,p2)];
```

revarbat, 2019-05-29:
Implemented in 0836e821f465217a268d234ae3e360f032c5bce4

closed 2019-05-29 by revarbat