BelfrySCAD/BOSL2 #18: add line intersection and determinants

adrianVmariano, 2019-05-16:
Here's code that I've been using in pathoffset that belongs somewhere.  I went ahead and implemented det3 even though I don't have an application for it right now just so it's there.  But I'm not going to do det4 unless someone asks for it.   (The code is either a kind of recursive mess or it's four copies of the det3 code, but each copy slightly different.)  

```
// Line intersection from two segments

// This function returns [p,t,u] where p is the intersection point of
// the lines defined by the two segments, t is the bezier parameter
// for the intersection point on s1 and u is the bezier parameter for
// the intersection point on s2.  The bezier parameter runs over [0,1]
// for each segment, so if it is in this range, then the intersection
// lies on the segment.  Otherwise it lies somewhere on the extension
// of the segment.

function _general_line_intersection(s1,s2) =
  let(  denominator = det2([s1[0],s2[0]]-[s1[1],s2[1]]),
        t=det2([s1[0],s2[0]]-s2)/denominator,
        u=det2([s1[0],s1[0]]-[s1[1],s2[1]])/denominator)
        [denominator==0 ? undef : s1[0]+t*(s1[1]-s1[0]),t,u];

// Returns the intersection point of the lines defined by two segments, or undef
// if the lines are parallel.  
function line_intersection(s1,s2) = let( int = _general_line_intersection(s1,s2)) int[0];

// Returns the intersection point of two segments and undef if they do not intersect
function segment_intersection(s1,s2) = let( int = _general_line_intersection(s1,s2))
        int[1]<0 || int[1]>1 || int[2]<0 || int[2]>1 ? undef : int[0];

// Returns the intersection point of a line defined by the first segment with a segment.
// If the segment doesn't intersect the line it returns undef
function line_segment_intersection(line,segment) = let( int = _general_line_intersection(line,segment))
         int[2]<0 || int[2]>1 ? undef : int[0];
        

// Return true if simple polygon is in clockwise order, false otherwise.
// Results for complex (self-intersecting) polygon are indeterminate
function polygon_clockwise(path) =
  let( 
       minx = min(array_subindex(path,0)),
       lowind = search(minx, path, 0, 0),
       lowpts = select(path, lowind),
       miny = min(array_subindex(lowpts, 1)),
       extreme_sub = search(miny, lowpts, 1, 1)[0],
       extreme = select(lowind,extreme_sub)
     )
  det2(  [select(path,extreme+1)-path[extreme], select(path, extreme-1)-path[extreme]])<0;

// determinant of 2x2 matrix
function det2(M) = M[0][0] * M[1][1] - M[0][1]*M[1][0];

function det3(M) =  M[0][0] * (M[1][1]*M[2][2]-M[2][1]*M[1][2])
                  - M[1][0] * (M[0][1]*M[2][2]-M[2][1]*M[0][2])
                 + M[2][0] * (M[0][1]*M[1][2]-M[1][1]*M[0][2]);

// Returns the segment normal to the input segment
//   normal_segment(p1,p2)
//   normal_segment([p1,p2])
// It returns the segment whose first point is the center of the input and whose second point extends left 
// The length of the segment is half the length of the input.  
function normal_segment(p1,p2) = let(center = (p1+p2)/2)
     [center, center + norm(p1-p2)/2 * normalize([p1.y-p2.y,p2.x-p1.x])];
```

I'm a bit unsure about normal_segment.  It's used by arc().  But maybe it needs to be designed differently to be less particular.

adrianVmariano, 2019-05-16:
Or maybe it's something that shouldn't be exposed and instead a function that returns the normal vector should be exposed.

revarbat, 2019-05-29:
Committed in ff1fa4e505da0716195bfb0cdda82ca4b8e0518f

closed 2019-05-29 by revarbat