BelfrySCAD/BOSL2 #24: list_pad() is slow as written.  Using concat is much faster

adrianVmariano, 2019-05-22:
I did a test on a large list and my run time was 54 s, to lengthen a large list by a small amount.  

Using concat is much faster at only 3 seconds:

```
function list_pad(v, minlen, fill=undef) =
	concat(v,replist(fill,minlen-len(v)));
```

In a test of extended a short list by a large amount the existing code took 33s and the concat version takes 19s.  So the concat version is always better.  This second result was a bit of a surprise, since replist just runs a for loop, but I guess the fact that there's no conditional test in the loop is making the difference?  

I also note that list_trim can be written without the maxlen<1 test by adding a step of 1 into the range.  And the documentation erroneously lists "minlen" as an argument.   And with list_pad rewritten as above, I think list_fit doesn't need the test for l==length.  That case is handled without extra run time cost by letting list_pad do it, where it pads on the empty list.  Then you can avoid the extra assignment and again the code is shorter and simpler.  

I also kind of feel like it makes more sense to supply list_lengths() and let users take max or min of the result instead of list_longest and list_shortest.   Hmmm.  Is there a way to get the index of the max or min?   I guess you can run search after running min/max.

revarbat, 2019-05-27:
I think you are looking for `min_index()` and/or `max_index()`.

revarbat, 2019-05-27:
Fixed in 8f2b9c4ef1977147a3ce7f2a75f23c60083976df

closed 2019-05-27 by revarbat