BelfrySCAD/BOSL2 #30: stroke: change close argument to closed

adrianVmariano, 2019-05-29:
I use the "closed" argument in several functions, so stroke is inconsistent.  I think "closed" makes more sense (describing the input curve as being closed) rather than "close" the verb form, which doesn't always apply.

revarbat, 2019-06-12:
Err, in `stroke()`, the verb form is correct, as `close=true` is a request to draw a closing segment, *not* a denotation that the path is already closed.  Since you can test if the first and last points of a path are the same, I cannot see a reason to ever need to specify that with an argument.

adrianVmariano, 2019-06-12:
Well, this is a matter of interpretation, and I think we need to pick a consistent form.  If you supply n points there is no way to know whether n edges or n-1 edges are intended, so an additional piece of input information to describe the curve is required.  My interpretation is that you write closed=true if the path is closed (n edges) and closed=false if the path is open (n-1 edges).  For stroke() it makes sense to use a verb form, being an instruction to close the path, but for round_corners, that makes less sense.  It is more general to treat this argument as part of a description of the input rather than to treat it as a verb describing what to do with the input.  

I would argue that supplying n+1 points where the first and last points are supposed to be the same is some kind of invitation to trouble (the points not being exactly the same).  It's also in many cases an extra nuisance to duplicate the first point, so I think it's a mistake from a UI design perspective to require it.  To me it makes a lot more sense to require no point duplication with a flag to indicate whether the curve is open or closed.

revarbat, 2019-07-13:
Implemented in e2d286bbf1b162c7c3879786bd48b21c0bcea96f

closed 2019-07-13 by revarbat