BelfrySCAD/BOSL2 #36: span_cube?

adrianVmariano, 2019-06-07:
Did span_cube go away?

revarbat, 2019-06-24:
I vaguely remember removing this when I first forked BOSL2.  Removed for redundancy with `p1`/`p2` in `cuboid()` as I recall.

adrianVmariano, 2019-06-24:
Yeah, it sort of is and sort of isn't.   I feel like so much has changed since I was using my equivalent of span_cube() that I'm not sure how important I think it is, but I found that I never wanted to specify a cube by corners.  I always wanted to specify it by bounds---that was what was natural.   Using corners means breaking up the bounds into the different coordinates, so it was clumsy.

revarbat, 2020-05-24:
```openscad

revarbat, 2020-05-24:
I don't really want to open the can of worms of adding span args to all kinds of shapes.  How about something like:

```openscad
function spans2pts(xspan,yspan,zspan) =
    let(
        p1 = [xspan[0],yspan[0],if(is_def(zspan)) zspan[0]],
        p2 = [xspan[1],yspan[1],if(is_def(zspan)) zspan[1]]
    ) [p1,p2];

p = spans2pts([-10,5],[2,8],[0,20]);
cuboid(p1=p[0], p2=p[1]);
```

adrianVmariano, 2020-05-24:
I think it's too clumsy to be useful.  I'm unlikely to use it.  If you could make it work directly like

cuboid(p=span(....))

then that might be a usable implementation.   When is specifying by corners useful, btw?

revarbat, 2020-05-24:
Mostly when you know the corner positions already.

adrianVmariano, 2020-05-27:
I guess for me it never happens that I know a cube's corners, but it does happen that I know its span.  So the corners option seems less useful than the span option.

closed 2020-07-31 by revarbat