BelfrySCAD/BOSL2 #37: uts screws

adrianVmariano, 2019-06-08:
You suggested I should supply code for UTS screws.  Here is my attempt to do so.   It parses a screw spec and produces a structure with screw data.  I made a head-drawing routine but didn't try to figure out how to draw the threaded shaft, or to put a recess into the head.  

Note also that there's a bunch of string stuff that belongs in a strings library.  I wonder if we should try to get permission to use the regular expression parser from the STring Theory library.   (It's GPL.)

```

function parse_screw_name(name) =
    name[0] == "M" || name[0] == "m" ?
       (let( split = strsplit(substr(name,1),"x"))
       len(split)==1 ? ["metric",atof(split[0]), undef] :
       len(split)==2 ? ["metric",atof(split[0]), atof(split[1])] :
       undef) :
    let(split = strsplit(name,"-"))
    len(split)<1 || len(split)>2 ? undef :
    let(
        diam = split[0][0] == "#" ? split[0] :
               substr(split[0],len(split[0])-2,2)=="''" ? atof(substr(split[0],0,len(split[0])-2)) :
               let(val=fraction2f(split[0]))
               val == floor(val) && val>=0 && val<=12 ? str("#",split[0]) :
               val != val ? atof(split[0]) : val
        )
    len(split)==1 ? ["english", diam, undef] :
    len(split)==2 ? ["english", diam, atof(split[1])] :
                    undef;
       
function atoi(str,base=10) =
    len(str)==0 ? 0 : 
    let(str=downcase(str))
    str[0] == "-" ? -_atoi_recurse(substr(str,1),base,len(str)-2) :
    str[0] == "+" ? _atoi_recurse(substr(str,1),base,len(str)-2) :
    _atoi_recurse(str,base,len(str)-1);

function _atoi_recurse(str,base,i) =
    let( digit = search(str[i],"0123456789abcdef"),
         last_digit = digit == [] || digit[0] >= base ? (0/0) : digit[0])
    i==0 ? last_digit : 
        _atoi_recurse(str,base,i-1)*base + last_digit;

// retrns substring starting from position, pos of specified length
// If length is -1 returns the rest of the string.
//
// default pos=0 (start at beginning of string)
// 
function substr(str, pos=0, len=-1, substr="") =
        assert(len>=-1,"illegal value of len")
	len == 0 || pos>=len(str) ? substr :
	len == -1 ? substr(str, pos, len(str)-pos, substr) :
	substr(str, pos+1, len-1, str(substr, str[pos]));

function strcat(list,_i=0, _result="") =
    _i == len(list) ? _result :
      strcat(list,_i+1,str(_result,list[_i]));

function strsplit_recurse(str,sep,i,result) =
   i == len(sep) ? concat(result,[str]) :
    let( pos = search(sep[i], str),
         end = pos==[] ? len(str) : pos[0]
      )
    strsplit_recurse(substr(str,end+1), sep, i+1,
                    concat(result, [substr(str,0,end)]));
                    
// splits the input string on the separator(s)
// If the separator is a string then the input is broken on any occurrence of a character in the string
// The output will be a list of strings with length n+1 if there are n occurences of separator characters
//
// If separator is a list of characters then split sequentially on each character
// 
function strsplit(str,sep) =
   is_list(sep) ? strsplit_recurse(str,sep,i=0,result=[]) :
   let( cutpts = concat([-1],sort(flatten(search(sep, str,0))),[len(str)]))
[... 373 lines cut ...]
   head_size = struct_val(screw_info, "head_size");
   head_height = struct_val(screw_info, "head_height");
   if (head=="flat") {
     angle = struct_val(screw_info, "head_angle")/2;
     full_height = head_size/2/tan(angle);
     height = is_def(head_height) ? head_height : full_height;
     d2 = head_size*(1-height/full_height);
     //down(height)
     zflip()
       cyl(d1=head_size, d2=d2, l=height, anchor=BOTTOM);
   }
   if (in_list(head,["round","pan","button"])) {
     base = head=="round" || head=="button" ? 0.1 * head_height : .5 * head_height;
     cyl(l=base, d=head_size, anchor=BOTTOM) 
       attach(TOP)
         rotate_extrude() 
           intersection(){
             arc(points=[[-head_size/2,0], [0,-base+head_height * (head=="button"?4/3:1)], [head_size/2,0]]);
             square([head_size, head_height-base]);
             }
   }
   if (head=="socket")
     cyl(l=head_height, d=head_size, anchor=BOTTOM);
}     
```

adrianVmariano, 2019-06-13:
Here's screw code that does metric and uts.  Is there a way to get internal threads?  (It looks like the existing nut code justs uses the same call to threaded_rod, which won't give the correct clearance.)  At some point I want to print some screws and see if they work.....   Also I think some of the old thread pitch tables in metric_bolts.scad are actually wrong (fine ones copies of the coarse one).  

My code does not do interpolation.  If you ask for a screw that wasn't on mcmaster you'll get undef back.  

```
// Things to do
//
// Recess size is confusing.  Does it really depend on the screw size and not just the recess type?
// The code that puts a recess into a flat head seems kind of bogus with regards to recess depth.  
// Do we need hex drive size table for headless screws?

include<BOSL2/std.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>
include <BOSL2/threading.scad>
include <BOSL2/phillips_drive.scad>
include <BOSL2/torx_drive.scad>

/*
http://mdmetric.com/thddata.htm#idx

Seems to show JIS has same nominal thread as others
https://www.nbk1560.com/~/media/Images/en/Product%20Site/en_technical/11_ISO%20General%20Purpose%20Metric%20Screw%20Threads.ashx?la=en

torx flat head
1.6, [3.67, 5
2, [3.8, 6
2.5, [5.5, 8
3, [6.72, 10
4, [8.96, 20
5, [10.4, 25
6, [12.6, 30

*/

/* 
   template:
   
      ["#0", [
      ["#1", [
      ["#2", [
      ["#3", [
      ["#4", [
      ["#5", [
      ["#6", [
      ["#8", [
      ["#10",[
      ["#12",[
      [1/4,  [
      [5/16, [
      [3/8,  [
      [7/16, [
      [1/2,  [
      [9/16, [
      [5/8,  [
      [3/4,  [
      [7/8,  [
      [1,    [
      [1.125,[
      [1.25, [
      [1.375,[
      [1.5,  [
      [1.75, [
      [2,    [
*/

function parse_screw_name(name) = 
    let( commasplit = str_split(name,","),
         length = str_num(commasplit[1]),
         xdash = str_split(commasplit[0], "-x"),
[... 663 lines cut ...]

module driver(spec)
{
  head = struct_val(spec,"head");
  diameter = struct_val(spec,"diameter");
  head_height = is_def(struct_val(spec,"head_height")) ? struct_val(spec,"head_height") :
                head=="socket" ? diameter :
                substr(head,len=4)=="flat" ? .5*struct_val(spec,"head_size") :
                head=="none" ? .7*diameter : undef;
  drive = struct_val(spec,"drive");
  drive_size = struct_val(spec,"drive_size");
  echo(drive = drive);
  echo(head_height = head_height, head=head, substr(head,len=4));
  echo(down=substr("head",len=4)=="flat" ? head_height : 0);
  down(substr(head,len=4)=="flat" || head=="none" ? head_height : 0) {
    if (drive=="philips") down(head_height/6) phillips_drive(size=drive_size, shaft=diameter);
    if (drive=="torx") up(1) torx_drive(size=drive_size, l=head_height+0.1, center=false);
    if (drive=="hex") up(1) linear_extrude(height=head_height) hexagon(id=drive_size);
  }
}

screw("8,.4", head="none", drive="hex", drive_size=2, details=true);
echo(get_screw_info("8-32,1", head="hex",thread=0));

```

revarbat, 2019-07-16:
This can use a refresh, since the strings.scad, structs.scad, and other things have already been integrated.

adrianVmariano, 2019-07-16:
At the moment waiting for a fix on the bug for generating the threaded rod.

revarbat, 2019-07-18:
Here's said bugfixes.  (Not a bug in the threading lib itself.)
```openscad
function thread_profile(thread) =
    let(
        pitch = struct_val(thread,"pitch"),
        basicrad = struct_val(thread,"basic")/2,
        meanpitchrad = mean(struct_val(thread,"d_pitch"))/2,
        meanminorrad = mean(struct_val(thread,"d_minor"))/2,
        meanmajorrad = mean(struct_val(thread,"d_major"))/2,
        depth = (meanmajorrad-meanminorrad)/pitch,
        crestwidth = (pitch/2 - 2*(meanmajorrad-meanpitchrad)/sqrt(3))/pitch
    ) [
        [0, -depth],
        [depth/sqrt(3), 0],
        [depth/sqrt(3)+crestwidth, 0],
        [crestwidth + 2*depth/sqrt(3), -depth],
        [pitch, -depth]
    ];
```
```openscad
module rod(pitch, profile, d, l) {
    trapezoidal_threaded_rod( d=d, l=l, pitch=pitch, profile = profile, bevel=false);
}
```

adrianVmariano, 2019-07-19:
Ok.  Here's where this stands.  I think it mostly works.  Needs some better error handling for when you request something that doesn't exist.  (What happens now is a 20 line error traceback and no insight.)
```
// Things to do
//
// Recess size is confusing.  Does it really depend on the screw size and not just the recess type?
// The code that puts a recess into a flat head seems kind of bogus with regards to recess depth.  
// Do we need hex drive size table for headless screws?

include<BOSL2/std.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>
include<BOSL2/threading.scad>
include<BOSL2/phillips_drive.scad>
include<BOSL2/torx_drive.scad>

/*
http://mdmetric.com/thddata.htm#idx

Seems to show JIS has same nominal thread as others
https://www.nbk1560.com/~/media/Images/en/Product%20Site/en_technical/11_ISO%20General%20Purpose%20Metric%20Screw%20Threads.ashx?la=en

torx flat head
1.6, [3.67, 5
2, [3.8, 6
2.5, [5.5, 8
3, [6.72, 10
4, [8.96, 20
5, [10.4, 25
6, [12.6, 30

*/

function parse_screw_name(name) = 
    let( commasplit = str_split(name,","),
         length = str_num(commasplit[1]),
         xdash = str_split(commasplit[0], "-x"),
         type = xdash[0],
         thread = str_float(xdash[1])
    )
    type[0] == "M" || type[0] == "m" ? ["metric", str_float(substr(type,1)), thread, length] :
    let(
        diam = type[0] == "#" ? type :
               suffix(type,2)=="''" ? str_float(substr(type,0,len(type)-2)) :
               let(val=str_num(type))
               val == floor(val) && val>=0 && val<=12 ? str("#",type) : val
        )
    ["english", diam, thread, 25.4*length]; 

// drive can be "hex", "philips", "slot", "torx", or "none"
// or you can specify "ph0" up to "ph4" for philips and "t20" for torx 20
function parse_drive(drive=undef, drive_size=undef) =
    is_undef(drive) ? ["none",undef] :
    let(drive = downcase(drive))
    in_list(drive,["hex","philips", "slot", "torx", "philips", "none"]) ? [drive, drive_size] :
    drive[0]=="t" ? ["torx", str_int(substr(drive,1))] :
    substr(drive,0,2)=="ph" ? ["philips", str_int(substr(drive,2))] :
    assert(str("Unknown screw drive type ",drive));

// Function: get_screw_info()
// Usage:
//   info = get_screw_info(name, [head], [thread], [drive], [drive_size], [oversize])
// Description:
//   Look up screw characteristics for the specified screw type.
//
//   For metric (ISO) `name` is M<size>x<pitch>,<length>, e.g. `"M6x1,10"` specifies a 6mm diameter screw with a thread pitch of 1mm and length of 10mm.
//   You can omit the pitch or length, e.g. `"M6x1"`, or `"M6,10"`, or just `"M6"`.  
//  
//   For English (UTS) name is <size>-<threadcount>,<length>, e.g. `"#8-32,1/2"`, or `"1/4-20,1"`.  Units are in inches, including the length.
//   Size can be a number from 0 to 12 with or without a leading '#' to specify a screw gauge size, or any other value to specify
//   a diameter in inches, either as a float or a fraction, so `"0.5-13"` and `"1/2-13"` are equivalent.  To force interpretation of the value
[... 920 lines cut ...]
 screw("#6-32", head="flat small",length=12,drive="torx");
 screw("#6-32,1/2", head="flat large",length=12);
 screw("#6-32", head="flat undercut",length=12);
 screw("#6-32", head="socket",length=12);
 screw("#6-32", drive="hex", drive_size=1.5, length=12);
}

fwd(10)xdistribute(spacing=8){
 screw("M3", head="flat small",length=12);
 screw("M3", head="button",length=12);
 screw("M3", head="pan", drive="philips",length=12);
 screw("M3", head="pan", drive="slot",length=12);
 screw("M3", head="flat large",length=12);
 screw("M3", head="flat", drive="hex",length=12);
 screw("M3", head="socket",length=12);
 screw("M5", head="hex", length=12);
}
 

/*
screw("1/4-20,1/2", head="hex",orient=UP,anchor="base");
right(15) nut("1/4-20", thickness=8, diameter=1/2*inch);
*/

```

adrianVmariano, 2019-07-24:
The phillips recess code is pretty wrong.  I have access to recess diameter, the width of the cross, and the depth of the recess.  We need a working recess maker.  

anchor=CENTER is broken.  No clue why.  

No slot for flat heads yet.  

```
// Things to do
//
// Recess size is confusing.  Does it really depend on the screw size and not just the recess type?
// The code that puts a recess into a flat head seems kind of bogus with regards to recess depth.  
// Do we need hex drive size table for headless screws?

include<BOSL2/std.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>
include<BOSL2/threading.scad>
include<BOSL2/phillips_drive.scad>
include<BOSL2/torx_drive.scad>

/*
http://mdmetric.com/thddata.htm#idx

Seems to show JIS has same nominal thread as others
https://www.nbk1560.com/~/media/Images/en/Product%20Site/en_technical/11_ISO%20General%20Purpose%20Metric%20Screw%20Threads.ashx?la=en

torx flat head
1.6, [3.67, 5
2, [3.8, 6
2.5, [5.5, 8
3, [6.72, 10
4, [8.96, 20
5, [10.4, 25
6, [12.6, 30

*/

function parse_screw_name(name) = 
    let( commasplit = str_split(name,","),
         length = str_num(commasplit[1]),
         xdash = str_split(commasplit[0], "-x"),
         type = xdash[0],
         thread = str_float(xdash[1])
    )
    type[0] == "M" || type[0] == "m" ? ["metric", str_float(substr(type,1)), thread, length] :
    let(
        diam = type[0] == "#" ? type :
               suffix(type,2)=="''" ? str_float(substr(type,0,len(type)-2)) :
               let(val=str_num(type))
               val == floor(val) && val>=0 && val<=12 ? str("#",type) : val
        )
    ["english", diam, thread, 25.4*length]; 

// drive can be "hex", "philips", "slot", "torx", or "none"
// or you can specify "ph0" up to "ph4" for philips and "t20" for torx 20
function parse_drive(drive=undef, drive_size=undef) =
    is_undef(drive) ? ["none",undef] :
    let(drive = downcase(drive))
    in_list(drive,["hex","philips", "slot", "torx", "philips", "none"]) ? [drive, drive_size] :
    drive[0]=="t" ? ["torx", str_int(substr(drive,1))] :
    substr(drive,0,2)=="ph" ? ["philips", str_int(substr(drive,2))] :
    assert(str("Unknown screw drive type ",drive));

// Function: screw_info()
// Usage:
//   info = screw_info(name, [head], [thread], [drive], [drive_size], [oversize])
// Description:
//   Look up screw characteristics for the specified screw type.
//
//   For metric (ISO) `name` is M<size>x<pitch>,<length>, e.g. `"M6x1,10"` specifies a 6mm diameter screw with a thread pitch of 1mm and length of 10mm.
[... 1026 lines cut ...]
    //drive="hex", drive_size=3,
       drive="hex",
     head="button",orient=UP,anchor_head=CENTER,tolerance="3A");

/*
xdistribute(spacing=15){
  nut("1/4-20", thickness=8, diameter=0.5*inch,tolerance="3B");
  nut("M6x1", thickness=8, diameter=0.4*inch,tolerance="5H");
  nut("M6x1", thickness=8, diameter=0.4*inch);
  nut("M6x1", thickness=8, diameter=0.4*inch,tolerance="4H");
}
*/

// recess sizing:
// http://www.fasnetdirect.com/refguide/Machinepancombo.pdf
//
/*   ASME B 18.6.3
http://www.smithfast.com/newproducts/screws/msflathead/

/* philips recess diagram

http://files.engineering.com/getfile.aspx?folder=76fb0d5e-1fff-4c49-87a5-05979477ca88&file=Noname.jpg&__hstc=212727627.6c577ef84c12d9cc69c819eea7be49d2.1563972499721.1563972499721.1563972499721.1&__hssc=212727627.1.1563972499721&__hsfp=165344926

*/
```

adrianVmariano, 2019-07-28:
Ok.  Here's an update.  Perhaps my last as I'm really tired of this stuff.   Limitations:  we need a real phillips driver module that takes diameter, width and depth.  The metric cheese screws don't have phillips width because I could only find their specs on ISO docs and ISO doesn't give that parameter.  (Is it derived somehow?).  Oval screws are missing, as are a few other more exotic shapes.  Someone else can add those....
```
include<BOSL2/std.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>
include<BOSL2/threading.scad>
include<BOSL2/phillips_drive.scad>
include<BOSL2/torx_drive.scad>

/*
http://mdmetric.com/thddata.htm#idx

Seems to show JIS has same nominal thread as others
https://www.nbk1560.com/~/media/Images/en/Product%20Site/en_technical/11_ISO%20General%20Purpose%20Metric%20Screw%20Threads.ashx?la=en

Various ISO standards here:  https://www.fasteners.eu/standards/ISO/4026/

Torx values:  https://www.stanleyengineeredfastening.com/-/media/web/sef/resources/docs/other/socket_screw_tech_manual_1.ashx

*/

function parse_screw_name(name) = 
    let( commasplit = str_split(name,","),
         length = str_num(commasplit[1]),
         xdash = str_split(commasplit[0], "-x"),
         type = xdash[0],
         thread = str_float(xdash[1])
    )
    type[0] == "M" || type[0] == "m" ? ["metric", str_float(substr(type,1)), thread, length] :
    let(
        diam = type[0] == "#" ? type :
               suffix(type,2)=="''" ? str_float(substr(type,0,len(type)-2)) :
               let(val=str_num(type))
               val == floor(val) && val>=0 && val<=12 ? str("#",type) : val
        )
    ["english", diam, thread, 25.4*length]; 

// drive can be "hex", "phillips", "slot", "torx", or "none"
// or you can specify "ph0" up to "ph4" for phillips and "t20" for torx 20
function parse_drive(drive=undef, drive_size=undef) =
    is_undef(drive) ? ["none",undef] :
    let(drive = downcase(drive))
    in_list(drive,["hex","phillips", "slot", "torx", "phillips", "none"]) ? [drive, drive_size] :
    drive[0]=="t" ? ["torx", str_int(substr(drive,1))] :
    substr(drive,0,2)=="ph" ? ["phillips", str_int(substr(drive,2))] :
    assert(str("Unknown screw drive type ",drive));

// Function: screw_info()
// Usage:
//   info = screw_info(name, [head], [thread], [drive], [drive_size], [oversize])
// Description:
//   Look up screw characteristics for the specified screw type.
//
//   For metric (ISO) `name` is M<size>x<pitch>,<length>, e.g. `"M6x1,10"` specifies a 6mm diameter screw with a thread pitch of 1mm and length of 10mm.
//   You can omit the pitch or length, e.g. `"M6x1"`, or `"M6,10"`, or just `"M6"`.  
//  
//   For English (UTS) name is <size>-<threadcount>,<length>, e.g. `"#8-32,1/2"`, or `"1/4-20,1"`.  Units are in inches, including the length.
//   Size can be a number from 0 to 12 with or without a leading '#' to specify a screw gauge size, or any other value to specify
//   a diameter in inches, either as a float or a fraction, so `"0.5-13"` and `"1/2-13"` are equivalent.  To force interpretation of the value
//   as inches add `''` to the end, e.g. `"1''-4"` is a one inch screw and `"1-80"` is a very small 1-gauge screw.  The pitch is specified
//   using a thread count, the number of threads per inch.  The length is in inches.  
//
//   If you omit the pitch then a standard screw pitch will be supplied from lookup tables for the screw diameter you have chosen.
//   For each screw diameter, multiple standard pitches are possible.  
//   For the UTS system these the availble thread types are:
//   - "coarse" or "UNC"
//   - "fine" or "UNF"
//   - "extra fine", "extrafine" or "UNEF".
//   The ISO system defines a coarse threading and three different fine threadings but does not give different names to the fine threadings.
//   The thread options for ISO are:
[... 1293 lines cut ...]
//
/*   ASME B 18.6.3
http://www.smithfast.com/newproducts/screws/msflathead/

/* phillips recess diagram

http://files.engineering.com/getfile.aspx?folder=76fb0d5e-1fff-4c49-87a5-05979477ca88&file=Noname.jpg&__hstc=212727627.6c577ef84c12d9cc69c819eea7be49d2.1563972499721.1563972499721.1563972499721.1&__hssc=212727627.1.1563972499721&__hsfp=165344926

*/

// To do list
//
// Metric hex engagement:
// https://www.bayoucitybolt.com/socket-head-cap-screws-metric.html
//
// Torx drive depth for UTS and ISO
// Handle generic phillips (e.g. ph2) or remove it?
//
// How do you insert a threaded hole into a model?
// Default nut thickness
//

//phillips_drive(size=3, shaft=.25*inch,anchor=BOTTOM);

```

adrianVmariano, 2019-08-04:
Here's a tweaked version where anchoring is (I think) working correctly.  

```
include<BOSL2/std.scad>
include<BOSL2/strings.scad>
include<BOSL2/structs.scad>
include<BOSL2/threading.scad>
include<BOSL2/phillips_drive.scad>
include<BOSL2/torx_drive.scad>

/*
http://mdmetric.com/thddata.htm#idx

Seems to show JIS has same nominal thread as others
https://www.nbk1560.com/~/media/Images/en/Product%20Site/en_technical/11_ISO%20General%20Purpose%20Metric%20Screw%20Threads.ashx?la=en

Various ISO standards here:  https://www.fasteners.eu/standards/ISO/4026/

Torx values:  https://www.stanleyengineeredfastening.com/-/media/web/sef/resources/docs/other/socket_screw_tech_manual_1.ashx

*/

function parse_screw_name(name) = 
    let( commasplit = str_split(name,","),
         length = str_num(commasplit[1]),
         xdash = str_split(commasplit[0], "-x"),
         type = xdash[0],
         thread = str_float(xdash[1])
    )
    type[0] == "M" || type[0] == "m" ? ["metric", str_float(substr(type,1)), thread, length] :
    let(
        diam = type[0] == "#" ? type :
               suffix(type,2)=="''" ? str_float(substr(type,0,len(type)-2)) :
               let(val=str_num(type))
               val == floor(val) && val>=0 && val<=12 ? str("#",type) : val
        )
    ["english", diam, thread, 25.4*length]; 

// drive can be "hex", "phillips", "slot", "torx", or "none"
// or you can specify "ph0" up to "ph4" for phillips and "t20" for torx 20
function parse_drive(drive=undef, drive_size=undef) =
    is_undef(drive) ? ["none",undef] :
    let(drive = downcase(drive))
    in_list(drive,["hex","phillips", "slot", "torx", "phillips", "none"]) ? [drive, drive_size] :
    drive[0]=="t" ? ["torx", str_int(substr(drive,1))] :
    substr(drive,0,2)=="ph" ? ["phillips", str_int(substr(drive,2))] :
    assert(str("Unknown screw drive type ",drive));

// Function: screw_info()
// Usage:
//   info = screw_info(name, [head], [thread], [drive], [drive_size], [oversize])
// Description:
//   Look up screw characteristics for the specified screw type.
//
//   For metric (ISO) `name` is M<size>x<pitch>,<length>, e.g. `"M6x1,10"` specifies a 6mm diameter screw with a thread pitch of 1mm and length of 10mm.
//   You can omit the pitch or length, e.g. `"M6x1"`, or `"M6,10"`, or just `"M6"`.  
//  
//   For English (UTS) name is <size>-<threadcount>,<length>, e.g. `"#8-32,1/2"`, or `"1/4-20,1"`.  Units are in inches, including the length.
//   Size can be a number from 0 to 12 with or without a leading '#' to specify a screw gauge size, or any other value to specify
//   a diameter in inches, either as a float or a fraction, so `"0.5-13"` and `"1/2-13"` are equivalent.  To force interpretation of the value
//   as inches add `''` to the end, e.g. `"1''-4"` is a one inch screw and `"1-80"` is a very small 1-gauge screw.  The pitch is specified
//   using a thread count, the number of threads per inch.  The length is in inches.  
//
//   If you omit the pitch then a standard screw pitch will be supplied from lookup tables for the screw diameter you have chosen.
//   For each screw diameter, multiple standard pitches are possible.  
//   For the UTS system these the availble thread types are:
//   - "coarse" or "UNC"
//   - "fine" or "UNF"
//   - "extra fine", "extrafine" or "UNEF".
//   The ISO system defines a coarse threading and three different fine threadings but does not give different names to the fine threadings.
[... 1296 lines cut ...]
// http://www.fasnetdirect.com/refguide/Machinepancombo.pdf
//
/*   ASME B 18.6.3
http://www.smithfast.com/newproducts/screws/msflathead/

/* phillips recess diagram

http://files.engineering.com/getfile.aspx?folder=76fb0d5e-1fff-4c49-87a5-05979477ca88&file=Noname.jpg&__hstc=212727627.6c577ef84c12d9cc69c819eea7be49d2.1563972499721.1563972499721.1563972499721.1&__hssc=212727627.1.1563972499721&__hsfp=165344926

*/

// To do list
//
// Metric hex engagement:
// https://www.bayoucitybolt.com/socket-head-cap-screws-metric.html
//
// Torx drive depth for UTS and ISO
// Handle generic phillips (e.g. ph2) or remove it?
//
// How do you insert a threaded hole into a model?
// Default nut thickness
//

//phillips_drive(size=3, shaft=.25*inch,anchor=BOTTOM);
```

revarbat, 2019-08-11:
Dependency on #92

revarbat, 2020-01-27:
I've committed new Phillips driver code and updated #92.

adrianVmariano, 2020-05-26:
Outstanding issues with the new screw code 
- [ ] Need default nut thickness 
- [ ] Torx engagement depth is missing for UTS and ISO (at least "flat small") which means you can't select torx.  At the moment the hex depth is used instead as a workaround
- [ ] Phillips drive code does not make use of the phillips drive dimensions of width and slot width
- [ ] Docs claim you can use ph1, ph2, etc.  This doesn't actually work.  Supporting this requires a standard phillips width and slot width, but in reality even screws that take the same phillips size have different sized recesses, so it's not obvious that this notion makes sense.   Either eliminate this as a supposed option or figure out how to implement it and add it.
- [ ] Need a way to insert a threaded hole.  Perhaps a threaded shaft with inverse nut-threads?  (e.g. so that when you subtract it you get the proper nut threading with proper nut threading tolerances and offsets.)

revarbat, 2020-05-29:
Quoth @adrianVmariano from Gitter, regarding Robertson/Square drive:
> It looks like the driver depth is directly determined by driver width.
> Gauge penetration is less than recess depth, I guess because of the sloped sides of the recess?
> What this is telling me is that the right way to implement it is to add columns for the square drive  size as a number (same way torx is done) rather than actually having length data in the tables.
> Then the recess making code can determine dimensions just from the size number.
> So for every head type that comes in square drive the drive size needs to be looked up....
> Now I suppose there's a possibility of using the diameter rules they give in that table to go from head diameter to recess size. That would avoid the need for tables.
> 
> So the way I'm seeing it, the recess doesn't vary its dimensions with different head sizes the way slot and phillips do. So we don't need to code dimensions, just the numerical size (like torx). And it appears that unlike hex (and torx?) the recess depth is fixed, not head size dependent. So we can get everything from the driver numerical size.
> Then there are two possibilities for getting the driver size: McMaster (or other sources perhaps) will specify the driver you need for each fastener they sell, so for each type of head you can just look it up. Or we could believe the doc above which gives size ranges of head diameter. The problem is that those size ranges might be specific to flathead and not apply to other shaped heads.

revarbat, 2020-05-29:
https://www.aspenfasteners.com/v/vspfiles/files/docs/square_drive_specification.pdf

revarbat, 2020-05-30:
https://www.kanebridge.com/intranet/refguide/machine.pdf

revarbat, 2020-05-30:
- [ ] Bolt shoulders.  Maybe as excess shank diameter?
- [ ] Wood screw threads?
- [ ] Sheet metal threads and/or digging tip/

revarbat, 2020-05-31:
Shoulders expect an extra recessed shank, so no, this cannot be implemented with excess shank diameter.

closed 2022-10-14 by adrianVmariano