BelfrySCAD/BOSL2 #41: more rounding flexibility for prismoid

adrianVmariano, 2019-06-11:
I needed to round some but not all of the edges of a prismoid...but also some top edges.  I ended up having to intersection with a cuboid to round the top and then union with an extra prismoid to unround the edges I didn't want rounded.  More flexibility (as with cuboid) would be nice.

revarbat, 2019-06-18:
Making `prismoid()` handle rounding and chamfering is doable, but rather complex.

adrianVmariano, 2019-08-12:
Doable but complex is why you put it into the library....

revarbat, 2020-05-23:
I was able to implement `prismoid()` rounding and chamfering for individual vertical(ish) edges, but I haven't yet worked out a way to do top or bottom edge tweaking in a reasonable way.  f3c69eb673f48c603992ff4aa4238555698331c5

revarbat, 2020-05-23:
Most of my problems here stem from how to make corners that allow various sized chamfers, round-overs, and unrounded edges to meet.

adrianVmariano, 2020-05-23:
Are these problems addressed at all by rounded_prism?

revarbat, 2020-05-24:
Maybe.  I don't see why it couldn't do that.

revarbat, 2021-02-02:
This is more complicated than I'm inclined to implement for the initial BOSL2 release.

revarbat, 2021-06-24:
Implementing this with `rounded_prism()` will need rounding.scad to be promoted into std.scad.

adrianVmariano, 2022-11-10:
Does the existence of rounded_prism() mean that we just don't need more rounding for prismoid?  Users should just use rounded_prism instead?

revarbat, 2023-03-26:
>Does the existence of rounded_prism() mean that we just don't need more rounding for prismoid? Users should just use rounded_prism instead?

I am inclined to agree.

adrianVmariano, 2023-03-26:
One consideration before closing this:  if a user wants to round *some* of the TOP edges of a prismoid, that is impossible by any means right now.  (Other than hand positioning rounding masks, I guess.)  A limitation of rounded_prism is that all top edges must have the same rounding.

revarbat, 2023-09-26:
I think that `edge_profile()`, `edge_profile_asym()`, and `corner_profile()` handle most of the missing rounding options.

adrianVmariano, 2023-09-26:
Ok, so how do I round the individual top edges of this prismoid?  

```
prismoid([50,30], [30,20], shift=[3,6], h=15);
```

revarbat, 2023-11-06:
```openscad
diff()
prismoid([50,30], [30,20], shift=[3,6], h=15, rounding=[5,0,5,0]) {
    edge_profile([TOP+RIGHT, TOP+BACK], excess=10, convexity=20) {
        mask2d_roundover(r=5,mask_angle=$edge_angle);
    }
}
```
[image]

revarbat, 2023-11-06:
I think it'd be good to have an example added to `prismoid()` for this.

revarbat, 2023-11-06:
I'll add that to my catenary()/version tweaking PR.

adrianVmariano, 2023-11-06:
I didn't know about $edge_angle.

Is there a way to round a corner?

revarbat, 2023-11-07:
At this time, rounding off corners of a non-cubic prismoid isn't doable via attachments.  I'm not clear how to do it, actually, as the $edge_angle would be continuously transitioning between two edges, much less three edges.

revarbat, 2023-11-07:
This has also made it clear to me that the round-over profile masks could use a cut= option in addition to it's r= argument.

adrianVmariano, 2023-11-07:
I really like the cut= method for specifying roundovers.   For producing roundings that match at the corner you actually would need the joint= argument.  

It seems to me like with a reference to rounded_prism() and to the use of attachable masks with $edge_angle that we can close this issue.  

Is $edge_angle new?  It's not documented at all in prismoid().  Or it's not a prismoid feature but an edge_profile() feature?  Seems like not only should there be an example but actually the paragraph that discusses rounding should also mention the option of using edge_profile with $edge_angle for rounding top edges.

adrianVmariano, 2023-11-07:
Should edge_profile() be listed under "see also"?  

Also does the above example still work with degenerate prismoids?

revarbat, 2023-11-07:
Okay, I've implemented `cut=` and `joint=` for `mask2d_roundover()` and `mask2d_teardrop()` based on the definitions in rounding.scad.  The main thing that this points out to me is that what is actually needed is a consistent mask height, so I've implemented `h=` for those masks as well.  And since the varying angle of the inset top on the mask makes for messy `edge_profile()` masking, I've also implemented a `flat_top=` boolean argument that forces the top inset to be horizontal for `mask2d_roundover()`.  And I made that the default.

The problem:
[image]

Fixed, using h= and flat_top=true:
[image]

adrianVmariano, 2023-11-07:
You will need the joint= arg if you want three of them meeting at a corner.  Meeting at a corner is a notable case because it's a case that cannot be done by rounded_prism().  That is, a prismoid with only 3 roundings that meet at a corner.    But it seems like joint= should also work for the above case.  Does it fail somehow?  

Regarding degenerate prismoids it seems like in principle you could round the top edge (degenerate edge) using the TOP anchor, and $edge_angle could be set appropriately for that case---as long as shift is not set.  With nonzero shift it's a mess.  You'd have to work around the fact that the TOP anchor isn't symetric wrt the edge, which seems like a kludge.

closed 2024-11-11 by adrianVmariano