BelfrySCAD/BOSL2 #55: rot should work in planar mode when given 2d input

adrianVmariano, 2019-06-27:
I did rot(45,[0,1]) and got back a 3-vector.  I had to explicitly give planar=true to get a 2-vector out.  This is wierd and unexpected.  If the input is 2d the output should be 2d.

revarbat, 2019-06-27:
Hmm.  As I recall, the `planar` argument existed for the case of `foo = rot(45);` where it had no input point to judge 2D-ness from, and had to decide whether to return an affine2d or affine3d matrix.  Yes, if given a vector or path as input, it should be able to figure out on its own without the `planar` argument.

adrianVmariano, 2019-06-27:
It doesn't.  

    echo(a=rot(45,p=[1,0]));
    ECHO: a = [0.707107, 0.707107, 0]

adrianVmariano, 2019-06-27:
And yes, you need the planar argument for that other case with no point input.

revarbat, 2019-06-27:
If given 2D input, but a 3D rotation, should it return a 3D value?  I would think so.

adrianVmariano, 2019-06-27:
So something like rot([20,45,33],[1,1])?

Seems like it should either return 3d or an error.

revarbat, 2019-07-11:
Fixed `rot()` to return 2D when passed 2D points in `p` and rotations are only around the Z axis.
If rotations aren't around the Z axis, 3D points will be returned.

closed 2019-07-11 by revarbat