BelfrySCAD/BOSL2 #56: replace (?) triangle_area2d with polygon_area

adrianVmariano, 2019-06-27:
```
function polygon_area(vertices) =
     0.5*sum([for(i=[0:len(vertices)-1]) det2(select(vertices,i,i+1))]);
```
Note if polygon self intersects then some areas have positive area, some have negative, and they are summed.  

This function is slower than triangle_area2d, so maybe it's worth keeping both?  (This takes about twice as long.)  I'm not sure how much of a difference it makes in the run time of hull() for example.  It took 9s to do 100k computations with polygon_area and 4s with triangle_area2d.   I don't know how many times hull() calls triangle_area2d().  

I do think that triangle_area2d should be renamed triangle_area if it is retained, unless you plan to add triangle_area3d.  

    function triangle_area3d(x1,x2,x3) = 0.5*norm(cross(x3-x1,x3-x2));

I would prefer to have one triangle_area function that checks the dimension.  The 3d version is not signed.

revarbat, 2019-06-27:
Since `triangle_are2d()` doesn't work for 3D triangles, it seemed a good idea to keep the `2d` to emphasize that.  Adding a `triangle_area3d()` seems like a good idea, even if it cannot be used for clockwise checks. I'd like to keep `triangle_area2d()` for speed reasons when we know we are dealing with a simple triangle, and add a `polygon_area()` for the general case.

adrianVmariano, 2019-06-27:
You could consider merging the two triangle commands into a single triangle_area that checks the point dimension and does the 2d or 3d computation as appropriate.  This doesn't incur a speed penalty.  

In the 3d case we are computing a normal to the triangle, but there's no point of reference with which to define clockwise.

revarbat, 2019-07-10:
Implemented in 9f0d55f5b38685bdcaf4e292c34ca6ff9bc02bc6

closed 2019-07-10 by revarbat