BelfrySCAD/BOSL2 #57: right_of_line2d vs point_left_of_segment

adrianVmariano, 2019-06-27:
These two functions basically do the same thing, except that point_left_of_segment detects the equality case.  It seems like we don't need both.   point_left_of_segment could be renamed point_left_of_line, maybe?  Or maybe something else that conveys the idea of the point's relationship to a line?  (Nothing leaps to mind.)  In any case, the docs also need to be beefed up to define what "left" means---in other words, that direction is defined by the order of the points that specify the line.  

Note that right_of_line2d is never used.

revarbat, 2019-06-27:
I honestly don't recall why I added `right_of_line2d()`.

revarbat, 2019-07-10:
Removed `right_of_line2d()` in e00ce0d81f3671615c0d22a8c917f1133a9340f3

closed 2019-07-10 by revarbat