BelfrySCAD/BOSL2 #58: Add puzzlecut (or something like it)?

adrianVmariano, 2019-06-27:
Consider adding something like this:

https://www.thingiverse.com/thing:35834

revarbat, 2019-06-27:
Seems useful, if for no other reason than to allow breaking up large prints into assemble-able pieces.

adrianVmariano, 2019-06-27:
Yeah, that was the point, I thought.  There's a peg version also, but that seems like it would be annoying to print due to supports.  

https://www.thingiverse.com/thing:310652

Perhaps it's a hair less obtrusive (no puzzle shape in the model)?   OF course, the locking nature of puzzle cut is nice.  

https://github.com/jernst/josl/tree/master/josl/cuts

Does some other cut lines, including zig-zag and comb.

revarbat, 2019-07-10:
Implemented in e2e6bcf2f9a7c4fd39fcd05918770e62cab1f617

closed 2019-07-10 by revarbat