BelfrySCAD/BOSL2 #59: universal joint?

adrianVmariano, 2019-06-27:
https://www.thingiverse.com/thing:125939

Consider adding something like this.

revarbat, 2019-06-27:
Hmm.  Not sure if this is really needed.

```openscad

revarbat, 2019-06-27:
```openscad
$slop=0.2;
module ujoint(r, d) {
    r = get_radius(r=r, d=d, dflt=50);
    rot_copies([[0,0,0], [180,0,90]]) {
        difference() {
            intersection() {
                union() {
                    zcyl(r=r, l=2*r, anchor=TOP);
                    ycyl(l=2*r, r=r/2);
                }
                zcyl(r=r, l=10*r);
            }
            ycyl(r=r/3, l=3*r);
            zcyl(r=r*0.75, l=r*2);
            xspread(2*r) cube([r,r*3,r*2], center=true);
        }
    }
    sphere(r=r*0.75);
    zrot_copies([0,90]) {
        xcyl(r=r/3-3*$slop, l=r*2);
    }
}
```
[image]

revarbat, 2019-06-27:
How often will this come up though?

adrianVmariano, 2019-06-27:
Don't know.  Could be interesting to make some mechanism with a bunch of these.  Assuming that they print well.  But I'm OK either way with keeping it or not.

revarbat, 2019-07-17:
```openscad

revarbat, 2019-08-09:
Eh.  I don't see this as being a library item.

closed 2019-08-09 by revarbat