BelfrySCAD/BOSL2 #60: add snap lock hinges

adrianVmariano, 2019-06-27:
https://www.myminifactory.com/object/3d-print-small-tetrahedron-folding-polyhedra-81860

I printed out the above model and it's neat how it just snaps together.  (Worked great right off the printer.)  It would be nice to have hinges like that available (at arbitrary angle).

revarbat, 2019-06-28:
If I'm interpreting this correctly, I see three modules needed.  One for making the folding hinge mask.  One for making the single snap-lock.  One for making the dual snap-socket.

revarbat, 2019-06-28:
```openscad
module folding_hinge_mask(l, thick, layerheight=0.2, fold_ang=90, anchor=CENTER, spin=0, orient=UP) {
    orient_and_anchor(size=[l, 0.01, thick], size2=[l,2*thick*tan(fold_ang/2)], anchor=anchor, spin=spin, orient=orient, chain=true) {
        up(layerheight*2) prismoid([l,0], [l, 2*thick*tan(fold_ang/2)], h=thick, anchor=BOT);
        children();
    }
}
```

revarbat, 2019-06-28:
```openscad

adrianVmariano, 2019-06-28:
I think you probably want a module that puts the snap lock on the hinge (with the dual on one side and single on the other, and slanted at the correct angle).

revarbat, 2019-06-28:
```openscad
module folding_hinge_mask(l, thick, layerheight=0.2, fold_ang=90, anchor=CENTER, spin=0, orient=UP) {
	orient_and_anchor(size=[l, 0.01, 2*thick], size2=[l,2*thick*tan(fold_ang/2)], anchor=anchor, spin=spin, orient=orient, chain=true) {
		up(layerheight*2) prismoid([l,0], [l, 2*thick/tan(fold_ang/2)], h=thick, anchor=BOT);
		children();
	}
}

module snap_lock(thick, snaplen=10, snapdiam=10, layerheight=0.2, fold_ang=90, anchor=CENTER, spin=0, orient=UP) {
	snap_x = (snapdiam/2) / tan(fold_ang/2) + (thick-2*layerheight)/tan(fold_ang/2);
	orient_and_anchor(size=[snaplen, snapdiam, 2*thick], anchor=anchor, spin=spin, orient=orient, chain=true) {
		back(snap_x) {
			cube([snaplen, snapdiam, snapdiam/2+thick], anchor=BOT) {
				attach(TOP) xcyl(l=snaplen, d=snapdiam);
				attach(TOP) xspread(snaplen-snapdiam/4/3) xscale(0.333) sphere(d=snapdiam-2);
			}
		}
		children();
	}
}

module snap_socket(thick, snaplen=10, snapdiam=10, layerheight=0.2, fold_ang=90, anchor=CENTER, spin=0, orient=UP) {
	snap_x = (snapdiam/2) / tan(fold_ang/2) + (thick-2*layerheight)/tan(fold_ang/2);
	orient_and_anchor(size=[snaplen, snapdiam, 2*thick], anchor=anchor, spin=spin, orient=orient, chain=true) {
		fwd(snap_x) {
			zrot_copies([0,180], r=snaplen+$slop) {
				diff("divot")
				cube([snaplen, snapdiam, snapdiam/2+thick], anchor=BOT) {
					attach(TOP) xcyl(l=snaplen, d=snapdiam);
					attach(TOP) left((snaplen+snapdiam/4/3)/2) xscale(0.333) sphere(d=snapdiam-2, $tags="divot");
				}
			}
		}
		children();
	}
}
```

revarbat, 2019-06-28:
[image]

revarbat, 2019-06-28:
```openscad
module folding_hinge_with_snaps(l, thick, snaplen=10, snapdiam=10, layerheight=0.2, fold_ang=90, anchor=CENTER, spin=0, orient=UP) {
	snap_x = (snapdiam/2) / tan(fold_ang/2) + (thick-2*layerheight)/tan(fold_ang/2);
	orient_and_anchor(size=[l+1, snapdiam+2*snap_x, 2*thick], anchor=anchor, spin=spin, orient=orient) {
		union() {
			difference() {
				children();
				folding_hinge_mask(l=l+1, thick=thick, fold_ang=fold_ang, layerheight=layerheight, $tags="hinge");
			}
			snap_lock(thick=thick, fold_ang=fold_ang, snaplen=snaplen, snapdiam=snapdiam, layerheight=layerheight, $tags="lock");
			snap_socket(thick=thick, fold_ang=fold_ang, snaplen=snaplen, snapdiam=snapdiam, layerheight=layerheight, $tags="sock");
		}
	}
}
```

revarbat, 2019-07-12:
Implemented in 8f20fa6263d976439bfb4e0bf967e1387499514e

closed 2019-07-12 by revarbat