BelfrySCAD/BOSL2 #62: Modules that don't use the children issue a warning or error?

adrianVmariano, 2019-06-29:
Should modules that don't use their children fail when children are passed?  

I just spent a baffled several minutes before realizing that I had left out a semicolon, so my code was being passed to a module that didn't use it.

revarbat, 2019-07-18:
It's doable to add a bunch of `assert($children==0);`s in various modules.

adrianVmariano, 2019-07-18:
Yeah, simple enough to do.   Just kind of tedious.

adrianVmariano, 2019-08-11:
Are there any modules that don't use all their children?  Like they only use the first one or something?

I was wondering about making a module like `no_children` to try to streamline this process, though it's not clear how much it really helps if you have to do `no_children($children)`.   I guess it avoids having to repeat the text of the error message.  

But I noticed that there is now a warning message from regular openscad modules that don't take children, so it seems like to keep everything consistent, it would be best to use the same mesage text, namely:

    assert($children==0,"Module modulename() does not support child modules")

I suppose there's no way to access the current module name so that this could be done generically, like `str("Module ",$modulename,"() does not support child modules")`

adrianVmariano, 2019-08-11:
```
module no_children(count)
{
  assert(count==0, str("Module ",parent_module(1),"() does not support child modules"));
}    
```

Call with `no_children($children)`

adrianVmariano, 2020-08-27:
- [ ] beziers.scad: bezier_polygon
- [ ] beziers.scad: bezier_sweep_bezier
- [ ] beziers.scad: trace_bezier
- [ ] beziers.scad: bezier_polyhedron
- [ ] common.scad: assert_approx
- [ ] common.scad: assert_equal
- [ ] debug.scad: trace_polyline
- [ ] debug.scad: debug_polygon
- [ ] debug.scad: debug_polyhedron
- [ ] hull.scad: hull_points
- [ ] involute_gears.scad: gear_tooth_profile
- [ ] masks.scad: rounding_cylinder_mask
- [ ] masks.scad: teardrop_corner_mask
- [ ] partitions.scad: partition_mask
- [ ] partitions.scad: partition_cut_mask
- [ ] paths.scad: modulated_circle
- [ ] regions.scad: region
- [ ] screws.scad: screw_head
- [ ] screws.scad: nut
- [ ] shapes2d.scad: stroke
- [ ] shapes2d.scad: arc
- [ ] shapes.scad: nil
- [ ] shapes.scad: slot
- [ ] shapes.scad: heightfield
- [ ] strings.scad: echofmt
- [ ] structs.scad: struct_echo
- [ ] torx_drive.scad: torx_drive2d
- [ ] version.scad: bosl_required
- [ ] vnf.scad: vnf_polyhedron
- [ ] vnf.scad: vnf_validate
- [ ] wiring.scad: wiring

I used grep to try to find the affected modules and have listed them above.  Some of these should probably call attachable and take children.  Others should have no_children($children) added.  No guarantee that my process for producing the list was correct or complete, but hopefully it's close.

revarbat, 2020-08-27:
> I suppose there's no way to access the current module name so that this could be done generically, like str("Module ",$modulename,"() does not support child modules")

Well, there *are* the builtins `parent_module(idx)` and `$parent_modules`:

> $parent_modules contains the number of modules in the instantiation stack. parent_module(i) returns the name of the module i levels above the current module in the instantiation stack. The stack is independent of where the modules are defined. It's where they're instantiated that counts. This can, for example, be used to build a BOM (Bill Of Material).

adrianVmariano, 2020-08-27:
I already did it generically with no_children(), which already exists in common.scad, and indeed uses parent_module(1).  Note also the code example above.   Now we just need to add the no_children($children) invocation everywhere it's needed.

adrianVmariano, 2020-08-27:
I added some of the checks and attachable() to vnf_polyhedron.  Probably hull_points should use attachable as well, but that one is complicated, because it's 2d and 3d.  Most of the other ones I'm not sure what they should do.

closed 2022-04-12 by adrianVmariano