BelfrySCAD/BOSL2 #66: Should move work on 1d data?

adrianVmariano, 2019-07-01:
Did I overlook some other way to add a constant to a vector?  

I tried to use move on 1d data, e.g. move(3,[1,2,3]) and it fails.

revarbat, 2019-07-02:
Do we need a `VADD()` function that assumes 0 fill for vectors of unequal length, and give it a special case for scalar rags?

adrianVmariano, 2019-07-02:
I'm not sure.  I feel like add two vectors of unequal length is a bad operation.  (IDL is a language that does that by default and that general category of behavior is basically a bug factory.)  It seems like the only case that's really of interest is adding a constant to a vector.   We could make a function that just does that, if you don't like using move() for it.

revarbat, 2019-07-03:
A `VADD()` of a 2D vector with a 3D vector seems like something we'd want to support.

revarbat, 2019-07-03:
Verified that OpenSCAD returns `[2,2]` for `[1,1]+[1,1,1]`

adrianVmariano, 2019-07-03:
What's the application for adding a 2d vector to a 3d vector?  The example of `[1,1] + [1,1,1]` is precisely the sort of thing that should be an error, because it is probably a bug.  You can do point3d([1,1])+`[1,1,1] or [1,1]+point2d([1,1,1])` as appropriate, which is more clear code.   If I want to add `vec+scalar` I can do `vec+replist(scalar,len(vec))`, I suppose.

revarbat, 2019-07-03:
Hmm. Would it make more sense to change the “casting” functions `point2d()` and `point3d()` to convert scalars?  Then `point3d(1)+[1,1,1]` would work.

adrianVmariano, 2019-07-03:
Well, I think it would be awkward.  Yes, `point3d(1)+[1,1,1]` would work, but the answer would be `[2,1,1]`, which isn't certainly what I was trying to do.  I mean, maybe these functions should work on scalars and singleton arrays (if they don't already).

But to add a scalar to a vector you'd need `point3d(1,fill=1)+[1,1,1]` which requires you to repeat the scalar (which could be bad if it's an expression) and its not simpler than [1,1,1]+replist(1,3).  Also it doesn't handle the case of arbitrary length vectors, like adding 1 to every element of a length 100 vector.  

Maybe we should create an addscalar() function that just adds a specified scalar to every entry in the argument.  It could work on vectors, matrices, etc.

adrianVmariano, 2019-07-03:
```
function add_scalar(list,scalar,ind=0,return=[]) =
   ind==len(list) ? return :
   add_scalar(list,scalar,ind+1,concat(return, [is_list(list[ind]) ? add_scalar(list[ind],scalar) : list[ind]+scalar]));
```

revarbat, 2019-07-03:
My original though had been to have `point3d([1])==[1,0,0]` (as it already does) and `point3d(1)==[1,1,1]`.  Maybe remove `scalar_vec3()`.

revarbat, 2019-07-03:
```openscadfunction add_scalar

adrianVmariano, 2019-07-03:
I think having `point3d(1)` expand to `[1,1,1]` is bad---confusingly different from the behavior on vectors.   If it does something it should do the same thing it does on `[1]`.  But that seems unecessary since the user can just invoke it on the singleton list instead.  

Oops.  Yeah, recursion not needed.  This is not the first time I've unnecessarily deployed recursion, but I caught the other ones before showing you the code.  :)

revarbat, 2019-07-04:
The dark parts of my mind that have had to tweak assembly language calls to low level glibc math routines wants to call this `vadds(v,s)`, but I realize this is probably inappropriate.

revarbat, 2019-07-04:
`vadd_scalar()` to go with `vmul()`, `vdiv()`, and `vabs()`?

adrianVmariano, 2019-07-04:
But it's not limited to vectors, so not really parallel.  It works on [[1,2],[3,4],[5,6]] and also [1,[3,4],[[3,4],3]]].

revarbat, 2019-07-04:
BTW, The few places I already do this, I currently do it like `v+[1,1,1]*s`

adrianVmariano, 2019-07-04:
Yes, but in my case the length of v was unknown and possibly very long.   So it would be `v+replist(1,len(v))*s`.

revarbat, 2019-07-04:
I'll go with `add_scalar()` then, but put it with the vectors.scad file, as it's a close fit, if not exact, to the other vector math calls.

revarbat, 2019-07-10:
This was implemented and the issue never closed.  Closing now.

closed 2019-07-10 by revarbat