BelfrySCAD/BOSL2 #78: [BUG] Grid inconsistencies?

ether42, 2019-07-17:
I was trying to get some shapes in a grid but it seems the children of `grid2d` are slightly rotated.

This simple example should clearly show how children aren't lined up together:
```scad
include <BOSL2/std.scad>
grid2d(spacing=2, cols=5, rows=5) cube(2);
```

[image]

Tested with OpenSCAD 2019.05.

revarbat, 2019-07-17:
Verified that the cubes are slightly rotated.

revarbat, 2019-07-17:
This is a leftover of the transition from `orient` as rotation angles to being a directional vector.
Fixed in 6ce93f75e009a4e26344a8f78f0323ff2817e8fc

closed 2019-07-17 by revarbat