BelfrySCAD/BOSL2 #81: Write text on objects?

adrianVmariano, 2019-07-28:
Do we need something like the write library for writing text on objects?

https://www.thingiverse.com/thing:16193

adrianVmariano, 2019-09-06:
https://github.com/brodykenrick/text_on_OpenSCAD

adrianVmariano, 2020-08-29:
It occurred to me that one way to ease the placement of text onto objects would be to make attachable text, maybe a `writing` module?  Not sure what it would be called.  `print`.   In essence, the align options of text() would be replaced by anchors and at least for simple tasks like subtracting or adding text from planar faces, it would be easy to use.  

However, this wouldn't replace the need for something that can make curved text to go on a cylinder, etc.  So maybe it's not enough by itself?

revarbat, 2021-05-28:
We have distributors for things like `line_of()`, `arc_of()`,  or `ovoid_spread()` which provide a $idx value that could index into a string, and .  Perhaps we need `spiral_spread()` or somesuch for cylindrical and spherical spirals?

revarbat, 2021-05-28:
I do like the idea of attachable text.

revarbat, 2021-05-28:
Thinking on it, though, attachable text is probably only implementable by making a custom version of the `attachable()` module, since the geometry of proportional letters is not accessible to user-space.  This needs further investigation.

revarbat, 2021-05-28:
May be able to make attachable text as leaf nodes that cannot have things attached to them.

adrianVmariano, 2021-05-28:
It seems like it's not necessary to allow things to be attached to text.   I think text that doesn't take children would be fine.

Note that Jordan (I think) was working on code to provide text metrics.  That might be useful for this.  I'm not sure if it was userspace, though.  

What does spiral spread have to do with attachable text?

revarbat, 2021-05-28:
```openscad

revarbat, 2021-05-28:
`atext()` is implemented for an anchorable 3D text block.

adrianVmariano, 2021-09-01:
text = reverse("A long text example to wrap around a cylinder, possibly for a few times.");
    L = 4*len(text);
    maxang = 360*L/(PI*50);
    spiral = [for(a=[0:-1:-maxang]) [25*cos(a), 25*sin(a), 10+30/maxang*a]];
    difference() {
        cyl(d=50, h=50);
        path_spread(spiral, len(text))
            atext(text[$idx], size=5, anchor=str("baseline",CENTER),orient=DOWN);
    }

We already have path_spread which can do this.  There are limitations, though.  You can really only do this with monospaced fonts, or it will look bad.  You have to guess the font spacing.  If we use the new text metrics this would be easier.  But still if the font has ligatures it's going to look funny placed letter by letter.  And kerning may mean that a single letter has different dimensions than the letter in a text string.   

The right solution is to make a path_text() module that places the text on a path so it's easy to control its position and orientation.  I had issues with the text being inverted and backwards.   We could do this without textmetrics by having the user specify a text spacing.  Or I suppose guessing a spacing from the specified text size.

revarbat, 2021-09-02:
When we can get access to kerning and character size info, I know we have a function that can split up a path at various lengths.

adrianVmariano, 2021-09-02:
Not sure we can get kerning data, though.  You can get the size of a piece of text, it looks like.

closed 2021-09-05 by adrianVmariano