BelfrySCAD/BOSL2 #85: nut catches?

adrianVmariano, 2019-08-07:
https://github.com/JohK/nutsnbolts

(Iis this already around somewhere?)

revarbat, 2019-08-09:
I've been making nut catches each time, and not even thinking about it.  This is sensible to add.

revarbat, 2019-08-09:
This may have a dependency on #37 so that nut size specification is consistent.

adrianVmariano, 2019-08-09:
Agreed about the dependency.  I never did a table of nut thicknesses...but nut widths are certainly there.

rjcarlson49, 2022-08-25:
I've put together a personal library for creating screw holes (masks) that include nut traps of various kinds. It uses the screws module from BOSL2 extensively. The code includes a very limited table of nut dimensions for the screws I use regularly. A key part of the code positions the screw hole in various ways to make it easier to move into position in the assembly.

Here it is:

`
include <BOSL2/std.scad>
include <BOSL2/screws.scad>

$fn = $preview ? 30 : 90;
_tiny = 1/128;

/*
The default tolerance is a measured amount. Vertical dimensions
in printed holes require this much extra so that a nut will always 
fit. A much smaller tolerance can work if the hole's cross section 
is in the XY plane. However, the .25 value works in the worst case.
*/ 
defaultTolerance = .25; 

/*************************************************************
* Screws and holes are rendered vertically with head at the top
* Slots for nut traps are in the direction of +X

* extendH extend the hole at the top of the head
* extendS extendthe hole at the bottom of the crew
* gap is the space between head and nut

* pos determines alignment with the XY plane, ie Z == 0
*    0 - Top of head end
*    1 - Screw head seat - the head side of the gap
*    2 - center of gap, ie half way between head and nut
*    3 - nut side of the gap 
*    4 - 1 mm beyond screw end
*    5 - Bottom
*
*    "top" - Top of the hole above the head
*    "head" - Screw head seat - the head side of the gap
*    "gap" - center of gap, ie half way between head and nut
*    "nut" - nut side of the gap 
*    "screw" - 1 mm beyond screw end
*    "bot" - Bottom of the screw hole
*
* nut - "none", "hex", "square" or "self", default: "hex"
* trap - "none", "side" or "inline", default: "inline"
***************************************************************/

/*
screw_context 

Creates a structure that includes all the needed parameters to define 
a screw and the holes necessary for it and its nut, if any. The nut 
information is added automatically.

The data structure allows many calls without the overhead of
rebuilding the set of parameters each time.

The screw can be described in one of three ways, in order of precedence
    specify spec (call screw_info to get the spec)
    specify name (as defined by BOSL2, eg "M5,16", metric 5mm diameter, 16 mm long)
    specify d and l, this assumes a name "Md,l"
    
    For the second and third options, drive and head can also be specified. 
    These will be passed to screw_info to retrieve the spec

name   - BOSL2 string name of the screw, "Md,l" - may replace D and L
d      - diameter of the screw in mm
l      - length of the screw in mm
head   - string head type as defined in BOSL2, defaults to "socket"
drive  - string drive type as defined in BOSL2, defaults to "hex"
spec   - the structure returned by screw_info
[... 353 lines cut ...]
          nut = scNut, trap = scTrap, pokeL = scPoke);

    echo(_HeadH(sc0));
          
    xdistribute(20) {
        ydistribute(20) {
            zrot(trapAngle) screw_hole(sc0);
            if (showPosition == "all") zrot(trapAngle) screw_hole(sc1);
            if (showPosition == "all") zrot(trapAngle) screw_hole(sc2);
            if (showPosition == "all") zrot(trapAngle) screw_hole(sc3);
            if (showPosition == "all") zrot(trapAngle) screw_hole(sc4);
            if (showPosition == "all") zrot(trapAngle) screw_hole(sc5);
        }

        ydistribute(20) {
            zrot(trapAngle) union() {screw_from_context(sc0); nut_from_context(sc0);}
            if (showPosition == "all") zrot(trapAngle) union() {screw_from_context(sc1); nut_from_context(sc1);}
            if (showPosition == "all") zrot(trapAngle) union() {screw_from_context(sc2); nut_from_context(sc2);}
            if (showPosition == "all") zrot(trapAngle) union() {screw_from_context(sc3); nut_from_context(sc3);}
            if (showPosition == "all") zrot(trapAngle) union() {screw_from_context(sc4); nut_from_context(sc4);}
            if (showPosition == "all") zrot(trapAngle) union() {screw_from_context(sc5); nut_from_context(sc5);}
        }
    }
}
`

adrianVmariano, 2022-08-30:
What is the "nut" anchor?  It says the nut side of the gap, but I'm not sure what that means.  Does it mean right above the nut?   Also, when do you use the "gap" anchor?  

You have "inline" and "side" for the nut trap.  What does that mean?

adrianVmariano, 2022-09-03:
An inline trap doesn't look like it traps anything.  What is the intended use for this?

rjcarlson49, 2022-09-04:
The inline trap keeps the nut from rotating. It's hex for a hex nut and square for a square nut. Once you get the screw started it will continue through and become as tight as you want.

rjcarlson49, 2022-09-04:
I considered adding anchor, spin and orient to the screw_context for use when these are always or usually the same in a given application. The anchor, spin and orient arguments in screw_hole(...), etc. would override the ones in the screw_context. It seemed to me that leaving them out of the context was more consistent with the rest of BOSL2. If you like it, that is an option. It's also an option to fold this functionality into existing calls by adding a context argument. Screw_hole, or screw_mask if you prefer, is probably the only one that seems to need to be added.

adrianVmariano, 2022-09-05:
The inline trap should have been more obvious to me.  It makes sense now.  

Right now my thinking on this matter is that you are duplicating too much functionality that already exists for it to make sense for us to directly use your code.   You have basically created a second screw context system even though one already exists.  And you have also duplicated the head-drawing code, which is problematic, since it creates two parallel lines of code to maintain.  If I'm reading your code correctly, the flat head code will not work right on 82 deg or 100 deg angled flat heads, for example.  And if one specified an undercut screw it looks like things would also go awry.   And with my latest code to fix sharp flathead corners, I think there will also be more problems, because you need to use the sharp diameter for a hole instead of the true diameter.  

But this functionality is important, so I do want to figure out a good way to incorporate it.

rjcarlson49, 2022-09-06:
I relied on the BOSL2 stuff as much as I thought I could. The only code I rewrote was the flat head because I couldn't see any other way. I used the angle from screw_info so I don't see why it wouldn't be correct for an 82 or 100 degree screw. As for the other heads, AFAIK they are all round and I use the head diameter from screw_info so I don't know why that would mess up. One reason I had to write the flat head code is that my custom anchors know about the "bottom of the head". For a flat screw I treat this as the bottom of the cone (usually also the top of the thread). This seemed most intuitive to me.

If you point me in a direction where I could use more BOSL2, I can adapt the code more closely.

I'm not sure what you mean about there already being a screw context system. I assume you are talking about screw_info. Screw_info is entirely a description of the screw itself. The data call comes from BOSL2 once the screw is chosen. Screw_context is, except for the incorporated screw_info, entirely a description of what's around the screw. Screw_info is incorporated only to make it more efficient. Because of the use of structs, I think the code is pretty resilient to changes in screw_info as long as clear incompatibilities are not introduced. Because OpenSCAD effectively only has constants, a user can't really add values to a struct. Yeah I know you can make a copy, but that's inelegant at best. So if you wanted to add my context information to screw_info, you'd have to a lot of arguments to screw_info or a lot of arguments to other modules.

There's a natural trade off. You can often add more arguments to a single module or you can split the module into 2 or three and each one will have fewer arguments. I love BOSL2 but IMHO it leans too far in the former direction. Case in point, I never even noticed the oversize argument for screw. I had no idea that functionality existed. This what at a minimum I think you need to a screw_hole (or _mask) module.

Typically I find I never have a single screw hole I need, it's always several. That's why screw_context. Specifying a bunch of arguments on a call and having several identical calls with all the arguments is error prone because any change requires changing all the calls. Putting all the arguments in "variables" allows you to make a change in one place, but that's kind of messy too.

adrianVmariano, 2022-09-07:
I've been working on fixing various bugs and deficiencies in screw() for the past couple months.  Revar and I had a discussion a couple weeks ago about hole making and I split off the hole making functionality into screw_hole() but I think the version currently posted has some bugs and deficiencies. I think I'm almost done, but a key thing that will be of interest to you which is still missing is flexible anchoring.  I got kind of stuck on the naming and anchoring for the threaded part of the screw (which might not be threaded), the shank, the shank + threaded part (shaft), and how to handle all the cases in a clear fashion.   Hopefully I'll finish this in the next couple days.      

You're right, the head size computation in your code looks reasonable for flat heads as things stand. (I was evidently looking at the wrong thing.)  But in the new code they look like this:

[image]

Screw on the left, and matching sized clearance hole mask on the right.  That's with the ASME "normal" hole tolerance and $slop=0.  If you use the actual head diameter (that's on the left) to make a clearance hole you'll get a clearance hole that leaves the flat part of the screw sticking up out of the hole.  This whole business with flatheads was pretty annoying to get nailed down.  

Fundamentally, I can't see a justification for generating your screw (or screw mask), including the head, by any other means than a call to screw() itself, because then you're duplicating functionality.  That was the original reason I tried to use screw() to also make masks---to avoid making the code more complex.   For uniformity of the interface the anchors should be identical between screw() and screw_hole(), so duplicating code creates an issue there as well, especially if we end up tweaking the anchors.  My new code has screw_hole() as basically a pass-through that invokes screw, and I hid the arguments to screw() that are only suited to holes, like counterbore and internal.  

If you have a struct and you rewrite it to add to or change entries you're making a new data structure.  If you create a new data structure that has the struct as a sub-part you're also creating a new data structure.   Maybe OpenSCAD is clever and uses a reference to the existing struct in your new one?  But the amount of memory is insignificant.  I don't think it matters which one you do from a performance perspective.  The struct_set() function will change or extend structures, so the code to do it is simple enough.  In fact, I even imagined that a user might want to create a screw and then tweak it.  

To me it seems like what you're doing with screw_context should be done with a user-defined module.  That is, if you need some specific screw arrangement a bunch of times you make a module to produce it and then you can create it as needed.  

I think that designing the library interface is hard, and there's no guarantee that we get it right the first time.  I'm curious if there are other specific cases that leap to mind where you feel BOSL2 has too much functionality in one module, and it should instead have been split.  Both Revar and I think that nut traps should be a separate module from the screw_hole module, so in this instance, you're leaning the other way, and in fact actually combining two things in one module.

rjcarlson49, 2022-09-07:
I'm in the process of simplifying the code I last posted. There is really not very much code that might need maintenance. Most of it is in setting up and accessing the screw_context. And I understand how structs work.

However, I think we are converging.

I certainly entertained the idea of making the traps a separate call and I don't think much is lost by separating it.

The real key is my custom anchors. All of them are actually found relative to the screw and nut position, rather than the entire screw hole. The only place where the anchors need to "know" about the head and screw extensions of the hole are when you use TOP or BOTTOM when placing a screw hole. In my experience, I've never found myself using those enchors except in test code.

The custom anchors "gap" and "nut" require knowledge of the nut position, AKA, the size of the gap. So in order to use those anchors with screw (and thus screw_hole in your pass thru), would you add a gap parameter? Something else I haven't thought of?

Other than gap, the entries in the context are all either not strictly needed (like spec (screw_info) and nutSpec because they can be handled by passing screw_info, OR they are only needed as arguments to nut_trap call.

So to implement my set of custom anchors and nut traps and have all the screws, nuts, holes and traps use the same set:

1) No screw_context is necessary.
2) Add a gap argument to screw() and nut(). 
3) Add screw_hole() that takes screw_info and gap arguments. Include self tapping in screw hole.
4) Add nut_trap() that would require screw_info and gap arguments, plus trap type, trapL, extendH, extendS, pokeL and pokeD and nut type (hex or square).

I think this will do everything my code does and within your constraints. The biggest change is the introduction of gap and the 2 anchors that depend on gap. Lest you be tempted to leave these out, they are the ones I use 90% of the time. The gap determines where the mating surface is and that is almost always the reference point I need when I move the screw hole or screw into place.

One more point. I often render a screw and nut in place in my rendering just to check for correctness, so I need the same anchors in screw, nut, screw_hole and nut_trap.

Well one more. I include a "poke hole" option. I quickly found I liked having a way to get a nut out of a trap.

rjcarlson49, 2022-09-07:
I notice also that your nut info is not nearly as robust as the screw tables in the code. I think this should be fixed and square nuts accommodated.

rjcarlson49, 2022-09-07:
Screw_hole also needs the extendH and extendS arguments.

adrianVmariano, 2022-09-08:
Getting the data and then entering it and organizing it is a major challenge in all of this.  Part of the big problem I was trying to fix is that torx drivers didn't have valid depth info anywhere, and there was generic torx depth in screw_drive.scad that was bogus (much too deep).  I would say we don't have nut info at all...I did recently download a nut spec standard, asme b18.2.2.   But I need to nail down the screws before thinking about nuts.  For the record, the long term plan is to eliminate metric_screws() completely and have screws.scad replace it.  I think the only nut dimensions are in metric_screws.scad right now.  To do nuts seriously presumably means building a nut_info() function analogous to screw_info with lookup tables or formulas for the different nuts.  I don't know how many different sizes/types of nuts there are.  "heavy"  "jam"  hex and square.  Is that all?   One nice thing about the standard I found is that it has formulas in appendix A1 so that means a lot less data entry and tables to worry about.  I found some formulas for flat head dimensions as well when I was trying to straighten that out.   ASME seems nicer about revealing underlying formulas than ISO.  

I have non-flat-head anchoring working in my screws code, but am at the moment confused about how anchors should be defined for flat heads.  Once I come to a conclusion about that, I think it shouldn't take long to implement and I can push my current code.   (Though...hmmm....maybe I should also write docs first.)  

I wonder if nut_trap should be drawn by the same code that draws nuts.  That would be a way to ensure/enforce uniformity.  

The poke hole seems like a great idea.  I've on occasion tried to extract hardware from nut traps with some difficulty.  

So "gap" is the point right above the nut, yes?  So if you put the nut somewhere...then the "gap" is automatically right above wherever you just put the nut?   I mean, I was thinking of placing the screw and then making the nut trap relative to the screw, but there's no reason it couldn't be done in reverse: place the nut trap so the "gap" point is where you need it and then place the screw relative to the nut trap.  

Could you post an example or two of your code being used to create a nut trap in an actual context that shows the typical use case and makes it clear why the gap is important?  (Something simple, not a huge model with 20 screws.)    I mean, if I understand this all (no guarantees) then you should be able to create the gap you need with something like:

screw_hole(....) position("head_bot") down(gap) nut_trap(....,anchor=TOP);

You mention screw_hole taking a gap argument.  What would it do with that argument?  

With your self-tapping screws you are making the holes flat walled but sized to match d_minor instead of d_major?  This is actually a significant issue because if I need access to threading information in screw_hole() I need to change its interface somehow.  (Basically you need to be able to specify threads and then say you don't want threads.  I guess maybe a tapping=true argument could work.)

adrianVmariano, 2022-09-08:
I don't understand the need for extendH and extendS.  Why not just make the screw longer?

rjcarlson49, 2022-09-08:
You can't make the head longer. And wouldn't you want the code to reflect the actual length of the screw as opposed to the length of the whole you need? Also the screws come in standard lengths, like M3,12 or M3,16.

I've done designs where the screw is as much as 2-3 cm below the surface, requiring an ExtendH and S of 30. If you have a side trap a standard length for ExtendS of 2mm could be used. But when an inline trap is used you have to know who long it must be.

You could make the extensions really long and assume that could cover all cases, but what if you have an unusual case where the extensions would run into some other part as you difference it out? If you went with this option, you would make head holes, side traps and inline traps very long and standard screw holes 2 beyond the end of the screw. But how long is long enough?

adrianVmariano, 2022-09-08:
So it sounds like extendH corresponds to my counterbore parameter.  

I'm not sure I understand why the code shouldn't reflect the length of the hole being created.  If mean, you could always write `length=12+4` or whatever to highlight the relationship to the actual screw.   The one benefit I can see to this extension idea is that it makes it easier to create the screw that goes in the hole.  This is basically a situation where we face the question: is it better to complicate the interface (add a parameter) or simplify the interface (user knows how to use "+" operation if necessary).  I wonder what @revarbat thinks about this.

rjcarlson49, 2022-09-09:
> Getting the data and then entering it and organizing it is a major challenge in all of this. Part of the big problem I was trying to fix is that torx drivers didn't have valid depth info anywhere, [...]

**No doubt that's a difficult issue. I suspect that even your screw data is incomplete if looked at by an actual ME. I think nut_info should take screw_info as an argument. That specifies a LOT. Then maybe only one more argument to add the nut type.**

**I'm pretty sure I am getting all my information about metric screws (the only ones I use) from screw_info. I did not realize there is a metric_screws().**

> I have non-flat-head anchoring working in my screws code, but am at the moment confused about how anchors should be defined for flat heads. Once I come to a conclusion about that, I think it [...]

**I really think a flat head should be regarded as having a "top", the flat part and a bottom, the spot where the taper intersects the shaft. That intuitively matches the definition of heads. Also, otherwise the "top" and "head" anchors would be the same and you would have no anchor and the bottom of the taper.**

> I wonder if nut_trap should be drawn by the same code that draws nuts. That would be a way to ensure/enforce uniformity.

**I really don't like that idea. The nut trap is really easy to make. You only need the distance between faces, the nut thickness and the type, hex or square. I really don't like the idea of drawing a screw hole with the screw code. I haven't looked yet but I imagine the "oversize" argument to screw REALLY complicates the code. Wouldn't you rather get rid of that? What does it even mean to oversize threads? I think these are prime examples where two different calls makes the code simpler and more understandable.**

> The poke hole seems like a great idea. I've on occasion tried to extract hardware from nut traps with some difficulty.
>
> So "gap" is the point right above the nut, yes? So if you put the nut [...]

**The "gap" anchor is the midpoint between the bottom of the head and the top of the nut. The "nut" anchor is the top of the nut.**

> Could you post an example or two of your code being used to create a nut trap in an actual context that shows the typical use case and makes it clear why the gap is important? (Something simple, not [...]

**I'll do that.**

> You mention screw_hole taking a gap argument. What would it do with that argument?

**It is used in calculating the custom anchors. Assuming the same anchors are supported by screw and screw_hole uses screw, it would pass gap to screw as well. I have code for the "anchors" argument that fits in a simple function that can take just screw_info as an argument. I'll recode it to use struct_val calls and then you can just drop it into your code. Hopefully tomorrow.**

> With your self-tapping screws you are making the holes flat walled but sized to match d_minor instead of d_major? This is actually a significant issue because if I need access to threading [...]

**You pass the screw_info structure to screw_hole. The thread info is in there.**

**Given a day or two I am pretty sure I can code up my stuff in a form pretty close to something you can just drop into yours that will conform to what I described in my long comment yesterday. I hope you'll wait to see that.**

rjcarlson49, 2022-09-09:
This is a simple ring that clamps onto a bar on my golf push cart. It has a mounting for a magnet. The clamp is formed by a cube with a screw hole with inline trap. A slice of the cube is removed from within the gap so that the screw pulls the two parts together. Crude but it works. It's on my cart now.

You can see the code that shows the screw in place and the cross section reference that lets me look at it that way.

```
include <BCLib/common.3.scad>

barD = 31 + 1; // +1 for pad
magnetD = 18;
magnetH = 3;

bandH = 10;
bandT = 3; 

screwName = "M3,12";

screwContext = screw_context(name = screwName, gap = 6, entendH = 6, extendS = 6, trap = "inline");

//crossSection("y") {
    left(bandH/2 + 1)
    clamp();

    right(barD/2)
    band();

    fwd(magnetD/2 + 1 - bandH/2)
    right(barD)
    setting();
//}

/*
left(bandH/2 + 1) {
    screw_from_context(screwContext);
    nut_from_context(screwContext);
}
*/

module clamp() {
    difference() {
        cube([bandH, bandH, 14], anchor = CENTER);
        
        // Remove
        screw_hole(screwContext, anchor = "gap");

        cube([10, 10, 2], anchor = CENTER);
    }
}

module band() {
    difference() {
        ycyl(h = bandH, d = barD + bandT*2);
        
        // Remove
        ycyl(h = bandH + t2, d = barD);
        
        left(barD/2)
        cube([10, 10, 2], anchor = CENTER);
    }
}

module setting() {
    right(bandT - tiny)
    difference() {
        xcyl(d = magnetD + 2 + t2, h = 3, anchor = LEFT);
        
        // Remove
        xcyl(d = magnetD+ t2, h = 4, anchor = LEFT);
    }

    xcyl(d = magnetD + 2 + t2, h = bandT, anchor = LEFT);
}
```

adrianVmariano, 2022-09-09:
I am making a PR for my current code, but I don't think anchoring works right yet.  (It turns out the existing posted version has a serious bug, so need to get something out.)   But you might want to take a look at screw_hole().  

My expectation is that very few users will use the screw_info structure.  I anticipate that a user will write something like `screw("M6",...) to make a screw and `screw_hole("M6",...)` to make a hole  As a user, why do I want some extra data structure to hold the information "M6"?   Also, I think the common use case is going to be to use only one of screw() or screw_hole(), not both, so there's no reason to believe a screw specification structure already exists.  I mainly expose that structure for users to enable them to do more complex stuff, like making nonstandard screws, tweak head geometry, or adjusting driver settings.  So all the modules should work without needing a screw_info spec as input.  

I can't think of any reason to oversize threads in a perfect world, but for 3d printing, oversizing the threaded hole may be necessary for parts to mate, depending on the accuracy of your machine, and I wouldn't be surprised if a screw needed to be *undersized* to mate with metal hardware, so a negative oversize.  The way this works is that the diameter just gets oversize added to it.  I don't think it's a massive complication.    The threads keep the same profile.  I don't recall at this point---I think with my printer I was able to get parts to mate by just using the tolerance settings.  But for pipe threads I know I needed to artificially adjust the size of the threads (with the $slop parameter) to mate with an existing pipe threaded fitting.   Since screw() doesn't respect $slop (because it applies to holes only) undersizing may be needed under some circumstances.  

The nut_info() function should definitely *not* take a screw spec as a required argument.   This would lead to the mess of having to do nut_info(screw_info(.....)) to create a nut.  You should be able to write something like `nut("M6")` and get a nut, and someone who wants to mess around with a nut_info structure should not need to explicitly call screw_info to do it.   It does seem that the way things are currently set up, nut_info will need to call screw_info with a headless screw to get threading info.  Perhaps threading sizing info should be split off into a subfunction?  

So for screw_hole and screw the anchoring has to be the same, and the anchoring alone is complicated enough that repeating it seems like a bad idea.   I didn't try to go through screw() and rip out everything that's not needed for screw holes to see if the code got significantly simpler, but why bother?  Keep in mind that users may want to make threadless screws to show in models without time consuming calculation, so we need the ability to make threadless screws anyway.  It's not like it's special for holes.   I do not understand what the point is of splitting the functionality and repeating code.  I see no way in which it doesn't result in increased complications for maintenance.  

The way I have it, screw_hole() is basically a pass-through to screw().  

Note that there are two different issues here, one having to do with the interface seen by the user and the other one about code internals.  You may be right that there is no benefit from combining nut trap functionality and nut functionality into a single module.

adrianVmariano, 2022-09-09:
Oh, and I have literally hundreds of pages of standards documents for screws and related stuff.  There's no way that the contents of BOSL2 is even close to "complete".   Wanna go add truss heads?  Oval heads?   Revar wants shoulder screws, which is still a complication.  And note that if you want to make a screw hole for a shoulder screw you need more information now, and more complication if you're repeating it in another module.

adrianVmariano, 2022-09-09:
Regarding standards, I took a look at asme b18.2.2 and it lists specs for 14 different kinds of nut.  

And mysteriously Table 1-1 says "dimensions of square and hex machine screw nuts" and shows a picture of a square nut with the label "no chamfer allowed".  And then two pages later, in Table 2 we see "Dimensions of Square Nuts" and a 25 deg chamfer is required.  

I haven't tried to track down the ISO standard for nuts to see what the metric scene looks like.

rjcarlson49, 2022-09-12:
Ok, here it is. Screw_context is gone. There are basically 3 calls, screw_hole, screw_in_hole and nut_in_hole. All use identical anchors and produce identical positioning for all three outputs when the same inputs are used.

If you look at line 217, there is a commented out call to screw() that can replace the call above it. Note I do not use the "internal" parameter because that is not in the copy of BOSL2 I current have. 

I did add the slight counterbore on the my flat head code so it should be flush or close when extendH is 0. 

I have kept all the "access" functions rather than use struct_val because I like to hide details of data structure as much as possible.

```
//   use <../!lib/utils.0.scad>
include <BOSL2/std.scad>
include <BOSL2/screws.scad>

$fn = $preview ? 30 : 90;
_tiny = 1/128;

/*
The default tolerance is a measured amount. Vertical dimensions
in printed holes require this much extra so that a nut will always 
fit. A much smaller tolerance can work if the hole's cross section 
is in the XY plane. However, the .25 value works in the worst case.
*/ 
defaultTolerance = .25; 

/*************************************************************
Screws and holes are rendered vertically with head at the top
Slots for nut traps are in the direction of +X

extendH extend the hole at the top of the head
extendS extendthe hole at the bottom of the crew
gap is the space between head and nut

The screw hole is generated vertically.
If there is a side trap, it extends toward the +X axis.
If there is a poke hole with the side trap, it extends on -X

nut - "none", "hex", "square" or "self", default: "hex"
trap - "none", "side" or "inline", default: "inline"

Several special anchors are defined. All these anchors
are on the Z axis. The designated anchor will be at the XYZ origin.

   TOP     - At the top of the screw hole
   "top"   - At the top of the screw head
   "head"  - At the bottom of the screw head
   "gap"   - Center of gap, ie half way between head and nut
   "nut"   - Nut side of the gap 
   "screw" - At the screw end
   "bot"   - Bottom of the screw
   BOT     - Bottom of the screw hole

   spin can be used to rotate a side trap to the desired direction.
   orient can be used to orient the screw hole as desired, 
   UP (default), DOWN, LEFT, RIGHT, etc.
***************************************************************/

/*
screw_hole(sc, anchor = "gap", spin = 0, orient = UP)
    sc - screw context
    
    Creates a screw hole includes any nut trap (including "poke hole").
    It is vertically positioned as designated in the context.
    If the trap is a side trap, it extends in the direction of the x axis.
    ANy poke hole is opposite the side trap.

screw_from_context(sc, anchor = "gap", spin = 0, orient = UP)
    sc - screw context

nut_from_context(sc, anchor = "gap", spin = 0, orient = UP)
    sc - screw context
[... 491 lines cut ...]
            if (show("head"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "head");
                nut_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "head");
            }
            if (show("gap"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "gap");
                nut_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "gap");
            }
            if (show("nut"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "nut");
                nut_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "nut");
            }
            if (show("bot"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "bot");
                nut_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "bot");
            }
            if (show(BOT))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = BOT);
                nut_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = BOT);
            }
        }
        //*/
    }
}
```

rjcarlson49, 2022-09-12:
I have a question about creating attachable objects. How is the problem of coincident faces handled. I made my objects larger than the outside anchor points by _tiny, which I have set to 1/128. How does the rest of BOSL2 handle it?

adrianVmariano, 2022-09-12:
There is an overlap argument and also $overlap global that makes attached objects sink in by that amount.  I think Revar had it default to something nonzero, but then that lead to some kind of confusing situation.  

My thinking on this matter is that if you are making a mask, then by all means, make it slightly bigger than its attached volume so that it subtracts in a well-behaved manner.  But if you are making an object, think hard before doing this kind of thing, because it means you'll be making an object explicitly different than what you said you were making.  This could result in a 3d model where objects aren't all on the same z base, which might conceivably create problems for printing.  (I once had a series of print failures that were a result of sphere() not actually producing the specified radius, resulting in a small offset of part of the model.)  We used to have $slop set with a nonzero default and that also created a problem where a user wondered why an object was the wrong size.  

But consider the implications of making things oversized:  you request a 1x1x1 cube and get a 1.001x1.001x1.001 cube.  It's a little puzzling.  And if you make the cube the right size and then put the anchors inside you can end up with a cube that's not where you expected it.  

I think that the majority of the objects in the BOSL2 library were designed before the attachments system was invented, so there may be room for improvement.  At this moment I still haven't quite nailed down how attachments will work for screws and screw holes, and shoulder screws are presenting various questions.   (For example on attachments, I want it to be natural to make holes that are spaced away from a wall by a specified amount, or whose heads are spaced away from a wall by a specified amount.)  Did you make "top" of a length L screw the point distance L from the screw's tip?  That is, the point above the head on a flathead and underneath the head on other screws?  

Note that you can still move objects around even after using attach or position.  So something like attach() down(tiny) thing() is a way to get an overlap, though admitedly it seems somewhat clumsy.  

Also I talked briefly with Revar and he does not like the idea of two length arguments (e.g. length and extendS).

rjcarlson49, 2022-09-13:
First, I found some problems in the self tap code posted above. I'm fixing them.

I see the problem with making objects the "wrong" size behind the scenes. As you said, no problem for screw_hole I think.

There aren't 2 length arguments. When you call screw_info, you specify the length of the screw. When you call screw_hole then you need to specify how far above and below the hole must go. I don't see a way around that at the moment.

Currently I assume that the head height in the screw tables is the max height of a given head type. I found that for flat screws, the head height was always 0. My code treats it differently for calculating where the "head" anchor goes because by my definition it is the bottom of the head. The virtual cylinder that the screw_hole anchors are using uses head_diameter, so if you want to place a hole n mm from a wall, that should be easy. By default, the screw_hole is aligned on the Z axis so placing the center of a hole n mm from a wall is also easy.

My "top" anchor is the top of the head, wherever that is. The "head" anchor is the bottom of the head. So "head" is L above the tip of an MDxL screw. I don't have an anchor for the end of a shank, but that would be easy to add. The TOP anchor of a screw_hole is the top of the hole including the head extension.

rjcarlson49, 2022-09-13:
I've realized that extendS really only has meaning when an inline trap is wanted. ExtendS could be replaced by trapL for the inline trap case and all other screw holes could be made 1 or 2 mm longer than L. 1 is probably enough in all cases though, right?

adrianVmariano, 2022-09-13:
Right now you can specify a screw with `screw("M6", length=25)` or you can do `screw("M6,25")`.  If we add an "extend" argument so you have `length=25, extend=2` then there will be two arguments that specify length.   The head height in the tables is the height of the head above the screw, or the length the head adds to the screw length.  Maybe that's a design error and it should be the actual head height.  The original code never even computed the head height for flat heads.  Is that value in the tables the maximum height?  Maybe, but I wouldn't guarantee it.  It's possible it might be the middle height height allowed by the tolerance range or typical head type or the head type of a screw sold by McMaster.  

You want to be able to place the screw head next to something but also the screw shaft.  I'm not sure if that means the default anchor should be a cylinder that contains the screw (or screw hole mask), or if the default should be a cylinder whose diameter is the screw diameter.  And I guess it seems like there should be an anchor that on a screw of length L is distance L from the tip, but I don't know what to call that anchor.   This would be on top of a flat head but underneath other heads and seems like the natural point of reference for a screw.

adrianVmariano, 2022-09-13:
I think the hole should be the length the user requested.  If the user wants it longer, the user can request it longer.  This is not a case where excess length will vanish, unless the user is making a through hole.  

If the inline trap is a separate module (which is how Revar and I think) then it will have its own length parameter, just like the sideways trap needs a length argument to specify how far it extends to the side.

rjcarlson49, 2022-09-13:
Extend (or trapL) is not relevant to screw_info, only screw_hole.

The cylinder for the anchors should be the diameter of the head. It's really hard for a user to obtain the head diameter so having anchors there helps the user. if the user really wants to reference the shaft, the shaft diameter and its center are both known.

adrianVmariano, 2022-09-13:
No, actually, shaft diameter is not known.  Only the **nominal** shaft diameter is known.

adrianVmariano, 2022-09-13:
There will need to be multiple anchor types.  I had actually wondered if we could make the anchor type system cleaner, but it seemed like maybe it's difficult on the implementation side.  I was thinking something like anchor=["head",BOT], anchor=["threads", TOP], or whatever, as compared to the more clumsy "atype" system currently used for this.

adrianVmariano, 2022-09-13:
I remembered why I didn't put the flathead height into the screw structure:  the flat head height isn't known until you determine the screw diameter, and the screw structure only contains the **nominal** diameter, not the actual diameter, so it's impossible to compute it.

rjcarlson49, 2022-09-13:
> No, actually, shaft diameter is not known. Only the **nominal** shaft diameter is known.

Technically, all the measurements are nominal. I don't see how that matters. The standards are mostly for maximum measurements just so you can ensure fit.

rjcarlson49, 2022-09-13:
> I remembered why I didn't put the flathead height into the screw structure: the flat head height isn't known until you determine the screw diameter, and the screw structure only contains the **nominal** diameter, not the actual diameter, so it's impossible to compute it.

I don't understand why this matters for screw holes. The nominal screw diameter is a known maximum and the hole has to be based on that.

rjcarlson49, 2022-09-13:
> There will need to be multiple anchor types. I had actually wondered if we could make the anchor type system cleaner, but it seemed like maybe it's difficult on the implementation side. I was [...]

This seems needlessly complicated since I found a rational and so far very useful set of anchors based on the current system with a single anchor argument.

I don't understand what you mean by "atype" system.

rjcarlson49, 2022-09-13:
> If the inline trap is a separate module (which is how Revar and I think) then it will have its own length parameter, just like the sideways trap needs a length argument to specify how far it extends to the side.

I get the impression that your desire for this is all based on wanting to enhance the existing screw code and use it for drawing screw holes. With that starting point, I don't see how you get a total system that allows you to position screw, nut, screw_hole and trap all simply and easily. The screw interface is already very complicated and hard to understand. Case in point is that I had no idea there were features intended to let you use it for screw holes.

My API for screw_holes is ONE module that creates screw_hole and trap and positions it with a wide variety of anchor points. Two other modules can position a screw or nut within that hole, all using identical arguments and anchors. I don't see how you can achieve that with what you are contemplating.

adrianVmariano, 2022-09-13:
"Technically all measurements are nominal"

Huh?  That doesn't make any sense.  In fact,  **no** measurement is nominal because if you have measured something then that makes it actual (with error bars if you like).  A nominal dimension is a a name, hopefully with some connection to the real size of the part, like "2x4" lumber which is actually a smaller size.  In the case of screws, if you request an M6 screw the model you get from screw() will **not** be 6mm in diameter, but will be adjusted according to the ISO tolerance formulas.  The actual dimension of the screw is the size that OpenSCAD actually creates the part at.  That is clearly a different thing than the nominal dimension.   Note that the nominal screw diameter is also not the same as the diameter for a hole.  ISO/ASME specify tolerances for this and the hole must be even larger.  If you base the flat head height on nominal rather than actual diameter then the flat head will have the wrong size, or the anchor point that is supposed to be under the head won't actually be at the intersection of the head and shaft, or you could potentially end up with a small gap between the shaft and head---depending on how the implementation works.  Basically something will go wrong somewhere, I think.  

Avoiding this complication was actually why I originally modeled flat heads as full cones and just unioned them with the screw, but this produced bad threads at the top and for very short screws, the cone would be too long.  So it wasn't really a satisfactory solution.  But the head height is a derived value so if I'm going to compute the head height, I want it to be correct, not just sort of correct.  

Regarding an anchor located at the underside of a flat head (top of the screw shaft): is there any use for this?  Neither Revar nor I could think of one.  

There needs to be a way to anchor on the side of the shaft.  There needs to be a way to anchor on the side of the head.  These are two inconsistent requirements.  The solution we have for this is to allow you to select a different attachment geometry using atype, which is short for attachment type.  So you might do `screw(....,atype="shaft", anchor=RIGHT)` to anchor on the right of the shaft, but `screw(...,atype="head", anchor=RIGHT)` to anchor on the right side of the head.   There are various other places around BOSL2 where we needed multiple conflicting attachment geometries, which is why we introduced atype.  But I do think it feels like a kludge, and it's inflexible because you have to do all your anchoring relative to a single anchor type.  The idea above is cleaner because it combines the anchor type with the anchor location, so at least syntactically it would allow for multiple different attachment geometries to be used at once. 

As far as I can tell, the complications for the screw model are there because modeling screws is complicated.  Modeling screw holes basically comes for free once the screw model works, so why wouldn't I take advantage?  (The one single addition needed is the table of screw hole tolerances from ASME B18.2.8 and ISO 273, and I guess forcing flatheads to have sharp heads.)  I don't understand what simplifications I could make if I decided to duplicate the code and have two entirely separate modules for screws and holes.  It's a negligible simplification of the screw() code and it means the hole making code is more complicated and creates a maintenance headache due to duplicated code.

I do not like the idea of the trap as part of the hole code, as I've mentioned before.  When I mentioned this idea to Revar he immediately said that the trap should be a separate attachable module.  I have not explored this yet to see if there is some compelling reason to bind together two separate objects like you want to do.

adrianVmariano, 2022-09-13:
Note, just to be clear about screw code:  all of the threading code in the whole library is intended to make threaded rods or screw masks.  The idea that I got this idea to enhance screw() to make holes is not right.  It's more like I realized that the capability that was supposed to be there was missing.  I originally wrote the screw code a couple years ago and got burnt out on it and didn't really finish, so implementation gaps and deficiencies are no big shock.

rjcarlson49, 2022-09-14:
> I do not like the idea of the trap as part of the hole code, as I've mentioned before. When I mentioned this idea to Revar he immediately said that the trap should be a separate attachable module. I [...]

I could easily separate the trap code from the hole code. However, the trap always attaches its TOP to the hole's "nut" anchor (in my set of anchors.) Why make the user do this if it is always the same?

adrianVmariano, 2022-09-14:
It seems like I might not want the trap there.  What if I have 20mm long screws and want the trap 8mm up from the bottom to work with my model?

adrianVmariano, 2022-09-14:
Is it possible to use two nut traps to connect three parts together?

adrianVmariano, 2022-09-14:
I suppose there's probably a better solution.

rjcarlson49, 2022-09-14:
The your gap argument would be set to 12, 20 - 8. Then "nut" gives you that spot.

rjcarlson49, 2022-09-14:
> Is it possible to use two nut traps to connect three parts together?

Can't see how.

rjcarlson49, 2022-09-14:
The gap argument IS the distance between bottom of head and top of nut. By definition that's where the nut trap goes.

adrianVmariano, 2022-09-14:
But this means you're making a different mechanism for positioning an object relative to another object instead of using the existing anchoring mechanism.    Also in my example I want to position the nut trap relative to the bottom of the screw, not the top.

adrianVmariano, 2022-09-14:
Here's another example:  my model has some external feature that defines where the trap needs to go, so I want to position the trap relative to that feature and wherever it hits the screw, I don't much care or want to try to figure out---maybe I don't know where that point is if it's relative to other objects.

rjcarlson49, 2022-09-14:
> all of the threading code in the whole library is intended to make threaded rods or screw masks.

The ability to draw screws and nuts in the total assembly is valuable, what Nop Head calls vitamins (does everyone call it this?) So very precisely drawing screws using ISO tolerances is not really necessary. It's only needed for what you want to print. I see that people print larger threaded items sometimes but they are not printing M5 screws AFAIK.

We do print screw holes though. Every document I have looked at about screws has listed max values for things like head diameter and screw diameter. I don't think I have encountered a case the max of something did not match the nominal value. The docs I looked at were engineering docs not the ISO standards. I did measure all the screws I have bought and they are indeed tiny bit below nominal (max) and they are not necessarily all the same. But since I am making holes, it seems only the max is really important.

BTW, I fully feel your pain re standards. Back in the 80s and 90s I was on various standards committees both US and international for networking protocols. Seeing the sausage made is perhaps worse than having to read and interpret it. People in the standards business also have to get used the fact that their standards are often very loosely interpreted.

adrianVmariano, 2022-09-14:
I printed 1/4-20 stuff and it worked fine, both screws and threaded holes.   I think M5 would work.   Would I trust it to hold my weight.  I don't think so.  But no reason someone is not printing it for light duty use. I mean, is someone printing separate screws?  Probably not a common application, though there are youtube videos about how to do this well, and the effect of print orientation on printed screw strength.   But parts that have male threaded components, it's a possibility.   Printing threaded holes seems like a better plan.  Also someone with an SLS printer could easily be printing much finer parts.  Someone printing in metal could also be printing small parts.  The point is that assuming that we can be inaccurate because "nobody cares" doesn't make sense to me.  If you want that, you don't need a screw library.  The point of having a library is that it does it in some substantially correct way---that is, it does the hard stuff.  If you just want to make 6mm holes for your M6 screws you don't need a library for that.  The other thing is that making proper threading work at large scales automatically makes it work at small scales---I mean, unless you build your thread lookup tables to fail below M16 or something.  All the core logic is all the same.  

ASME b18-2.8 lists the smallest allowed hole size for a 1/4" screw at 0.266 inches and that's with the "close" tolerance.  With the "normal" tolerance the smallest allowed size is 0.281".  Smallest allowed hole for an M6 bolt is 6.4mm. 

If you're drawing screws in assembly you probably don't want to waste time drawing the threads, which is one of the reason that drawing screws isn't different than drawing screw holes.  You can set tolerance to zero if you want to make them at nominal size---and if you're confident CGAL won't barf on the overlapping hole and screw surfaces.  But the point is that you're now basically arguing for a separate capability:  saying that **sometimes** you don't need it accurate doesn't mean that there aren't other times where you do.  It's easy to make it inaccurate.  That's not the hard part.   Also there's no **harm** in drawing the screws at the correct size (other than code complexity, I suppose).  It doesn't take more time.  The only thing that sucks up time is drawing the threads.  I feel like I've been a little overly verbose here to make this main point:  there are times when you need the screws or holes to be accurate.  Therefore the code needs to be able to produce accurate screws and (possibly threaded) holes.  As a result, there's no benefit in cutting corners, because it just multiplies the number of cases: the accurate case and the corner-cut case.  

I've never modeled something where I wanted to bother including the screws.  It doesn't seem like a useful thing to do from a modeling perspective.  It's seems like it would only be useful if you want a pretty picture of the assembly, put together with its screws.  I've never heard the term "vitamins" applied that way before.  Nophead builds very big, complicated assemblies with lots of parts, so he may know something I don't.  But basically if you are trying to use this scheme to "debug" your model but the screw making code is the same as the screw_hole making code then it will just automatically work, so if you have problems, it doesn't seem like they would be revealed by adding the screws.  They might even be concealed (consider screw holes that intersect each other).

rjcarlson49, 2022-09-15:
I never meant to argue against that kind of accuracy and detail for modeling of screws. I just don't think it can be used as a reason not to draw screw holes in the manner that I do.

If there are standards for holes related to screws I'd be happy to incorporate them. I'd love to see such numbers in screw_info. I'd certainly use them.

I think it's also true that standards are written with machining in mind. The kind of 3D printing most of us do has errors of .1-.2 mm for features in the 1-10 mm range. I suspect the standards are aimed at machined parts, drilled, CNC, etc. Anyone doing 3D printing is probably adjusting for their particular print process. I know with my own printer, different numbers are good for a horizontally printed hole and a vertically printed hole. So a "t" or "oversize" parameter is required regardless of whether we have "min hole size" data element.

Anyway I appreciate all the discussion here. I think my code is a lot better now as a result. If you don't include something like it, it will be no problem for me to continue to use it.

rjcarlson49, 2022-09-15:
> If you're drawing screws in assembly you probably don't want to waste time drawing the threads, 

>
I've never modeled something where I wanted to bother including the screws. It doesn't seem like a useful thing to do from a modeling perspective. It's seems like it would only be useful if you want a pretty picture of the assembly, put together with its screws. 

I do this about half the time I'd guess, and I do like to visualize how everything fits together. I want a screw to look like a screw for this purpose. I came up with a crossSection() module so I can see inside quickly as well. I use that a lot.

You have a point about how the screw being positioned "automatically" defeats some of the purpose of this kind of visual checking. However, part of that is because the method makes it hard to make a mistake. For example if I just make a mistake positioning the screw, it's not really a mistake because the screw is just there for visualization.

rjcarlson49, 2022-09-15:
You mentioned that you don't see a need for my "head" anchor, the bottom of the head. Though it would not be used often it is a key feature of the hole. A screw and nut bind two parts together. The binding force is applied at the top of the nut and the bottom of the head. Both spots are clearly important. Imagine the simplest case, a 2 mm sheet to be screwed to a 4 mm sheet. If the head is to sit right on the 2 mm sheet then you can use the "head" anchor and move it to the top surface of the 2 mm sheet. Or move the "head" anchor to the "mating surface" then up 2 mm to be at the top of the top sheet.

Anyway, the feature is important so I am pretty sure somepeopel will find uses for it.

adrianVmariano, 2022-09-15:
I have a test part I printed with 10mm holes and I measured them at somewhere between 10.00 and 9.99 mm.  So **my** printer at least is more accurate that 0.1-0.2 mm!  But also, really, there is other tech out there.  I haven't checked the accuracy of printing in aluminum or stainless steel on Shapeways.   But I know that SLA printers are ten times more accurate than the type of printer I have.  Maybe even more accurate than that.  And while they're more expensive than the extrusion type machines, they're not out of reach for home use.  (I have no clue how strong the material is, though.)  

Right now the standard hole clearance data is embedded in the screw_hole() module in my code, which is posted to github, so you can take a look if you want.   Personally I found the size of the clearance holes bigger than I would have expected, so I'm just obediently following the standards in this matter.   The problem with putting it in screw_info is that screw_info has nominal data.  The tolerances are in a different structure.  I'm not sure it makes sense to put the clearance tolerances there, though, because there's a different tolerance terminology for the clearance holes than for threading, so combining it doesn't buy you anything, really.  It may make sense to split it into a separate function, though.  

I was not talking about your anchors but my anchors.  I've just been trying to figure out what I think is the fully general set of anchors.  The anchor I don't see a use for is the bottom of a **flat** head.  Under a regular head is very important, I totally agree.  In fact I'd think it might be the most important anchor point for a screw.   The exact location of the "bottom" of a flat head seems like it's a little bit uncertain, since it will change a bit if you change screw tolerance, or use the nominal diameter, or make a clearance hole.  

How is crossSection any different than either projection() or the BOSL2 front_half(), half_space(), etc?

adrianVmariano, 2022-09-15:
I'm not using accuracy as the argument against your approach but rather that your approach seems to lead to duplication, that is, we end up with two different screw_hole functions.  In the context of a library we want to try to make the interface clean and not have two modules that do the same thing.  We haven't been entirely successful at this, but mostly, I think.  I put a lot of effort into going over the library and just trying to decide whether everything was consistent and and seemed to have reasonable names and so on.  And if it seemed like we had duplicate functionality, I asked: can we merge it or eliminate one of the duplicates.   (There are places where Revar and I wrote the same code and we didn't realize it.  Revar wrote a polygon rounding and chamfering function that duplicated my round_polygon except it also did chamfers, so trying to merge that functionality was complicated.)

rjcarlson49, 2022-09-18:
The problem is that your current code can make a screw hole, but only maybe 40% of what is needed. The head part of the hole must have an extension. The head (hole) height needs to be increased as well as the diameter. If a screw is to be used self tapping, the "tapping" part of those must be smaller in D than the threads and a couple of mm longer. If there is a trap, you need to have all sorts of dimensions for that trap. Both inline and side traps may need to be of arbitrary length and depend on whether a hex or square nut is being used. Lastly you need to position the trap relative to the screw hole.

I agree with you that code should exist in one place. The place for a screw_hole code is not as part of screw(). The code you have added to screw() so far only gets you part way. The only code that might be duplicated in mine is the code for heads and that appears to be less than a half dozen lines. And it would be easy to call your _head module instead if you wanted.

Since the screw() code cannot realistically be expanded to the whole screw hole job, there should be a screw_hole() module and all screw functionality should be removed from screw() to make it simpler and clarify that screw_hole is the way to do that. Two other modules, nut_in_hole() and screw_in_hole() exist only to draw a nut or screw in a screw_hole() that has  been created. There are there just for convenience and call the existing nut() and screw() modules so they do not duplicate any code.

adrianVmariano, 2022-09-18:
Have you actually looked at the current code?  

For screw hole you use the counterbore argument which extends a flathead or replaces the specified head for other types of head.  The clearance size by default should be adequate---I used the ISO/ASME clearance that applies to the threads on the head as well---but if it's not right you can adjust with the various oversize parameters.  And of course $slop applies to aid with printer overextrusion.  

I have asked this before:  what can I delete from screw if I remove the functionality that is there only for holes?  I think the answer is I can delete two things:  the (hidden) _counterbore argument and the _internal argument.   Doing this does not create some huge simplification.  The cost of this is that I have to make screw_hole() basically a copy of all the code from screw() instead of a pass-through module, and screw() is about 135 lines of code.  I really don't understand how a second copy of that code is desirable.   Note that most of screw() is parameter wrangling, error handling, and attachment support, and a screw_hole() module needs to do a lot of that same stuff.  Your code is lacking in error handling.   My screw() implementation has 19 lines of error handling.  I wish I didn't have to write error handling, but it's essential in a library.  We don't have enough.   Attachment support is about 40 lines.   Taking out _counterbore would eliminate two lines of code I think and simplify some others.  Taking out internal has almost no impact.  So where's the big gain?  

As I have said before, traps appear to me to be separate objects, and should therefore be produced by separate modules.  I do **not** think that screw() should produce traps.  Definitely not.   I actually think that inline traps and side traps should be two separate modules---they are totally different shapes and will have different anchoring.  

Regarding what you're calling "self tapping", that is something I need to look in to.  I'm not sure what a hole like that should be called.  The hole isn't self-tapping, and neither is the screw---it's just a normal machine screw.   If I look up self-tapping screws on McMaster I get sheet metal screws.  In any case, it seems like it should be possible to make these using the existing screw() code, though it might be annoying because of having to evaluate the tolerance outside of screw to figure out the dimensions.  That seems kind of ugly.  

Hmmm.  I went and looked at your code.  One observation is that all of your extra calls like _ScrewD() conceal what's actually happening, so they function as code obfuscators.  When I see _ScrewD I don't know which of the four different screw diameters it's going to produce.  But what I see is that you have based your hole size for the "self tapping holes" on the **nominal** diameter.  That makes it trivial to implement this using the existing screw() module---the shoulder screw support that Revar requested can be used to make the larger top section.  I also notice that you subtract a tolerance from the head but not the shaft, which seems suspicious.  I guess it depends on the intended purpose of the tolerance parameter, but there should be a way to adjust the hole size----including the "self-tapping" part---in order to compensate for printer over-extrusion.  It appears that this capability is missing.  

You are claiming that screw() can only do 40% of what is needed.  Other than traps, what can screw() not do that is needed?  

The way I see things right now, the code structure would be:

screw() - core code for screw and screw hole making, exposed to users for screws only
screw_hole() - code for making screw holes, pass through to screw()
shoulder_screw() - code for shoulder screws, pass through to screw()
nut_trap_inline() - draws inline traps
nut_trap_side() - draws side traps
nut() - draws nuts

rjcarlson49, 2022-09-19:
You asked before for an example. This is a portion of something I build that supports 2 water bottle cages. The type that mount on bikes. Two hex shaped blocks hold screw holes with nut traps and poke holes. I believe you will be able to actually run this code if you also have my screw hole code. A couple of other library modules are copied here to make the example. As it happens this is an example of a "head" anchor.

```
include <BOSL2/std.scad>

$fn = $preview ? 45 : 90;
t = .1;
t2 = 2 * t;
tiny = 1/128;

use <BClib/screws.4.scad>;

blockH = 12;
blockW = 28;
blockTent = 2;

mountOffset = 64;

m5x8 = screw_info(name = "M5,8", head = "socket", drive = "hex");

crossSection("y", fwd = -3)
block();

up(mountOffset)
crossSection("y", fwd = -3)
block();

module block() {
    difference() { 
        hexCyl(id = blockW, h = blockH, tent = blockTent);
        
        // Remove
        zrot(-60)
        fwd(blockW/2 + tiny)
        screw_hole(m5x8, anchor = "head", gap = 3, nut = "hex", trap = "side", orient = FORWARD, spin = 90);

        zrot(60 + tiny)
        fwd(blockW/2 + tiny)
        screw_hole(m5x8, anchor = "head", gap = 3, nut = "hex", trap = "side", orient = FORWARD, spin = 90);
    }
}

/*
Below are library modules copied here
*/

module crossSection(view = "x", right = 0, fwd = 0, up = 0) {
    an = view == "x" ? LEFT
       : view == "y" ? FRONT 
       : view == "z" ? BOTTOM 
       : view == "-x" ? RIGHT 
       : view == "-y" ? BACK 
       : view == "-z" ? TOP 
       : LEFT; 

    difference() {
        union() {
            children();
        }
        right(right)
        fwd(fwd)
        up(up)
        cube(1000, anchor = an);
    }
}

/*
    hexCyl
    
    Produce a hexagonal cylinder along z axis

    id - face to face
    h - length of column
    tent - size of "tents" on both ends
*/

module hexCyl(l = -1, h = -1, id = -1, d = -1, tent = 0, anchor = CENTER, orient = UP, spin = 0) {
    H = (h > 0) ? h : (l > 0) ? l : 1;
    ID = (id > 0) ? id : (d > 0) ? d : 1;

    attachable(anchor = anchor, 
               spin = spin, 
               orient = orient, 
               d = ID, 
               h = H + tent*2) 
    {
        hull() {
            if (tent > 0) up(H/2 + tent/2) sphere(tent);

            down(H/2)
            linear_extrude(H)
            hexagon(id = ID);

            if (tent > 0) down(H/2 + tent/2) sphere(tent);
        }
        
        children();
    } // attachable
} // hexCyl
```

rjcarlson49, 2022-09-19:
And here's the current version of screw hole. You should be able to comment out the BClib references.

```
//   use <../!lib/utils.0.scad>
include <BOSL2/std.scad>
include <BOSL2/screws.scad>

include <BClib/values.scad>
use <BClib/basics.4.scad>

/*
The default tolerance is a measured amount. Vertical dimensions
in printed holes require this much extra so that a nut will always 
fit. A much smaller tolerance can work if the hole's cross section 
is in the XY plane. However, the .25 value works in the worst case.
*/ 
defaultTolerance = .25; 

/*************************************************************
Screws and holes are rendered vertically with head at the top
Slots for nut traps are in the direction of +X

extendH extend the hole at the top of the head
extendS extendthe hole at the bottom of the crew
gap is the space between head and nut

The screw hole is generated vertically.
If there is a side trap, it extends toward the +X axis.
If there is a poke hole with the side trap, it extends on -X

nut - "none", "hex", "square" or "self", default: "hex"
trap - "none", "side" or "inline", default: "inline"

Several special anchors are defined. All these anchors
are on the Z axis. The designated anchor will be at the XYZ origin.

   TOP     - At the top of the screw hole
   "top"   - At the top of the screw head
   "head"  - At the bottom of the screw head
   "gap"   - Center of gap, ie half way between head and nut
   "nut"   - Nut side of the gap 
   "screw" - At the screw end
   "bot"   - Bottom of the screw
   BOT     - Bottom of the screw hole

   spin can be used to rotate a side trap to the desired direction.
   orient can be used to orient the screw hole as desired, 
   UP (default), DOWN, LEFT, RIGHT, etc.
***************************************************************/

/*
screw_hole(sc, anchor = "gap", spin = 0, orient = UP)
    sc - screw context
    
    Creates a screw hole includes any nut trap (including "poke hole").
    It is vertically positioned as designated in the context.
    If the trap is a side trap, it extends in the direction of the x axis.
    ANy poke hole is opposite the side trap.

screw_from_context(sc, anchor = "gap", spin = 0, orient = UP)
    sc - screw context

nut_from_context(sc, anchor = "gap", spin = 0, orient = UP)
    sc - screw context
*/
     
/*

Nut Info information

[... 477 lines cut ...]
            if (show("head"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "head");
                nut_in_hole(si, nut = nut, gap = gap, extendH = extendH, extendS = extendS, anchor = "head");
            }
            if (show("gap"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "gap");
                nut_in_hole(si, nut = nut, gap = gap, extendH = extendH, extendS = extendS, anchor = "gap");
            }
            if (show("nut"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "nut");
                nut_in_hole(si, nut = nut, gap = gap, extendH = extendH, extendS = extendS, anchor = "nut");
            }
            if (show("bot"))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = "bot");
                nut_in_hole(si, nut = nut, gap = gap, extendH = extendH, extendS = extendS, anchor = "bot");
            }
            if (show(BOT))    union() {
                screw_in_hole(si, gap = gap, extendH = extendH, extendS = extendS, anchor = BOT);
                nut_in_hole(si, nut = nut, gap = gap, extendH = extendH, extendS = extendS, anchor = BOT);
            }
        }
        //*/
    }
}
```

[29 more replies]

closed 2022-11-01 by adrianVmariano