BelfrySCAD/BOSL2 #100: bosl_version()

adrianVmariano, 2019-08-12:
Seems like there should be a way to find the library version.

revarbat, 2019-08-12:
What format, though?
- bosl_version = 2.0000;
- bosl_version = "2.0000";
- bosl_version = "2.00.00";

adrianVmariano, 2019-08-12:
Do you have a preference or plan on how you want to do versioning?  If you did three part versions what do the different parts mean?  Like would the second part indicate API changes and the third part only bug fixes?  Is there any reason to return a vector, [2,3,2] instead of 2.3.2?   Is there any way to have versioning happen automatically when stuff gets checked in?  I suppose the resulting version codes might be ugly.   

The advantage of doing a function is that you could request the version in different forms. (I suppose multiple variables would achieve the same result.)  

Would it make sense to make a function that fails if the BOSL verision is too old?  So something like `bosl_required(2.3)` and it would error out with "BOSL 2.2 is installed but version 2.3 or better is required."

revarbat, 2019-11-04:
Making versioning automatic on checkin is theoretically doable, but I'll have to do some research.  It'll probably need us to make versioning based on date like OpenSCAD does.  I do like the idea of `bosl_require()`.

revarbat, 2019-11-04:
Hmm.  Maybe major=2, minor=0 to start with, rev=an auto-ingrementing number that resets to 0 on every minor version change.  Major gets changed manually only on breaking backwards compatability, minor upgrades manually per official release, and rev gets incremented on every checkin.  Give functions to retrieve the version as a float, string, or list.  As float, formatted as M.mmmrrr (like `2.001043`), as string formatted like `"2.1.43"`, and as list, `[2,1,43]`.

revarbat, 2019-11-05:
version.scad added in commit 1170bbb489510b14ba9f8f938d9308958898798d

closed 2019-11-05 by revarbat