BelfrySCAD/BOSL2 #103: Add knurling to pco1881_cap()

sej7278, 2019-08-25:
Just spent an hour making a pc01881 cap using BOSL1, then noticed you already had a function in BOSL2!

Only problem being that you never get a bottle top that's smooth. So could you either add knurling or vertical serations?

i managed to do it but it seemed a bit messy as i had to figure out the dimensions of the cylinder to diff against e.g:

```
difference()
{
    knurled_cylinder(l=12.2, d=32, helix=45);
    up(0.1) cyl(l=12.2, d=30.6);
}

down(6.1) pco1881_cap(1);
```

[image]

revarbat, 2019-08-30:
Knurling and ribbing implemented via `texture=` argument.

revarbat, 2019-08-31:
Added PCO 1810 neck and cap in bf407307c78902cd2dacbdeea1aca7807ba6b0e9

closed 2019-08-31 by revarbat

sej7278, 2019-08-31:
that's excellent thanks!!!