BelfrySCAD/BOSL2 #107: add snap pins

adrianVmariano, 2019-09-08:
Quoting a post to the forum:

The best way to find out how the snap pins work is to print Emmet's gear
cube https://www.thingiverse.com/thing:213946
There are two versions of the hole inset, one which allows rotation and one
which is fixed. They hold very solidly, but can be pulled apart with enough
effort. The smallest I have been able to make useful pins is 4mm long with
a 2.5mm diameter, anything smaller than that is too fragile to use.
The fully customisable pins are at https://www.thingiverse.com/thing:3218332,
and the output options include a test socket you can print to try it... The
code generates an STL for the pin, which is printed flat, and an STL for
the socket, which is difference'd from the surface you want to embed the
pin in.

I printed the geared cube and the pins work quite well.  Seem like a nice part for the toolbox.

revarbat, 2019-09-19:
Verrrry interesting.  Definitely a good candidate.

adrianVmariano, 2019-09-19:
Note also that I did use lube on the pins, which definitely helped.

adrianVmariano, 2020-08-18:
Now have snap_pins and rabbit_clips.

closed 2020-08-18 by adrianVmariano