BelfrySCAD/BOSL2 #109: remove knurling from std.scad?

adrianVmariano, 2019-09-18:
Does knurling belong in std.scad?

revarbat, 2019-09-19:
No.  Unfortunately it's there because threading.scad needs it for the PCO18XX modules.  Which, now that I think of it, should probably be moved out to their own library file, which will make knurling.scad not needed in std.scad.

adrianVmariano, 2019-09-19:
Hmmm.  Can threading.scad "use" knurling.scad?   I'm not sure I undersatnd why moving the PCO18xx stuff into its own module makes things any different than the existing setup with regards to the includes.  

I wonder, actually, whether switching over completely to "include" is the right strategy, but I haven't really thought carefully about it.   It seems like there are some important functions where performance is important where using "use" seems like it might slow down computations.  But for rounding.scad, for example, the modules are slow, so if there's a bit of extra overhead it makes no difference.

revarbat, 2019-09-19:
Fixed in d4f79e87a113954f63b74691df5739078d912aca

closed 2019-09-19 by revarbat