BelfrySCAD/BOSL2 #114: Use large cube size for *_half modules

adrianVmariano, 2019-12-16:
Prioritize producing the correct model over a subtle display issue.  Document that users bothered by the display issue can use a small s.  I think s should be at least 1000, maybe even 10000.

revarbat, 2019-12-17:
The behavior in previous versions of OpenSCAD was far from subtle.  Zooming to view all used to include the shapes that were in the subtractive side of `difference()`.  Thus, if you diff a humongous cube from a shape, it would zoom to the size of the subtracted cube.  It looks like they fixed that.  I concur that a large default size is fine now.

revarbat, 2019-12-17:
When s > ~35000 then OpenSCAD preview rendering becomes dodgy.
I've updated s to 10000 for all the _half() modules.
Committed in 60b92e1d5619060792825516a6ee8cf126729332

closed 2019-12-17 by revarbat