BelfrySCAD/BOSL2 #116: rename "spread" modules

adrianVmariano, 2020-01-17:
I can never remember what the spread modules do, and don't find the name intuitive...or easy to distinguish semantically from "distribute"...so I can never remember the right names for these.  

I think distribute is a natural name---distributing the children---but spread means the same thing as distribute, hence the confusion.  They need to be renamed to copy or repeat or replicate....something like that.

adrianVmariano, 2020-01-17:
Your arguments against are basically just one argument:  you're used to the current names.   But accessibility for other users needs to be considered.  

Your analogy shows exactly how the name is wrong:  "spreading out a hand of cards along a line".  This is precisely what I expect it to do:  take a set of things (the children) and spread them out.  But what it *actually* does it take a single thing and copy it and spread out the copies.   The "distribute" name means the same thing as spread, which is the problem.  The solution is not to rename distribute, which has a name that matches its function and which seems fine to me, but to rename spread, which has a name that does *not* match is function.  

If you want short names xcopy(), ycopy() and zcopy() are short.  Less short would be xrepeat(), yrepeat and zrepeat().  And xrep(), etc would be really short.  

I suggest you add better names and keep your own names as undocumented or deprecated compatibility commands.  It's kind of ugly, but the current names are so bad and confusing that I have to look them up every time I want to use the commands because it's impossible for me to remember which one copies.   I agree that these are important commands and the name should be short.  I don't like "spread_copies" particularly, though I suppose it does address the naming problem.   It's kind of clumsy, and as you say, sort of long for a function that is pretty important.

adrianVmariano, 2020-01-17:
Actually if I was coming to this fresh I'd suggest that "spread" should do what distribute does and "repeat" should do what spread does.

revarbat, 2020-01-17:
Eh.  I see spread and distribute as having pretty identical meanings. I looked them up in the dictionary, and they even define themselves in terms of the other word. Distribute always bugged me as being long, so "spread" seems the obvious choice. The issue seems to me that the words are too identical in meaning, so we shouldn't use them both.

adrianVmariano, 2020-01-17:
Yes, that is precisely the problem.   

Distribute is long, it's true.  I don't know if I've ever used it except for when I did examples of all the polyhedra.  If it's rarely used, brevity is less important.  It would make sense to use "spread" for what distribute does now...but that would be confusing to anybody who has used the library before. If you really wanted to to be shorter you could do xdist(), etc.

revarbat, 2020-01-17:
No seriously, I did a grep of my files.  I really *do* have hundreds of uses of `*spread()`.  Every single time I wanted a pair or triplet of items, I used them.  There's dozens of instances even in other BOSL2 files.

revarbat, 2020-01-17:
I honestly don't see either "spread" or "distribute" as more indicative or dealing with children versus making copies.

    Spread cards along the line. (multiple individual objects)
    Spread holes along the line. (copies of the hole)
    Distribute cards along the line. (multiple individual objects)
    Distribute holes along the line.  (copies of the hole)

The multiplicity or duplication is implied entirely by the noun, not the verb.
So, both meanings being equal, and "spread" being both already in place, well used, and shorter, I'd keep "spread" as the base name, and add "_children" for the (currently) distribute versions.

revarbat, 2020-01-17:
And now I realize your last comment was saying the opposite of what I first read it as.  Oops.  Sorry. We seem to broadly agree here.

If we were starting from scratch, I wouldn't be opposed to `xdist()` etc.

adrianVmariano, 2020-01-17:
To me the word "spread" means that you are starting with something and putting it out over an area.  There is no notion of making copies.  Duplication is not implied.  It's also not implied by "distribute" which suggests that you have a box of things and you're going to put them out in an area.  

So if you say "spread holes" then now you are no longer copying.  The copying happened when hole became holes.  Spread is operating on a collection (of holes).  The word "spread" is NOT doing the copying.  It's the same with the use of distribute, since the two words are synonyms.  If you apply spread or distribute to a set of copies then it makes sense.  But that's not what's going on.  Try this on for size:

Spread a hole along the line.   (A really long hole?)
Spread this box along the line.  (Does the box get stretched out?  Cut into pieces?)

My efforts to interpret spread applied to a singular object are forced and also not what we mean.  

If you have "spread_copies" and "spread_children" that actually makes sense and works, but it suffers from being long.  I think that "spread" by itself is confusing unless it does what distribute currently does.

adrianVmariano, 2020-01-17:
I do not think we agree.  My position is that "spread" and "distribute" are synonyms and neither one makes copies, so the existing "spread" functions are confusing.  I swear I have had to consult the manual every time I've used spread.  It's this situation that made me realize that there's a problem.

revarbat, 2020-01-17:
This has no favorable outcomes, as far as I can see.

By this logic, `*spread()`, `*distribute()`, and by extension `*dist()` are invalid names, as they are unclear on copy vs children.  We *definitely cannot* rename `*distribute()` to `*spread()`, as that will have BOSL1 users (and myself) debugging broken smears of subparts instead of copies. Far worse than just getting a syntax error.  It's bad enough that users of BOSL1 `trapezoid()` will get a 2D part instead of a prismoid in BOSL2.

That pretty much means `*distribute()` would have to be named something like `*distribute_children()` or `*spread_children()`.  All other things being equal, the latter wins by being shorter.

We do *not* want to use both "distribute" and "spread", again to avoid confusion.

This means that we either have to lengthen `*spread()` to `*spread_copies()` for consistency,  or we have to rename it to something like `*repeat()` or `*copies()`.

When I originally added `*spread()`, I replaced my old uses of `grid_of(xa=[-X/2,X/2],ya=[-Y/2,Y/2])` with the much cleaner `xspread(X) yspread(Y)` constructs.  With the "_copies" renaming, that would expand to `xspread_copies(X) yspread_copies(Y)` or `grid2d(spacing=[X,Y],rows=2,cols=2)` neither of which is concise in any way.

If `*spread()` is renamed to `*repeat()` that would keep the same length.  However, nothing about `xrepeat()` says to me that it's spreading copies over a line.  So instead of being unclear on copies vs children, it's unclear on what it does with the copies.

If `*spread() is renamed to `*copies()`, it has exactly the same issues as `*repeat()`.

If our choices are doubling the name length, or changing the name without improving it semantically in any demonstrable way, why rename it?

I do agree that `*distribute()` needs renaming, at least.

revarbat, 2020-01-17:
There's also the argument that what is currently called `*distribute()` is a lot less useful than what is currently `*spread()`.  I only added `*distribute()` on a whim a few years after I added `*spread()`, and it's not much useful except for example images and parts tray layouts.  And it's finicky at doing the latter.

The more useful one really should have the shorter name.

adrianVmariano, 2020-01-17:
No, I think *spread, *distribute, or *dist are valid names and are clear:  they clearly mean that the children are being arranged along a line.  I think it's fine to keep the *distribute names.   What is NOT clear is using any of these names to refer to making copies.   So from a semantic point of view, it would be ok to have `distribute` and `distribute_copies` and I don't think I would get confused.   The point is that in the same way that you listed examples "spread these cards" or "distribute these objects" when you write `distribute(){children}` you have provided the stuff to be distributed.  

When you write `spread() child()` you have provided only one thing to be spread.  And like I said above, it doesn't make sense.  How do you spread one thing?  

Now with regards to using repeat, yes, it's true that simply repeat doesn't say what exactly is done with the copies, but consider from a user's point of view in trying to remember the commands.  What might `repeat` do?  What are the possibilities---what makes sense?  What are the points of confusion?    If you try to list all the possibilities for what `xrepeat() child()` might do...what can you come up with?  Because I can't really think of anything but one thing.  Maybe I'll ask around and see what the reaction is.

adrianVmariano, 2020-01-17:
I tried talking to two people who write code, but don't use OpenSCAD about the terminology.  Neither one liked spread (the current name).  Ideas that got suggested:  repeat, replicate, make_copies, line_of. 

I also have identified the following commands that position copies:  place_copies, arc_of, *rot_copies, ovoid_spread, path_spread, grid_2d, grid_3d.  I'm not sure if we should be trying to make things uniform across this set of commands.  But it's something to keep in mind.

revarbat, 2020-01-18:
Other inconsistencies are that we renamed `grid_of()` to `grid3d()` partly because "_of" was unnecessary. But `arc_of()` still exists. I'd actually argue that the "_of" indicates the copying.  And it seems to me that `spread()` (no prefix) would have to become `line_of()`.  Renaming `*spread()` to `*copies()` would be consistent with `place_copies()`.  And then there's `path_spread()`.

adrianVmariano, 2020-01-18:
I'm not sure, actually, what would be consistent with place_copies.  What would replace the word "place"?    I'm also not sure that trying to make all of these commands exactly parallel is the best strategy.  I don't feel like "xcopies()" exactly makes sense.  It seems like "xreplicate" is better.  (One of my victims thought replicate was better than repeat, for what that's worth, and I think I agree.)  

Yes, arc_of exists.  And it certainly can't be reduced to "arc" because that's something else.  It could perhaps be arc_copies.   Oh, and actually `line_copies` was also suggested in my discussion today, so I guess xline_copies, yline_copies, etc?  I can't say that the "right" answer is popping out of all of these options.

adrianVmariano, 2020-01-18:
Actually is arc_of redundant given zrot_copies?   I suppose with the latter the user has to compute the desired angles.

adrianVmariano, 2020-01-18:
Note also that the *rot_copies modules are rotational analogs of place_copies, not of spread.  Perhaps it makes sense to name the commands that do the distributional math for you differently.

revarbat, 2020-01-18:
"Replicate" is just using extra letters to say "repeat" in a way that sounds fancy.

revarbat, 2020-01-18:
I can see renaming `place_copies()` to `move_copies()`, so that it aligned with `move()` in the same way `rot_copies()` aligns with `rot()`.

revarbat, 2020-01-18:
Actually, "replicate" and "copy" are even closer than "replicate" and "repeat".  We have been using a "_copies" suffix already to indicate duplication, with `xrot_copies()`, `place_copies()`, `xflip_copy()`, etc.

revarbat, 2020-01-18:
And, BTW, `rot_copies()` is totally distributional.  It can make N copies around a ring.  Or at each given rotation.

revarbat, 2020-01-18:
I think `arc_of()` would become redundant if `zrot_copies()` could take an end angle argument to allow partial circles.

revarbat, 2020-01-18:
Oh.  It'd need X and Y scaling of placement as well.

revarbat, 2020-01-18:
My non-coder friend comments that old Corel Draw called spreading copies along a line "arrange".

adrianVmariano, 2020-01-20:
Doesn't the `rot` argument to `zrot_copies` enable you to effectively give a start and end angle argument, by supplying a range of points that start and end at the desired angles?

revarbat, 2020-01-20:
Well, you can do something like `zrot_copies(rots=[30:15:150])`, yes, but you have to calculate the stepsize youself.

lar3ry, 2020-03-20:
I rather like distribute, and I don't mind spread, but I think the two could be combined.
A optional argument to x-, y- or z- distribute could be num_copies.

So 
zdistribute(sizes=[20],numcopies = 5, spacing=30) {
	sphere(10);
}

Would change it from being a list of 5 children, would make it an implied list of 5 identical children.

I find the spread as well as the distibute quite handy and time-saving, but they seem to me to be the same thing but with different names.

As an aside, I reported a bug in zdistribute, but in looking at it further, I think perhaps I misinterpreted what the description means in distributors.scad. 
Neither 
    zdistribute(spacing, [sizes])
    zdistribute(l, [sizes])
does what I expected.

revarbat, 2020-03-30:
I renamed `[xyz]spread()` to `[xyz]copies()` and `spread()` to `line_of()`.  `place_copies()` was renamed to `move_copies()` to be orthogonal with `move()`.  The `grid2d()` arguments were changed so that `grid2d(spacing,n)` is a close match to how `[xyz]copies(spacing,n)` works.

revarbat, 2020-03-30:
I've not determined what to do with `[xyz]distribute()` as yet, so I'm leaving them as is for the time being.

My main remaining related issues:
- [ ] `arc_of()` should be merged into `*rot_copies()`
- [ ] `*rot_copies()` use of `sa=` needs sanity fix.  Needs offset for first copy angle.
- [ ] `grid2d()` should be enhanced to work in YZ, XZ, and arbitrary planes.
- [ ] `grid3d(xa,ya,za)` functionality should be merged into `grid2d()`, and the entire thing renamed to `grid()`

revarbat, 2021-02-02:
Executive decision that I'm not going to do the remaining tweaks.

closed 2021-02-02 by revarbat