BelfrySCAD/BOSL2 #118: Define final skin() behavior

adrianVmariano, 2020-01-22:
In the interest of nailing down and stabilizing skin() I’m suggesting  the following as a conversation starter.   Part of my intention is (A) to make the module usable, but not overly complicate the API.  And part of my intention is that it (B) should generally produce good looking comprehensible output.   I do not think the goal is to have the module always work and produce some output…however horrible…just for the sake of producing output.   It also can’t do everything, so the case of small vertex count polygons can be handled in an automatic way, but it might do the wrong thing.  Getting this right would complicate the API and the fully general problem is very hard. 

So the basic idea is that skin links a sequence of polygons that have the same number of vertices, but with some flexibility for noncompliant input lists:

 

1.  Optionally split vertices of input polygons so that they all match (e.g. to the largest polygon (vertex_split=”max”), or to a user specified N (vertex_split=12) that is even larger).  This is for aligning polygons of small, but different vertex count and should produce decent looking results, but maybe not always what the user wants.   This would be done using repeat_entries() so it would be quasi-uniform.   This option is the worst on my list for user comprehensibility since it may split polygons in ways the user didn’t expect or doesn’t really understand, but it’s the only general mechanism for producing non-horrible skins of incompatible low vertex count polygons. 
1.  Optionally resample the input polygons to a constant point count with subdivide_path, either using the method=”length” based resampling (which is good for “continuous” shapes) or the method=”segment” resampling (which is good for polygons).  User species total number of points?  (Or user specifies number of points per segment?   Is this option needed, or user can do multiplication?)  Note that if the input vertex count is small results will only be OK if the shape has coplanar faces.  If it is a curved surface, this resampling is essential to produce non-horrible results.   (You need both vertex_split and resampling with polygons.) 
1.   Optionally reindex polygon[i+1] to match polygon[i] (using reindex_polygon) for the whole list. Note challenge here is that reindex_polygon only works for 2d inputs because it tries to make the polygons have the same direction (cw vs ccw).   I think the solution is to make reindex_polygon skip cw/ccw alignment for 3d input and just run as usual, and I think this will be better than trying to do some complex mapping from 3d to 2d.    
1.   Add slices.  Either take a user input number (or list), or perhaps have the program calculate a slice count designed to produce roughly 45 deg triangles.  Perhaps the latter automatic method as a slices=”auto” default with user specified slice counts (slices=25) as an option, which would insert 25 slices between every input pair of polygons.  User can give slices=0 if slices aren’t needed.   Is there some other way to add slices based on separation between the polygons?  (Separation is not well defined for non-parallel polygons.)  Note that inserting slices is essential to get non-horrible input if the shape is curved. 
1.    Skin should handle polygons in either orientation and still produce a valid polyhedron.   (In my dovetail code I have to call reverse on both of my two polygons or I get an invalid reversed polyhedron.)   Is this possible?    Note if the user gives mixed input where some polygons need to be reversed and others do not it would be nice to handle that case (during the reindex phase) if we can, but that’s not the issue here.  I’m talking here just about the case of valid input, entirely reversed.  
1.    Skin needs to check its input polygon list to make sure it is 3d.  It’s a simple check with arraydim, but right now the code fails catastrophically and actually crashes openscad if you give it a list of 2d polygons…which I do all the time. 
1.    We could provide a mechanism for specifying the z coordinates of 2d polygons since it seems like this would be a common use case and I think would make cleaner code than having to run path3d on every polygon like I currently do.   If we take 2d input like this we can potentially reindex it before promoting to 3d and handle inconsistent winding directions.  

So what’s missing, or excluded from my list.  

*   Complex alignment algorithms to map polygons to other polygons such as the angle or distance methods.   I haven’t seen evidence that these work, I couldn’t understand WFT they were doing (so users are mystified), and the output either looks the same as uniform (so why bother) or looks bad, so they fail condition (B) above.   Note that a “distance” method would be interesting and useful if it would map, a square with rounded corners in the obvious fashion to a square with pointed corners.  This is basically the equivalent of doing reindex_polygon but for polygons with non-equal vertex count.   It seems to me to be computationally intractable, since there are exponentially many mappings.  I suppose you could map each point to its nearest neighbor on the next polygon and check if the result is a proper ordering.  If yes, declare victory, of no, error.  Not sure if this would have broad applicability or not---it’s too local to work in a lot of interesting cases where you need the algorithm I think is intractable. 
*    General alignment code for polygons with different vertex counts, done with vertex duplication.    I think the general case is too complex to handle within skin(), possibly too complex to handle generally within OpenSCAD.   Beyond repeat_entries, which is a little uncontrolled,  I have a function that can handle the case of mapping together a sequence of polygons as long as the vertex count is monotonically increasing (and by reflection the reverse case of decreasing).  I could make a version that would handle one peak, so increase for a while and then decrease.   (Not sure it’s worth the trouble.  The user can call the simpler program twice.)   But this sort of mapping should be done outside of skin() by the user.  I’ve given up on writing something more sophisticated.   A user who needs it can work it out for the specific case.   
*    Ability to change resampling method (between “segment” and “length”) for every polygon.  Complicates the API and if the user needs it, then do it outside of skin.
*   Ability to actually rotate polygon (not just reindex) using align_polygon.  Out of the scope of skin itself.  User can do this if desired and we should show examples.   (Note that doing this requires first adjusting vertices to be compatible either by repeating points or subdividing.)

adrianVmariano, 2020-01-25:
Maybe it would work to do distance alignment by mapping the shapes to 2d, aligning their centroids, and then associating points with the nearest point on the other polygon.  And then checking if the result is a valid sequential alignment.  I wonder how much of the time this would actually succeed in producing an alignment and how much of the time it would fail.

revarbat, 2020-01-25:
That last comment pretty much exactly describes the “distance” method as it is currently.

adrianVmariano, 2020-01-25:
Did you change it?  Because I'm pretty sure you did something different before.

adrianVmariano, 2020-01-25:
```
ell = yscale(2,p=circle(r=5));
skin([path3d(ell,0), zrot(90, p=path3d(ell,4))], method="distance");
```
Above is a counterexample to your claim.  I can see edges that are not the shortest edge.  

There is another issue, which is that you still need to be able to add slices.  The section of the shape that looks like it's appropriately aligned looks terrible because it needs slices.

adrianVmariano, 2020-01-25:
I think the algorithm I proposed would fail in this case because it wouldn't map points onto every point of the destination curve.   Maybe it would work better if the source curve had extra points.  It seems like there would be a point count such that it would work.

adrianVmariano, 2020-01-25:
A key distinction I think between my proposal and what the code currently does is that mine will fail with some kind of error ("unable to distance map polygons"), whereas the existing code produces a bizarre, unpredictable mapping in the two ellipse example above.  So it fails condition (B) above in that the output is not predictable and not good looking.    

Note that the existing distance map does produce a good looking result for mapping a pointed square to a rounded corner square.  

I'm not sure how you handle this stuff internally, but it seems like a clean way to do it, if we want to have other mappings, would be for the alternative mapping to add points to the polygon.  Then linear interpolation to produce slices is simple.  Otherwise it seems like there's a messy situation, with slicing code that is different in some cases than in others.

adrianVmariano, 2020-01-25:
I should have split my conditions into three conditions:  (A) usable, (B) predictable (C) good looking result.

adrianVmariano, 2020-01-25:
For the two ellipse case the distance algorithm I proposed simply fails.  It will not map anything to the ends of the destination ellipse.  This seems to suggest that a broadly applicable algorithm needs to be more global than my proposal, which is what I feared.

adrianVmariano, 2020-01-26:
I got this idea that the distance alignment could be done as a quadratic programming problem, which is O(NM) to align length N and length M paths.  But then it would need to be done with every different index shift, I think, which means it would end up as O(N^2*M) where N>M.  Not sure if that's worth implementing.

adrianVmariano, 2020-01-27:
Looks like it's still too slow.  I wrote the alignment code and it takes 40s to do a single index shift case for a 128 pt and 32 pt ellipse.  The full method would be 128 times slower when it tries every index shift of the larger polygon.  

So I guess this leaves a question about whether there's some utility in a more limited (local?) algorithm such as the current one, and if it can be clear when it is likely to work (the rounded square case) vs when it produces garbage (the two ellipse case).

adrianVmariano, 2020-01-28:
Actually I realized something else about this.  The distance alignment concept only makes sense if at least one of the shapes is "discrete".   It seems like this "discrete" vs "continuous" notion is fairly important in determining how shapes should be connected.   I ran the full code to align a 128 point ellipse to a 32 point ellipse (trying all reindexings) and after optimizing my code, this ran in about 7 minutes.  But the results is bad because it collects groups of vertices together and maps them to the same point (which of course it does) but in this continuous context, that's not what makies sense.  The mapping of ellipse to ellipse should never map two points onto one point.   (So the distance mapping problem for this case would need to be cast as a continuous problem where you resample the points on one of the polygons.  I have no idea how that could be implemented algorithmically.)

My code is actually fast (enough) and usable if you want to find a minimum distance mapping between polygons with low vertex count.

adrianVmariano, 2020-01-28:
I wonder if it makes sense to use the distance code instead of the repeat_entries call to align polygons by vertex.   It would produce better results in many cases, though there is the problem of how to chain the alignments together...and possible long run times if the input has a high vertex count.

closed 2020-02-14 by adrianVmariano