BelfrySCAD/BOSL2 #119: rename trace_polyline

adrianVmariano, 2020-02-02:
Consider renameing trace_polyline to stroke3d.  Also change the size arg to width to match stroke.   (Maybe also pline arg should be path?)

revarbat, 2020-03-09:
I enabled use of `stroke()` for 3D lines as well as 2D lines.  Implemented in b6f39ce77b8765bcdc009fcdb2969ee4b0cd8b54
I think I'll leave `trace_polyline()`, though, as it's mostly for debugging purposes, with features for showing bezier controls and such.

revarbat, 2020-03-09:
Thinking about moving `trace_polyline()` to debug.scad, and maybe renaming it to `debug_path()` or `trace_path()`.

Also, changing all references to "polyline" to "path" or "path3d".

adrianVmariano, 2020-03-09:
Is trace_polyline necessary now that you have stroke working for 3d paths?

revarbat, 2020-03-09:
`trace_polyline()` is specialized to do things like show the control points of a bezier, or the pathpoints in a polyline.

closed 2020-04-10 by revarbat