BelfrySCAD/BOSL2 #122: add "face" or "edge" anchor to regular_ngon

adrianVmariano, 2020-02-12:
There should be an anchor that points outward from the center of the first edge.  I think it may need to be a named anchor.  I am not sure if it can be cleanly handled by a general method.  Having it be the first edge also means the location of the first edge is well-defined.  

I never quite understood why these start indexing at (0,1) so they are incompatible with the output of circle.  Is there intent there?

revarbat, 2020-04-10:
The starting vertex being towards Y+ seems like the natural orientation I would expect.  If I make a `pentagon()`, or a 5 pointed `star()`, I kind of expect one vertex on the Y+ axis line.

adrianVmariano, 2020-04-10:
There's a sort of conflict here, I think, between what is natural and what is consistent.  I'd say that what seems natural to me is to have a face down.  For n odd that means a vertex on the y axis.  For n even, no.

But if you try to combine regular_ngon output with circle then things are misaligned, which seems bad.  It seems like it would be better to make regular_ngon be consistent with circle and allow face and corner anchors to allow the user to position things as desired.

revarbat, 2020-04-10:
Well, if the ngons are natural, why not just change `circle()`, `square()`, et al to be consistent with that?

revarbat, 2020-04-10:
I agree about even vs odd count ngons, BTW.

adrianVmariano, 2020-04-11:
Yeah, so as it stands, only the odd ngons are natural.  

I think that because angle 0 is the x axis people will expect circles to start there.  So having them start on the y+ axis is a surprise.  I know I was surprised when I discovered that ngons did that.  If I can do something like `regular_ngon(....,anchor=EDGE)` or something like that to get an edge down then the natural form is easy to get (and works for both odd and even n) and mathematical consistency is retained for the default, so skinning with a circle works, for example, without having to make an adjustment.  

Is there any other shape that doesn't start on the x axis?  The supershapes start on the x axis...though I notice they go counterclockwise, which probably needs to be changed.   It's sort of annoying that the OpenSCAD standard is backwards.

revarbat, 2020-04-11:
I've double-checked, and `polyhedron()` expects clockwise faces from the outside viewpoint.  I figure that means that CW should be the consistent standard for 2D polygons, though `polygon()` itself doesn't care.

revarbat, 2020-04-11:
We currently have a `realign=` argument for `regular_ngon()`, `pentagon()`, etc. but it only rotates the shape by exactly half a face.  However, this is not sufficient to get a "natural" ngon.  Maybe if we had an argument that took a vector pointing the way you want the starting index, and default it to RIGHT.

adrianVmariano, 2020-04-11:
That's not good enough because it doesn't give a clean way to orient an edge in a desired direction, such as to put a decagon flat side down.  (And in other cases where you can do it the intent "put an edge down" isn't clear from code that says "point this point to the right".)    

What about the idea of having named anchors of "edge0", "edge1", "edge2", etc?   "vertex0", "vertex1", etc.    Hmmm.  I realized there's a complication I completely forgot about with rounded corners.   It seems like this creates a situation where the first point of the shape really should be the one at the vertex, which is in the middle of the roundover.  

What if I want attach shapes to certain edges of a polygon?    Is there a better solution than the above named anchor system?  

And will there also be some generalized 2d anchor system that might work for supershape using that anchor code I wrote a while ago?

adrianVmariano, 2020-04-11:
Oh, I'm not sure that polyhedron exactly "expects" clockwise.  That is, it seems that it likes clockwise, and shapes look yellow in "thrown together" but if you do everything ccw then CGAL is happy and everything works.  It's just a problem when you mix them.  But it seems like we should be as consistent as possible in how the paths are produced, so all shapes have the same orientation and start on the same axis rather than some going one way and some going another way.

adrianVmariano, 2020-04-11:
The behavior of `offset` by the way, is orientation dependent, and shapes have to be clockwise for it to match the way offset works on geometry.

revarbat, 2020-04-11:
I think ngons need the named face and vertex anchors anyways, but that doesn't give you an immediate way to get natural ngons.

revarbat, 2020-04-11:
`spin=90` would rotate unanchored odd ngons to "natural".  Works poorly with anchors, though.  Maybe `first_edge=BACK`/`first_vertex=BACK` as mutually exclusive args?

And supershape is already using path extent anchoring.

adrianVmariano, 2020-04-11:
Why wouldn't `anchor="edge0"` give a natural result?

adrianVmariano, 2020-04-11:
Or indeed any edge you happen to pick?

revarbat, 2020-04-11:
anchors don't chang the orientation at all.  It's a pure translation of the selected point to the origin.

revarbat, 2020-04-11:
Well, except when used with `attach()`, which adds rotation on top of the translation.

adrianVmariano, 2020-04-11:
Hmmm.  I guess I was confused because of the attach() situation.    Add an `edgedown` argument?

adrianVmariano, 2020-04-11:
Or would that be `edgefront`?

I have to admit I get confused about having to use front and back when working in 2d.

revarbat, 2020-04-11:
I think a more flexible solution would be `align_edge=FWD`.  If given, it rotates so that the center of an edge is facing towards that vector, and the first vertex is always the closest clockwise vertex to X+.

adrianVmariano, 2020-04-11:
I actually think that the aligned edge should be edge 0, so you know where the edges are.   How does this work with `orient`?

adrianVmariano, 2020-04-11:
Would some special arg to `orient` make sense?

revarbat, 2020-04-11:
`orient=` isn't really used with 2D shapes.

adrianVmariano, 2020-04-11:
BTW, your spec above is ill-defined for a bunch of cases when there are two equally close vertices to the X+ axis.  But it seems like that spec, if somehow nailed down, is going to be confusing.  Making the aligned edge edge0 is clear and also consistent as n varies.

adrianVmariano, 2020-04-11:
Ah, because you have `spin`?

adrianVmariano, 2020-04-11:
So if you have `align_edge=FWD` is there also `align_vertex=RIGHT`?

revarbat, 2020-04-11:
First vertex clockwise of the X+ axis, is what I meant, with vertexes clockwise from there.
If you want to skin a shape with an edge aligned to BACK, you don't necessarily want the first vertex to be near the back.

revarbat, 2020-04-11:
A mutually exclusive `align_edge=` and `align_vertex=` could work well.

revarbat, 2020-04-11:
If we implement these, then `realign=` is not really needed.

adrianVmariano, 2020-04-11:
Realign just turns by enough to move a face to a point, right?  

What's the argument for first vertex near the X+ axis?  It means it's hard to right a general program that attaches something to a known edge.  In practice it would probably mean for me that I would make the model, look at it to see where the edges were, count, and then I'd know the identity of the edge I needed to work with.    I mean, the more flexible way would be to specify which edge is aligned where, though that's more complex.

revarbat, 2020-04-14:
I think that `zrot()` and `realign=` takes care of all rotation cases.  Still need to do edge and vertex anchors, though.

adrianVmariano, 2020-04-14:
I thought we were getting rid of realign?

So are you planning to do align_edge and align_vertex?   Or are you saying that *all* cases are handled manually with zrot?   Not sure I understand.   If you do what you were talking about before, that align_edge puts an edge down (parallel to the x axis) such that the polygon starts on or just after the X+ axis then to place edge0 down so that I know where it is I need to zrot by something which is not easy to compute.  Looks like it would be zrot(360/n*floor((90*n-180)/360).  And even if you only want to do the case for a single n....well let's just say I drew a pentagon and my diagram was bad and it tricked me, so I got the wrong answer.   So I think there needs to be a way to align edge0 with the x axis that doesn't require the user to figure out the magic rotation to do it.

revarbat, 2020-04-17:
Err, this seems simple enough.  If you need to align an edge to an axis, use `realign=true`, and if you need to align a vertex to an axis, then use `realign=false`.  Then you can rotate that edge or vertex to whichever axis you want.  To make a pentagon have the forward side parallel to the X axis, `zrot(-90) pentagon(d=10,realign=true);`

revarbat, 2020-04-17:
Admittedly, segment0 of an ngon with `realign=true` will be the segment after the X+ axis, clockwise.  I could change it so that segment0 is the one centered on X+.  Pro: you know just where the first segment is, and it's aligned.  Con: this violates the clockwise from X+ rule.

adrianVmariano, 2020-04-17:
It seems to me that ditching realign but providing a mechanism that points edge0 or vertex0 in a known direction is a better solution.   The clockwise from x+ rule applies otherwise.  

I suppose letting spin take text arguments of some kind might be another option (vs what I suggested before).  So `spin="edgefwd"` or `spin="vertexfwd"` perhaps?

revarbat, 2020-04-19:
I'm going to think a bit on named spins.  There may be something interesting there.

adrianVmariano, 2020-10-22:
This has been fixed by the new polygon alignment stuff.

closed 2020-10-22 by adrianVmariano