BelfrySCAD/BOSL2 #127: more robust is_path and is_vector?  is_matrix?

adrianVmariano, 2020-02-28:
I'm wondering if is_path should call array_dim so it makes more of an effort to check that the data is actually a path.  Or perhaps instead of is_path we should have path_dim that returns the length and dimension of the path, and undef or empty list or something if it's not compliant.   Or it could assert() out if the input isn't a path (maybe optionally).   Or it could fail if the path dimension isn't the required one.

The main issue I see with this is that array_dim is kind of slow, so the question is whether that would ever actually be a problem.   Run time on a length 10^5 path on my machine is 0.7s, so pretty slow.   I tested a non-recursive path_dim routine and it was quite a bit faster at 0.2s, though still arguably sort of slow.  I guess the question is whether these runs times would be noticeable compared to any interesting operation that would actually get run on the path afterwords.    I also tried running is_num on all the entries.  That drove the run time up to 2.0s.   It would be nice to trap paths that contain [undef,undef,undef], for example, because they fail later with cryptic errors.  But that's really slow.

adrianVmariano, 2020-02-28:
As another point of reference, running path3d on 10^5 points takes 0.75s.

RonaldoCMP, 2020-03-01:
I have been using a bunch of test functions like those but a little more discriminating. Here are some of them:

function is_3Dvec(v)      = 0 * v==[0,0,0] ;
function is_2Dvec(v)      = 0 * v==[0,0] ;
function is_3Dlist(v,n)   = is_list_of(v,[0,0,0],n);
function is_2Dlist(v,n)   = is_list_of(v,[0,0],n);
function is_num_list(v,n) = is_list_of(v,0,n);
// check if the elements of v have the same numerical pattern 
// and structure of 'pat'
function is_list_of(v,pat,n) = 
  is_list(v) 
  && ( is_undef(n) || (is_num(n) && len(v)==n) )
  && [for( i=0, pat=0 * pat, vi=0 * v[i], va=pat; 
         i<len(v) && pat==va ;
           i=i+1, va=vi, vi=0 * v[i] ) if( pat!=vi) 0 ]==[];
  
v = [ [2,[3]], [3,[4]], [4,[5]] ]; 
echo(is_list_of(v,[1,[2]]));            // ECHO: true
echo(is_list_of(v,[1,[2]],2));          // ECHO: false
w = [ [2,[3]], [3,[4]], [4,["a"]], [5,[8]] ];
echo(is_list_of(w,[1,[2]]));            // ECHO: false
u = [ [2,[2==3]], [3,[undef]], [4,["a"]], [5,[false]] ];
echo(is_list_of(u,[1,[undef]]));        // ECHO: true
t = [ [1], [2], [3] ];
echo(is_list_of(t,1));                  // ECHO: false
echo(is_list_of(t,[1]));                // ECHO: true
echo(is_list_of(t,[[1]]));              // ECHO: false

adrianVmariano, 2020-03-02:
I can't think of a case that requires complex structures such as lists of the form [x,[y]], but it's definitely an interesting method.  I did a speed comparison of your method compared to my best approach so far.   My best method that tests that all the entries are numeric took 0.51 s on a 10^5 length input.  A test based on your method took 0.11 s.  Huge speed advantage.  

Note that I tried testing your actual code and it took 0.4 s.  I didn't try to figure out why exactly your code was slower, but I previously observed that the C-style for loops are very slow.  I tried one for this application, thinking that I could accumulate the boolean value and quit early if input was bad, but it was much slower than just checking all cases with a for loop that outputs the bad indices.  

Based on this it seems like there's no reason to keep the existing hacks that just check the first entry.  The question is what tests exactly do we want.

adrianVmariano, 2020-03-02:
I suppose the one possible downside of this approach is that inf and nan test as invalid.   That might be desirable but also it might not be.

RonaldoCMP, 2020-03-02:
You are right. The code with the usual 'for' is faster then with 'C-like
for':

function is_list_of(v,pat,n) =
  is_list(v)
  && ( is_undef(n) || (is_num(n) && len(v)==n) )
  && [let(pat=0*pat) for(vi=v) if(pat!=0*vi) 0]==[];

I was not aware of that.

Sometimes, I use some more complex structure than simple lists or lists of
lists and the general check is handy. For instance, it may be more
practical and legible to have a list of aggregates with two 3D points than
of a list of 6 numbers.

adrianVmariano, 2020-03-02:
When I was trying to think of cases with complex structure I had in mind in the existing BOSL2 code.  I'm sure it's possible to devise some situation where such a structure is handy.  

It looks like is_list_of is going to go in and is_path and is_vector will call it.

RonaldoCMP, 2020-03-02:
Nice study. I use 'C-like for' when a for loop is infeasible, the recursion would be more complex and/or I suppose the loop can be exited earlier. I will avoid to use of 'C-like for' with long lists anyway.

Allow me draw your attention to another derivation of is_list_of(). 

function is_homogeneous(v,n) = is_list_of(v, v[0],n);

It checks whether all list components have the same pattern/structure. It is useful for instance as a parameter check for carefully written Bezier functions wherein the operations of OpenSCAD scalar product and sum are explored as operators of a vector algebra. A simpler example is:

function sum(v) =
  assert( is_homogeneous(v) , "function sum(): the input list is not homogeneous")
  _sum(v,v[0]*0);

function _sum(v, tot, i=0) =
  i>=len(v) ?
    tot
  : _sum(v, tot+v[i], i+1);

echo( sum([ [ [1,2], [2,3] ], [ [4,5], [5,6] ] ] )); // ECHO: [[5, 7], [7, 9]]
echo( sum([ [ [1,2], [2, undef] ], [ [4,5], [5,6] ] ] )); 
// ERROR: Assertion 'is_homogeneous(v)' 
// failed: "function sum(): the input list is not homogeneous" in file libraries, line 4

adrianVmariano, 2020-03-02:
I implemented something like is_homogeneous for is_path in BOSL2, where a "path" is defined to be a list of vectors of the same length.  But you point out some broader applicability.  And in particular, of great value, the use of this technique for setting the base case for example in sum.  We recently ran into trouble with that and overlooked this elegant solution.

RonaldoCMP, 2020-03-02:
Note that, depending on the application, we should also check the first
element to see whether it has the appropriate structure.

revarbat, 2020-03-03:
This has been implemented.

closed 2020-03-03 by revarbat