BelfrySCAD/BOSL2 #143: [BUG] zdistribute

lar3ry, 2020-03-19:
When I use zdistribute(), the spaces between the objects see to vary depending on the size of the object.

// Code goes here.
include<BOSL2/std.scad>
zdistribute (sizes = [20,5,50],spacing = 10) {
	sphere(20);
	cube(5,center=true);
	cube(50,center=true);
}

**Expected behavior**
I expected the distance between each object to be 10mm

**Screenshots**
Produced by the above code:
[image]

**Additional context**
Add any other context about the problem here.
- OpenSCAD Version: 2019.o5
- Other libraries used: None

revarbat, 2020-03-23:
Hmm.  The default argument for `sphere()` is `r=`, not `d=`, which makes the diameter of that sphere 40, while you told `zdistribute()` it is 20.  So, this is not a bug in the code, but a misunderstanding of `sphere()`.  Behavior verified correct:
```openscad
zdistribute (sizes = [20,5,50],spacing = 10) {
	sphere(d=20);
	cube(5,center=true);
	cube(50,center=true);
}
```
[image]

closed 2020-03-23 by revarbat

lar3ry, 2020-03-23:
> Hmm. The default argument for `sphere()` is `r=`, not `d=`, which makes the diameter of that sphere 40, while you told `zdistribute()` it is 20. So, this is not a bug in the code, but a misunderstanding of `sphere()`. Behavior verified correct:

> zdistribute (sizes = [20,5,50],spacing = 10) {
> 	sphere(d=20);
> 	cube(5,center=true);
> 	cube(50,center=true);
> }

Ah yes. Sorry about that. However, the reason I was investigating this is that zdistribute definitely did not work as expected in the script I was writing. Turns out it fails when I use tube();

Heres an example
include<BOSL2/std.scad>

zdistribute(
	sizes=[25,20,20,10,20],spacing=5)
		{
			tube(h=25, od=50.5, id=28);
			sphere(10);
			cube(20,center = true);
			sphere(5);
			tube(h=20, od=50.5, id=28);
		}
[image]

revarbat, 2020-03-23:
Add `center=true`.  There is no way for the distribute functions to know if a child is off-center.
[image]

revarbat, 2020-03-23:
Add `center=true` to the `tube()` and `cylinder()` calls, that is.

lar3ry, 2020-03-23:
> Add `center=true` to the `tube()` and `cylinder()` calls, that is.

Thanks!
Need any help with the Wiki?

revarbat, 2020-03-23:
The Wiki is currently, except for the Home, Index, and _Sidebar pages, auto-generate from the documentation comments in the code.  If you find errors or unclear sections of those documents, or examples and images that need work, you can make the changes in the comments of your own fork of the code, and file a PR with the changes, or file an issue, and Adrian or I can fix them, and regenerate the pages.

I do want to add a number of tutorial pages, as well.  I'm currently working on a syntax and a script to generate images for a standardly formatted tutorial markdown page.