BelfrySCAD/BOSL2 #148: [BUG] Function gear2d use of attachable causes warnings and gears are not drawn.

kingadami, 2020-03-30:
**Describe the bug**
The function gear2d tries to use the module attachable which causes the following warning is openscad:
```
WARNING: Ignoring unknown function 'attachable', in file ../.local/share/OpenSCAD/libraries/BOSL2/involute_gears.scad, line 281.
```
As far as I can tell because of this warning, the gear is not being drawn.

While functional programming languages are not my strong point, and I am relatively new to OpenSCAD, I think you just need to change `attachable` to `reorient` and appear to get the intended behavior.  After making this change I get the gear as expected.

**Code To Reproduce Bug**
```
gear(pitch=5, teeth=50, shaft_diam=5);
```

**Expected behavior**
I would expect the gear to be drawn.

**Additional context**
- OpenSCAD Version: 2019.05
- Other libraries used: N/A.

kingadami, 2020-03-30:
Assuming that just changing the attachable to reorient is the correct answer, I would like to try to fix it and do a pull request if that is OK with you.

kingadami, 2020-03-30:
One more item which I forgot and is related to getting the module gear2d to work is that OpenSCAD complains with the following warning and error:
```
WARNING: Ignoring unknown variable 'pr', in file ../.local/share/OpenSCAD/libraries/BOSL2/involute_gears.scad, line 304.
ERROR: Assertion 'false' failed: "Unrecognizable geometry description." in file ../.local/share/OpenSCAD/libraries/BOSL2/attachments.scad, line 239
...
```
To fix this I added the following line to the gear2d module:
```
pr = pitch_radius(pitch=pitch, teeth=teeth);
```
I don't know if this is the intended fix or not, but it did address the warning and allowed the gear to be drawn.  Let me know.

revarbat, 2020-03-30:
Your analysis was right on the mark.  Error introduced in the attachable refactoring.
Fixed in 0a741b36d1c9cea34ebf917124f515d4a942abda

closed 2020-03-30 by revarbat