BelfrySCAD/BOSL2 #151: [BUG]

RonaldoCMP, 2020-04-05:
Function triangulate_face() in triangulation.scad fails when the face is not a simple polygon. For instance:
```
p = [ [0,0,0], [1,0,0], [1,1,0], [0.5,-1,0],[0,1,0] ];
echo(triangulate_face(p,[0,1,2,3,4]));
```
WARNING: Invalid type of parameters for cross(), in file triangulation.scad, line 148

For this example, find_convex_vertex() can't find a convex face vertex and returns an undef value that is an operand of a cross().

We have two possibility to overcome the issue:
1) check the returned find_convex_vertex() value and call an assert if it is undef;
2) the triangulate_face() returns an undef value when the check fails as a signal that the triangulation is impossible because the face is not simple.

The later alternative would be interesting as it allows the caller function to identify the issue and take an appropriate action. Ideally those two alternative might be optional to the caller function by an additional parameter whose default to the first alternative.

RonaldoCMP, 2020-04-05:
There is another issue with triangulate_face(). It fails for some simple faces like:

```
p = [ [0,0,0],   [1,0,0],     [1,1,0],     [0.7,0.5,0], 
      [0.6,1,0], [0.5,0.1,0], [0.4,0.5,0], [0,1,0] ];
echo(triangulate_face(p,[0,1,2,3,4])); // ECHO: [[1, 2, 3], [3, 4, 0], [0, 1, 3]]
t = triangulate_face(p,[0,1,2,3,4]);
translate([2,0,0]) polygon([for(pi=p) [pi.x,pi.y]]);
for(ti=t) 
   color(ti/7)
     polygon([for(i=ti) [p[i].x,p[i].y]]);
```
The triangulation in this example doesn't refers to the vertices 5 an 6.

revarbat, 2020-04-06:
It should pretty definitely assert an error if it finds a self-crossing face perimeter.  Returning undef, while in theory allowing more flexibility, in practice tends to hide problems, and propagate bad data to places that make more cryptic errors.

RonaldoCMP, 2020-04-06:
I agree. An option would be to return an empty vnf when the face is not simple and a parameter 'check' is true. If 'check' is false (the default), an error would be asserted. The objective here is to avoid a previous simple face test before the triangulation.

revarbat, 2020-04-07:
Regarding the second example given, that is behaving correctly.  It never references vertices 5 and 6 because you never gave it a face with them in it.

```openscad

revarbat, 2020-04-07:
Added the assert at this time.

closed 2020-04-08 by RonaldoCMP