BelfrySCAD/BOSL2 #153: Rectangular tube?

lar3ry, 2020-04-10:
I like BOSL2 a LOT.  I am starting to open a new project and typing in the include for std.scad before starting on the object. No actual problem, just a 'would be nice'.

Unless I just haven't looked hard enough, I would like to suggest that a good enhancement would be a rectangular tube. Of course, I would like to also have parameters for chamfer and rounding for at least the outsides of the tube, but I have no idea how much more work that would be.

revarbat, 2020-04-10:
Well, at the moment, you can do a rounded rectangular tube like this:
```openscad
size = [40,50,60]; r=5; wall=10; $fn=24;
diff("hole")
cuboid(size, rounding=r) {
	cuboid(size-[2*wall,2*wall,-0.1],rounding=-r,$tags="hole");
}
```

revarbat, 2020-04-10:
Playing with this, I seem to have found a minor bug where a rounded `cuboid()` is ever so slightly larger than it should be.

revarbat, 2020-04-10:
Fixed `cuboid()` with rounding being slightly oversized.  0cdc94cc0a8c7a14eb00a989c0d543aa6aae3425

revarbat, 2020-04-10:
Alternatively, using the old non-attachment method:
```openscad
size = [40,50,60]; ir=2; or=5; wall=10; $fn=24;
difference() {
	cuboid(size, rounding=or);
	cuboid(size-[2*wall,2*wall,-0.01],rounding=-ir);
}
```

adrianVmariano, 2020-04-11:
I have waffled about whether it would be useful to have a function that makes boxes, which I do frequently.  The main problem with writing a function to do it is that it doesn't really simplify the code very much and you have a ton of parameters, with rounding inside and outside at the top and bottom.  

On the other hand, `tube` exists.  I suppose you could do $fn=4 to get a square tube.  But does the existence of tube support creating these other modules as well?

revarbat, 2020-04-11:
I've been waffling about whether or not we need `tube()`.

revarbat, 2020-04-11:
It wouldn't be hard to make a `rect_tube()` module. The number of arguments you'd need to specify rounding/chamferring inside and out, though seems like a bit much effort, given how easy it is to do without said module.  Maybe a simple `rect_tube()` without rounding or chamfering?

revarbat, 2020-04-11:
Eh, I guess it'd be nice to do something like:
```openscad
rect_tube([40,50,60], wall=10, irounding=2, orounding=5, $fn=24);
```
If we did do `rect_tube()` with rounding, we'd pretty much HAVE to make `tube()` handle rounding and chamfering for consistency.  I don't think we need per-edge specs, though.  Just per-tube-end, inside and outside.

lar3ry, 2020-04-11:
I do like tube() as it is now, but I think I would ge more use out of a rect_tube() simply because I would cut down on the fiddling with calculating the translattions to position an inner rectangle, and especially when making a box.

I think that inner and top/bottom rounding would be far less useful than outer rounding of the side corners of the rectangle. 

I would make a box someththing this. It would produce a box with default rounding only on edges aligned with the Z axis.

union() {
rect_tube([40,50,60], wall=5, rounding=5, $fn=24);
cuboid([40,50,5]);
}

lar3ry, 2020-04-11:
BTW, when I said I invoked BOSL2 on a project, I was not speaking of a project I was just starting, but rather as a first step in any projects. Love the shapes, axis directions, distribute, etc.

adrianVmariano, 2020-04-11:
Hmmm.  See, I think rounding the top edges of a box is really important because that's often where I'm most likely to touch it.  And if I pick it up I'll touch the bottom outside edges.  And if I want to clean it I want the inside bottom to be rounded.  Really there's no justification for any edge not being rounded.  I make boxes a lot, and I typically make them like this:

```
include<BOSL2/std.scad>
include<BOSL2/rounding.scad>

height = 34;
wall = 2;
inwidth = 60+wall;
inlength = 70+wall;

base = round_corners(square([inwidth, inlength]), size=2, $fn=64);

difference(){
  offset_sweep(base, bottom=os_teardrop(r=2), top=os_circle(r=1), height=height);
  up(wall)
    offset_sweep(offset(base,r=-wall,closed=true), height=height-wall, bottom = os_circle(r=2), top = os_circle(r=-1,extra=1), steps=8);   
}
```

This one is for toothpicks.  (The code is not clean because I originally tried to make it by starting with the inside and offsetting outward, which turns out not to work.  It produces a valid looking model and then my slicer fails on it.  I'm still trying to figure out exactly why that happens.)   

I used the fancier `offset_sweep` module instead of cuboid because I wanted teardrop rounding on the outside to ensure printability.  But the same approach would work with `cuboid`.

adrianVmariano, 2020-04-11:
Oh, also in your hypothetical example if rect_tube rounds the outside corners then you need to add rounding arguments to cuboid.  

Also I notice you said something about "calculating translations" which is puzzling to me:  just have the inside and outside both centered.  Nothing to calculate.  The only calculation is the size of the inside vs outside, unless you use offset().  If you're making a rounded box you have to calculate the length of the inner section, as it cannot be oversized.

lar3ry, 2020-04-11:
> Hmmm. See, I think rounding the top edges of a box is really important because that's often where I'm most likely to touch it. And if I pick it up I'll touch the bottom outside edges. And if I want [...]

Good points. Inside rounding is something I don't use as often as square insides, as most of my boxes are for cases or parts drawers, so it just hasn't been a priority for me

> I used the fancier `offset_sweep` module instead of cuboid because I wanted teardrop rounding on the outside to ensure printability. But the same approach would work with `cuboid`.

sweep is something I've been seeing a lot lately, but I have yet to figre it out. I just now discovered the index on the Wiki, so I have a lot to study now.

> Oh, also in your hypothetical example if rect_tube rounds the outside corners then you need to add rounding arguments to cuboid.

True. Forgot to put that in. 

> Also I notice you said something about "calculating translations" which is puzzling to me: just have the inside and outside both centered. Nothing to calculate. The only calculation is the size of [...]

You know, it may sound strange to you, but I actually never thought of doing it that way. I have always just used the default origin, then translated and resized the inner object. I'll do it that way from now on. I do have to do one translate, of course, to raise the inner box by the 'floor' thickness, but it's a lot easier than what I do now.

Is there a preferred place to just ask questions about BOSL2? I hesitate to use github, because questions don't seem to be either bugs or enhancement requests.

adrianVmariano, 2020-04-11:
I might actually want a more aggressive interior roundover for parts drawers to make it easier to scoop out the parts and to prevent dust from sticking in the corners.  

Note that offset_sweep is not the same as general sweep.  As far as I know, nobody else has ever implemented offset sweep, but general sweep, of forming 3d objects by a sequence of transforms of a cross section, has been implemented over and over by many folks.  We have general sweep stuff in skin.scad right now (probably needs a rename).  And offset_sweep is in rounding.scad because I originally wrote it specifically for making rounded boxes.  

I don't understand what you mean about using the default origin and translating the inner object for box making.  If you use cuboid, the objects are centered on the origin by default, so no action is needed to line the inner part up with the outer.  I guess if you use cube instead then you would need a translate for alignment other than raising it to create the floor.  

Revar has a chat that we use to talk about development and issues.  Maybe that would be a suitable place?

lar3ry, 2020-04-11:
> I might actually want a more aggressive interior roundover for parts drawers to make it easier to scoop out the parts and to prevent dust from sticking in the corners.

I haven't bothered with most of the drawers I have made, primarily because of the effort involved. Now, with BOSL2, I think I will be finding things a lot easier. For example, when I designed a parametric box, and allowed for rounded sides and top, I used a rather brute force method. You can see it if interested, at https://www.thingiverse.com/thing:2739499/files
 
> Note that offset_sweep is not the same as general sweep. As far as I know, nobody else has ever implemented offset sweep, but general sweep, of forming 3d objects by a sequence of transforms of a [...]

I'll have a look and see if I can understand it. 

> I don't understand what you mean about using the default origin and translating the inner object for box making. If you use cuboid, the objects are centered on the origin by default, so no action is [...]

Right. And, as above, I'll be trying to do things better, using BOSL2.

> Revar has a chat that we use to talk about development and issues. Maybe that would be a suitable place?

If you don't mind me joining in, sure.

adrianVmariano, 2020-04-11:
My first big project I wanted to round things over and it was a big struggle, so I've been really interested in how to make tools to make that easier/possible.  You should hopefully find offset_sweep easy to understand and use.  If you don't, we should try to figure out how to make things better.  

I think it would be advantageous to have a third voice in the chat because a lot of the time we are talking about issues of what would be natural for users or what would make sense to users, what kind of interface would be good.  And the two of us may not have a broad enough perspective to find the best answers.   Also Revar is starting to write tutorials, and if you ask questions it may help us understand what things need more explanation.

revarbat, 2020-04-11:
I've sent the Gitter invite, but if you have issues with that, you should be able to just join the channel at https://gitter.im/revarbat/BOSL2

lar3ry, 2020-04-11:
Got it. Managed to get in with the link. Definite problems with the invite. Apparently I forgot to update my password safe when I last changed my github one.

revarbat, 2020-05-19:
Implemented basic `rect_tube()` in 4fab4f56c3c86a1e4a098779162697776b67013e.
Does not yet implement rounding or chamfer.

revarbat, 2020-05-19:
Implemented extensive chamfers and roundovers. 892e98e77339113ff96c1953e6842ea442385917

closed 2020-05-19 by revarbat