BelfrySCAD/BOSL2 #155: supershape is counterclockwise

adrianVmariano, 2020-04-11:
Supershape output should be clockwise for consistency.

revarbat, 2020-04-11:
This is definitely tied to the orientation decision in #154

revarbat, 2020-04-11:
Related to #154

revarbat, 2020-04-12:
`glued_circles()` also appears to be CCW.

revarbat, 2020-04-14:
Actually, I verify `supershape()` as already being clockwise.  However, it starts somewhat clockwise of X+, instead of on X+.

adrianVmariano, 2020-04-14:
Uh, I'm confused:

```
s = supershape(step=0.5,m1=16,m2=16,n1=0.5,n2=0.5,n3=16,r=50);
echo(cc = polygon_is_clockwise(s));
echo(s[0]);
```

That was the code I wrote before posting this issue.  And the output I get:

```
ECHO: cc = false
ECHO: [31.8073, 0]
```
Output says it's NOT clockwise and it starts on the X+ axis.  

Does the direction it turns depend on parameter choice?

revarbat, 2020-04-14:
Huh.  I have no idea.  Maybe it does, because the `polygon_is_clockwise()` test I ran with it returned true.

revarbat, 2020-04-14:
```openscad
path = supershape(m1=4, n1=1/2, n2=1/2, n3=4);
stroke(path,width=0.01);
echo(polygon_is_clockwise(path));
// ECHO: true
```

revarbat, 2020-04-14:
Nevermind.  Examination shows that I already applied a fix for it in my code, and forgot I had.

adrianVmariano, 2020-04-14:
What about the business of not starting on the X+ axis?

revarbat, 2020-04-14:
That was a side effect of my original fix, and has been corrected.

revarbat, 2020-04-14:
`supershape()` and `glued_circles()` are both fixed for CW from X+ standardization, as is everything in shapes2d.scad.

closed 2020-04-14 by revarbat