BelfrySCAD/BOSL2 #162: Add log/exp of SE3 transformations

adrianVmariano, 2020-05-05:
Add code that computes log and exponentials of SE3 transformations.   This also relates to the `trajectory` function from list-comprehension-demos.

adrianVmariano, 2020-08-02:
This code is all written....but I'm just not sure if it's useful...or the right way to do things.  I was pondering adding powers of arbitrary transformations (not just SE3) using eigenvalue decomposition, but that's been quite difficult to implement.

revarbat, 2020-08-02:
I don't even have the faintest idea what this would be for.

adrianVmariano, 2020-08-02:
It lets you interpolate between transformations that belong to SE3, which is the set of rotations and translations.  So you can say I start at T1 and and at T2 and automatically insert 100 uniformly chosen transformations in between T1 and T2.  You could say it would be the equivalent of adding slices in skin but for generic sweep.  This could be a nice thing.  The problem is the lack of generality to handle other kinds of transformation.  

The basic reason that log/exp of transformations are involved in interpolation is that suppose you want to interpolate from the identity to T.  Well, since transformations multiply if you want N transformations it means you need S=T^(1/N).  Then the solution to the interpolation problem is S^0, S^1, S^2, ..., S^10=T.  So it's ten uniform steps that arrive at T at the end.   How the heck can you get T^(1/N)?  Well if log and exp are available you can do it from exp(log(T)/N).   So that's where the log and exp of transformations come in.  The problem is that it doesn't work on all transformations, so it's not generic.  There is a way to do this generically but the math is like 100 times harder to implement---the eigenvector decomposition.  Normally folks call LINPACK.  Nobody tries to code their own.   The "bible" of numerical analysis, "Matrix Computations" doesn't bother to give all the details, only explaining the simpler cases and leaving the more complex cases for you to look up in the original papers.  

Here's another interesting thing:  the log of a transformation in SE3 is actually just a vector of rotation angles and the translation vector.  Would it ever be useful to be able to take a rotation and extract its angle?  (I don't know the answer.)  

This is also connected to the 3d turtle, which is written, but it acts in ways that are not particularly intuitive.  And so it's not clear that it's really the right way to do a 3d turtle.

RonaldoCMP, 2020-08-07:
I have been studying dual quaternion as an alternative way to represent and process transformations in SE3. Dual quaternion is an extension of dual numbers to quaternions. With dual quaternions it is possible to do interpolations between transformations in SE3 the same way Slerp is an interpolation between two rotations represented by quaternions. I have no idea how that would compare with exp/log interpolation techniques but I would expect good performance as dual quaternions is commonly used by the people of real time animation. I am about to finish the debugging of a dual quaternion implementation and will start to test it against exp/log stuff.

adrianVmariano, 2020-08-07:
What is the advantage of this compared to working in SE3?  From what I have heard from Revar, it's much slower to use quaternions than matrices.   And the SE3 interpolation method seems to work well.

RonaldoCMP, 2020-08-08:
Dual quaternion is an alternative to exponential map but I can't say what
are the advantages or disadvantages yet. It is true that in some cases
quaternions can be slow as I reported in a now closed issue named Slerp.
That was the reason I have added Q_Rotation_path to quaternions.scad. The
function is intended to do spherical linear interpolations between two
quaternion rotations at a list of n uniformly distributed coefficients
returning the corresponding list of matrices. Instead of using the
generality of Q_Slerp to do it, the function computes the difference
quaternion to the power 1/n, convert it to a matrix and return the
cumulative product of that matrix by itself. This function has shown to be
faster than calling a Q_Splerp. That is an interpolation process in SO3.

Dual quaternions allows this process to be extended to SE3. The theory
involves dual numbers and quaternions and it is hard to say in advance what
will be its performance because it requires the cascade of a bunch of small
functions. Animation people defend its use to interpolate movements of
synthetic characters even in real time. They often work in high performance
languages. In OpenSCAD, operations with matrices are relatively faster
though. Only a test will show advantages and disadvantages.

adrianVmariano, 2020-10-22:
The new rot_decode() does this.

closed 2020-10-22 by adrianVmariano