BelfrySCAD/BOSL2 #167: [BUG] structs.scad does not parse

tcurdt, 2020-05-19:
**Describe the bug**
structs.scad does not parse (for me)

**Code To Reproduce Bug**

Just opening the file, this gives me a syntax error:
https://github.com/revarbat/BOSL2/blob/master/structs.scad#L108

**Additional context**

- OpenSCAD Version:  2019.12.20 (git 2a943a57)

tcurdt, 2020-05-19:
Also a problem with version 2020.05.17 (git b0dabf69)

tcurdt, 2020-05-19:
This seems to work OK:
```
function is_struct(struct, ind=0) = let(dim = array_dim(struct))
    is_list(struct) && (
	struct == [] ||	
	len(dim)==2 && dim[1]==2
    );
```

revarbat, 2020-05-19:
Argh.  Yes, OpenSCAD introduced a parsing bug recently that made having a `let()` immediately follow a logical operator like `||` or `&&` a syntax error, despite it being a valid expression prefix.  Putting the `let()` and its following expressions all in a paren `()` block should fix it.

revarbat, 2020-05-20:
Fixed in d20caf7d59ff419b1cd59749b5447b1c335e9983

closed 2020-05-20 by revarbat