BelfrySCAD/BOSL2 #174: document circle_point_tangents, remove "find" from find_circle functions

adrianVmariano, 2020-05-30:
The docs say that circle_point_tangents returns "ANG" but never says what this return value is.  (Is this return, whatever it is, necessary/useful?  If you have the point of tangency you can compute angles if needed.)  

The find_circle_3points and find_circle_2tangents might be renamed without "find".  Also they return arrays of undef in the collinear case.  Should they instead issue an error?

revarbat, 2020-09-25:
Agreed that the ANG is easily derived and just makes using `circle_point_tangents()` more complex.  Changing it to return just the list of tangent points.

revarbat, 2020-09-25:
Renames implemented in 07beb712a2686a7e4ec738cfd8867736bb77352e

closed 2020-09-25 by revarbat