BelfrySCAD/BOSL2 #180: stroke fails for 3d inputs

adrianVmariano, 2020-06-14:
include<BOSL2/std.scad>
    stroke([[0,0,0],[0,0,5]]);

fails with errors about path2 unknown.

lar3ry, 2020-06-14:
Works for me. No errors in preview or render.

adrianVmariano, 2020-06-14:
Are you using last night's revision?  That's when it broke.

lar3ry, 2020-06-14:
On 14 Jun 2020 at 15:10, adrianVmariano wrote:
> Are you using last night's revision? That's when it broke.

No. I had downloaded it a few days ago, and when I saw your posting "Latest 
update seems to have broken stroke for 3d inputs." I realized what had 
happened, and downloaded the latest today.

Sorry about that.

adrianVmariano, 2020-06-14:
On Sun, Jun 14, 2020 at 04:01:13PM -0700, lar3ry wrote:
> On 14 Jun 2020 at 15:10, adrianVmariano wrote:
>
> No. I had downloaded it a few days ago, and when I saw your posting "Latest
> update seems to have broken stroke for 3d inputs." I realized what had [...]

I'm not sure downloading the latest version is the right response to
learning that it's broken.

lar3ry, 2020-06-14:
On 14 Jun 2020 at 16:03, adrianVmariano wrote:
> I'm not sure downloading the latest version is the right response to
> learning that it's broken.

Well, I don't mind if it's broken. In this particular case, I don't (so far) use 
stroke() anyway, and if something else is also broken, I don't mind finding 
out. I do realize it's a work in progress.

Actually, from now on, I'll keep the old one when I download a new one.

adrianVmariano, 2020-06-14:
On Sun, Jun 14, 2020 at 04:08:44PM -0700, lar3ry wrote:
> 
> Actually, from now on, I'll keep the old one when I download a new one.

It's pretty easy to download old revisions if you need to.

revarbat, 2020-06-15:
Argh.  Fixed in 3b07cf210b79956fe72708eda7757d029e57583e

closed 2020-06-15 by revarbat