BelfrySCAD/BOSL2 #181: 3D Supershapes

CloudReadCoder, 2020-06-15:
Hello, 

first of all, kudos for the great work you're doing with BOSL/BOSL2, your library is a real life & time saver.

I've started playing with the supershape module/function, and trying to create 3D Supershapes (using the 3D formula from http://paulbourke.net/geometry/supershape/ ).

I've found an OpenSCAD implementation here https://www.thingiverse.com/thing:2703438 ,
and I'd like to implement this in a BOSL2 compatible & elegant way,  but I'm kind of getting lost because my understanding of OpenSCAD & BOSL2 isn't yet where I want it to be.

Are you guys by any chance working on such an implementation, or is there maybe already a procedure which could be fed 2 supershapes and return a polyhedron that I am not aware of?

My code here below spits out some kind of result as a lot of small cubes, with more or less the intended shape, but I'm kind of lost from there on.

sth.like:
```
overall_r=30;
sh1=supershape(8,m1=3,n1=1,r=1,spin=-90); //xy shape
sh2=supershape(8,m1=3,n1=1,r=1,spin=-90);  //zshape
supershape3d(sh1,sh2);

module supershape3d(shape1,shape2){
   pts=
   [ for (c1=shape1) 
       for (c2=shape2)
          let
          (x=c1[0]*c2[0]*overall_r,
           y=c1[1]*c2[0]*overall_r, 
           z=c2[1]*overall_r
           )
           [x,y,z]
   ]; 
   
  for (apt=pts)
    translate(apt)cube([1,1,1]);     
  
};   
```

CloudReadCoder, 2020-06-15:
Worked a bit on the example, using part of the code from the thingiverse code mentioned earlier, now creates polyhedron.

```
overall_r=30;
sh1=supershape(8,m1=3,n1=1,r=1,spin=0); //xy shape
sh2=supershape(8,m1=2,n1=1,r=1,spin=0);  //zshape

supershape3d(sh1,sh2);

module supershape3d(shape1,shape2){
    cnt=0;
   pts=
   [ for (c1=shape1) 
       for (c2=shape2)
          let
          (x=c1[0]*c2[0]*overall_r,
           y=c1[1]*c2[0]*overall_r, 
           z=c2[1]*overall_r
           )
           [x,y,z]
   
   ]; 
   
   parallels=len(shape1);
   meridians=len(shape2);    
   inilat=0;    
   fces = 
concat(
    [
    for(z = [0:parallels-2])
        for(s = [0:meridians-1])
            let(
                // clockwise from left-down corner
                f1 = s + meridians*z,
                f2 = s + meridians*(z+1),
                f3 = ((s+1) % meridians) + meridians*(z+1),
                f4 = ((s+1) % meridians) + meridians*z
            )
        
            [f1,f2,f3,f4]
 
    ],
    [[ for(s = [0:meridians-1]) ((inilat==0)?0:meridians)+s]], // bottom
    [[ for(s = [1:meridians]) meridians*(parallels)-s]] // top
 
) ;   
       
  polyhedron(points=pts,faces=fces);
    
   
}; 
```

revarbat, 2020-06-15:
‘supershape()` has been implemented for a while now:
https://github.com/revarbat/BOSL2/wiki/shapes2d.scad#supershape

CloudReadCoder, 2020-06-15:
Thx for your quick reply, supershape() gives a 2D path (which I am aware of), what I'm looking to achieve is a 3D object using 2 2Dsupershapes, one modulating the other to create "extrude" a complex object. Kind of linear_extrude without the linear.

The code I posted here above creates a polyhedron using 2 supershape() as input to create a 3D object, it kind of works, but doesn't exactly work as expected.

CloudReadCoder, 2020-06-15:
Fixed the code a bit, works a little better to illustrate:
```
overall_r=30;
//sh1=supershape(step=1,m1=16,m2=16,n1=0.5,n2=0.5,n3=16,r=1); //xy shape
//sh2=supershape(step=1,m1=16,m2=16,n1=0.5,n2=0.5,n3=16,r=1); //xy shape

sh1=supershape(1,m1=3,n1=1,r=1,spin=0);
sh2=supershape(1,m1=3,n1=1,r=1,spin=90);  //zshape

//translate([0,100,0])polygon(sh1);
//translate([0,-100,0])polygon(sh2);

supershape3d(sh1,sh2);

module supershape3d(shope1,shope2){
    cnt=0;
    shape1=concat(shope1,[shope1[0]]);
    shape2=concat(shope2,[shope2[0]]);
    parallels=len(shape1);
   meridians=len(shape2);   
    
    pts=
   [ for (i=[0:parallels-1]) 
       for (j=[0:meridians-1])
          let
          (c1=shape1[i],
           c2=shape2[j],
           x=c1[0]*c2[0]*overall_r,
           y=c1[1]*c2[0]*overall_r, 
           z=c2[1]*overall_r
           )
           [x,y,z]
   
   ]; 
   
   
   inilat=0;    
   fces = 
concat(
    [
    for(z = [0:parallels-2])
        for(s = [0:meridians-1])
            let(
                // clockwise from left-down corner
                f1 = s + meridians*z,
                f2 = s + meridians*(z+1),
                f3 = ((s+1) % meridians) + meridians*(z+1),
                f4 = ((s+1) % meridians) + meridians*z
            )
        
            [f1,f2,f3,f4]
 
    ],
    [[ for(s = [0:meridians-1]) ((inilat==0)?0:meridians)+s]], // bottom
    [[ for(s = [1:meridians]) meridians*(parallels)-s]] // top
 
) ;   
       
  polyhedron(points=pts,faces=fces);
    
   
};    

```

revarbat, 2020-06-15:
Sorry, I missed that you were asking about a 3D supershape.  Interesting.

revarbat, 2020-06-15:
```openscad

revarbat, 2020-06-15:
Here's my first pass BOSL2-ification:
```openscad
overall_r=30;
//sh1=supershape(step=1,m1=16,m2=16,n1=0.5,n2=0.5,n3=16,r=1); //xy shape
//sh2=supershape(step=1,m1=16,m2=16,n1=0.5,n2=0.5,n3=16,r=1); //xy shape
sh1=supershape(1,m1=3,n1=1,r=1,spin=0);
sh2=supershape(1,m1=3,n1=1,r=1,spin=90);  //zshape
supershape3d(sh1,sh2);

module supershape3d(shope1, shope2, anchor=CENTER, spin=0, orient=UP)
{
    cnt=0;
    shape1=close_path(shope1);
    shape2=close_path(shope2);
    parallels=len(shape1);
    meridians=len(shape2);
    pts = [
        for (i=[0:parallels-1], j=[0:meridians-1])
        let(
            c1=shape1[i],
            c2=shape2[j],
            x=c1[0]*c2[0]*overall_r,
            y=c1[1]*c2[0]*overall_r,
            z=c2[1]*overall_r
        ) [x,y,z]
    ];
    inilat = 0;
    fces = concat(
        [
            for(z=[0:parallels-2], s=[0:meridians-1])
                let(
                    // clockwise from left-down corner
                    f1 = s + meridians*z,
                    f2 = s + meridians*(z+1),
                    f3 = ((s+1) % meridians) + meridians*(z+1),
                    f4 = ((s+1) % meridians) + meridians*z
                ) [f1,f2,f3,f4]
        ],
        [[ for(s = [0:meridians-1]) ((inilat==0)?0:meridians)+s]], // bottom
        [[ for(s = [1:meridians]) meridians*(parallels)-s]] // top
    );
    vnf = [pts, fces];
    attachable(anchor,spin,orient, vnf=vnf) {
        vnf_polyhedron(vnf);
        children();
    }
};
```

CloudReadCoder, 2020-06-16:
Hy again, and thanks, I've added your code, except the close_path, and I had to extend & slightly modify the supershape() function, copied and renamed it to supershape2() it because I needed the ability to created only supershape from as exemple -45 to 45.

You may want to to run this and eventually add the changes in the supershape() function if you deem it reasonable and it doesn't break things.

```
sh1=supershape2(step=2,m1=2,n1=0.1,n2=2.5,n3=1.7,r=1,spin=0,ang_beg=0,ang_end=360);
sh2=supershape2(step=1,m1=40,n1=8,n2=40,n3=0,r=1,spin=0,ang_beg=-25,ang_end=45);  //zshape

//overall_r=30;
supershape3d(sh1,sh2,30);

function supershape2(step=0.5,m1=4,m2=undef,n1=1,n2=undef,n3=undef,a=1,b=undef,r=undef,d=undef,anchor=CENTER, spin=0,ang_beg=undef,ang_end=undef) =
    let(
        r = get_radius(r=r, d=d, dflt=undef),
        m2 = is_def(m2) ? m2 : m1,
        n2 = is_def(n2) ? n2 : n1,
        n3 = is_def(n3) ? n3 : n2,
        b = is_def(b) ? b : a,
        
        ang_beg = is_def(ang_beg) ? ang_beg : 0,
        ang_end = is_def(ang_end) ? ang_end : 360,
        ang_dif = ang_end-ang_beg,
         
        
        steps = ceil(ang_dif/step),
        step = ang_dif/steps,
        angs = [for (i = [0:steps]) step*i+ang_beg],
        rads = [for (theta = angs) _superformula(theta=theta,m1=m1,m2=m2,n1=n1,n2=n2,n3=n3,a=a,b=b)],
        scale = is_def(r) ? r/max(rads) : 1,
        path = [for (i = [steps:-1:0]) let(a=angs[i]) scale*rads[i]*[cos(a), sin(a)]] 
             
            
         
    ) reorient(anchor,spin, two_d=true, path=path, p=path);

module supershape3d(shope1,shope2, overall_r,anchor=CENTER, spin=0, orient=UP){
    shape1=shope1;
    shape2=shope2;
     
    pts=
   [ for (c1=shape2) 
       for (c2=shape1)
          let
          (           
           x=c2[0]*c1[0]*overall_r,
           y=c2[1]*c1[0]*overall_r, 
           z=c1[1]*overall_r
           )
           [x,y,z]
   
   ]; 
   
   parallels=len(shape2);
   meridians=len(shape1);    
       
   fc1 =[
    for(z = [0:parallels-2])  //from top to bottom
        for(s = [0:meridians-1])
            let(
                // clockwise from left-down corner
                f1 = s + meridians*z,
                f2 = s + meridians*(z+1),
                f3 = ((s+1) % meridians) + meridians*(z+1),
                f4 = ((s+1) % meridians) + meridians*z
            )
            [f1,f2,f3,f4]
    ];
  fc2 = [[ for(s = [0:meridians-1]) (meridians*(parallels-1))+s]]; //bottom
  fc3 = [[ for(s = [0:meridians-1]) s]];  //top  
       
  fces=concat(fc1,fc2,fc3);
  
  vnf = [pts, fces];
    attachable(anchor,spin,orient, vnf=vnf) {
        vnf_polyhedron(vnf);
        children();
    };
     
};   
```
Example produced:

[image]

revarbat, 2020-08-19:
It strikes me that this is _NOT_ a supershape-specific thing.  It's more of a 3D modulation of a 2D path by another perpendicular 2D half-path.  You can just pass it supershape paths.  It should totally be able to modulate a hexagon with half a pentagon to make a 3D shape.

adrianVmariano, 2020-12-16:
I seems like it's an effort to implement the "spherical product" described on the linked web page.  It seems like it's hard to tell how to make this produce valid output.  I played around with various options and got a lot of invalid polyhedra.  The general utility of the spherical product isn't obvious to me.  It's sort of like a linear extrusion that scales the extrusion variably by the second path----only in spherical coordinates instead of linear.   I'm not sure that there's a call for a spherical product function, but if we have one it needs better defined parameters and clarity on how to ensure that the output is valid.   Is there a 2d function that takes r(theta) and produces the curve defined by that r in polar coords?    This is the 3d generalization of that.

I suggest closing this issue as being overly exotic/specialized.

closed 2020-12-17 by adrianVmariano