BelfrySCAD/BOSL2 #182: [BUG] type_of wrong result

RonaldoCMP, 2020-06-18:
I have found this:

echo(type_of(0/0)); // ECHO: "range"

"nan" is not among the possible outputs of type_of() but it should.

revarbat, 2020-06-18:
Should it return “nan” or is this technically a float?  What about inf?

RonaldoCMP, 2020-06-18:
As "nan" is not a number (even for is_num()), it should be in its own category: "nan".
inf is a number and accepted in arithmetic operations although:
   (1/0-1/0)
returns nan.

My proposal for typeof() would be:

function typeof(x) =
	is_undef(x)? "undef" :
	is_bool(x)? "boolean" :
	is_num(x)? "number" :
	is_string(x)? "string" :
	is_list(x)? "list" :
        x!=x? "nan" :
	"range";

RonaldoCMP, 2020-06-18:
BTW, in boolean expressions both "nan" and "inf" have value true.

RonaldoCMP, 2020-06-19:
There is another place where "nan" is classified as "range" with bad consequences.

`echo(compare_vals(0/0,[0:1])); // ECHO: 0`

compare_vals() in that case relies on OpenSCAD comparisons but "nan" is not comparable:

( (0/0)>a ) = ( (0/0)<a ) = ( (0/0)==a ) = false

for all "a".

To correct that its enough to change _type_num(x) to:

```
function _type_num(x) =
	is_undef(x)?  0 :
	is_bool(x)?   1 :
	is_num(x)?    2 :
	is_string(x)? 3 :
	is_list(x)?   4 : 
        x!=x?         6 :
  5;
```
and change the description of compare_vals() to 

// Description:
//   Compares two values.  Lists are compared recursively.
//   If types are not the same, then undef < bool < num < str < list < range < nan.

After that:

`echo(compare_vals(0/0,[0:1])); // ECHO: 1`

closed 2020-06-22 by RonaldoCMP

RonaldoCMP, 2020-07-18:
Due to OpenSCAD idiosyncrasies, `typeof()` still has a bug:

```
echo(is_undef([0:NAN:INF])); // false
echo(is_undef([0:"a":INF])); // true
echo(is_undef([0:[]:INF])); // true
echo(is_undef([0:true:INF])); // true
echo();
echo(is_type([0:NAN:INF], "range")); // true
echo(is_type([0:"a":INF], "range"));   // false
echo(typeof([0:NAN:INF]));  // "range"
echo(typeof([0:"a":INF]));  // "undef"
```

The only way I know to catch a valid range is with:

`function is_range(x) = ! is_list(x) && is_finite(x[0]+x[1]+x[2]) ;`

Function `typeof` should be able to return `"none"` or `"invalid"` for invalid ranges.