BelfrySCAD/BOSL2 #187: undef, nan, inf and data checks

RonaldoCMP, 2020-06-21:
I have been reading BOSL2 code for a while and found very few places where undef leaks out of control. Here is one of them:
`product([]) = undef`
However, I have found many places where nan and inf are not adequately dealt with. A nan or inf value may propagate and cause damages as severe as undef; they may become undef in something as simple as:
```
vect[1/0]
vect[0/0]
```
So, if we care about undef we should care about nan and inf. If something is ilegal or produces a nan or inf, an assert might be the appropriate response.

A simple example is found in vdiv:
`vdiv([1,-1,0],[0,0,0]) = [-inf,-inf,nan]`
In places like that it is cumbersome and inefficient to check each coordinate of the vectors for a dangerous value. We need something short and fast. In vdiv, we expect simple vectors fulfilled with finite numbers but is_vector resort to is_list_of() that may be expensive for long vectors. I recently found a better and faster way to check finite numbers, vectors and matrices:
```
function is_finite(v)       = is_num(0*v);
function is_vector(v,n)     = is_list(v) && is_num(0*(v*v)) && (is_undef(n)||len(v)==n) ;
function is_matrix(M,n,m,square=false) =
  is_vector(v[0],n) && is_vector(v*(0*v[0]),m)
  && (! square || len(M)==len(M[0])) ;
```
Those functions excludes nan and inf values and are faster because they rely just on elementary OpenSCAD array operators. Most operations in BOSL2 expect finite numbers and/or arrays with finite numbers and those checks should be used on the input parameters. Virtually any place where is_num is used is better served by is_finite. The function is_list_of() should be reserved to the rare occasions more general arrays are expected like in lookup tables. 

I offer myself to the job of reviewing the functions in std.scad to clean them from potential leaking of undef, nan and inf in order to have safer codes. I just need an orientation on how to do it because I have no previous experience with the github environment.

Another point I would like to consider is the data checking strategy. It has a direct impact on how to deal with dangerous values. In the current state of BOSL2, the data check is provided by each individual function that calls assert() when needed. I think it is better to have a unique point in the code where errors are reported. That allows for changes in the way the library reports errors and/or issue warnings without having to change all places throughout the code that issues an error. Warnings are important too so the user might be informed, for instance, that a radius was taken although he has specified a diameter too. 

In my libraries, I have a check point like that: a file included in all API files where there are 2 main functions to issue error and warning messages. The individual functions and modules call those functions to report any case of error and warning. Warnings might be silenced or limited by a global variable that the user can set. Recently I have been rewarded by having used such scheme: OpenSCAD abolished the HTML in echos and I had to simplify the fancy way I reported errors and warnings but just in one file.

revarbat, 2020-07-05:
I am not against you and others looking over the code for things like this.  The standard way to contribute is thus:

1. Go to my BOSL2 GitHub repo page `https://github.com/revarbat/BOSL2/`.
2. Click on the Fork button.
3. Use `git clone` to clone your fork of the repo to your local machine.
4. In your local repo, do the command `git remote add upstream https://github.com/revarbat/BOSL2.git`
5. Make changes in the source code that you want to make.
6. Do `git status` to see which files you changed.
7. Optionally use `git diff FILENAME` to see what changes you have made.
8. When the code is ready, use `git add FILENAME` for each changed file.
9. Commit the added changed files with `git commit -m "COMMIT DESCRIPTION"`.
10. Merge in any upstream changes I've made.  `git pull upstream`
11. Push the changes back to your GitHub repo with `git push`.
12. Go to your GitHub BOSL2 repo page.
13. Click on the `Pull Request` button, enter the description, then create the PR.
14. If a change you made fails to pass the regressions or docs validations, this will be noted at the bottom of your Pull Request page, and you will get an email about it.

revarbat, 2020-07-07:
BTW, @RonaldoCMP, you should join the BOSL2 Gitter chatroom at https://gitter.im/revarbat/BOSL2?utm_source=share-link&utm_medium=link&utm_campaign=share-link

revarbat, 2020-07-07:
I'm adopting your improvements to `is_vector()`, and `is_matrix()`.  Also adding `is_finite()`.
Implemented in 6bde1c231dd133e2f032d8a4f5eb28b400c682e9

revarbat, 2020-07-07:
Also added a CONTRIBUTING.md file to the repo.

adrianVmariano, 2020-07-31:
I think everything in this has been addressed except for two things:

1. is_num should change to is_finite everywhere in the code
2. Nothing has been said about the proposal to centralize error reporting.  Advantages?  Disadvantages?

RonaldoCMP, 2020-07-31:
Yes, `is_num` has been changed to `is_finite` in my PRs. The question of centralizing error reporting is quite complex although I have defended it already in #102 

In the current non centralized error reporting, we just concatenate a bunch of asserts with expressions. A centralized error reporting could not be done this way because the language syntactically doesn't allow the concatenation of expressions . So, instead of:

```
function unit(v) =
    assert(is_vector(v), "Input is not a vector." )
    assert(norm(v)>EPSILON, "Input is a null vector." )
    v/norm(v);
```
we would had something like:

```
function unit(v) =
    on_error(is_vector(v), "Input is not a vector." ) ? any_value :
    on_error(norm(v)>EPSILON, "Input a null vector." ) ) ? any_value :
    v/norm(v)
```
which would be more verbose and convoluted. That is one disadvantage. Another one is to change virtually all functions changing asserts to calls to a centralized error reporting.

The library already have a central point to report errors and warns: `error.scad`. Their functions and modules are not used anywhere though. And they should not because they resort to HTML styled echos still available in the stable version but banned from the following experimental versions. If that central was in large use, it would be simple to update it to the new echo behavior. That is one advantage of the centralization. Another is the possibility under the control of the user to silent any warnings and/or report just the first one although I didn't find any warning in the library yet.

After starting the job of finding leaking `undefs` and adding argument checks, I realized that would be painful to change everything (again) for error central reporting. So I am unsure about the issue.

It is interesting to note that `assert()` and `echo()` (in expressions) have a special syntactic nature not found elsewhere in the language. They are not expressions but may be the right hand side of an assignment. And they can be followed by an expression without any operator. We could call it a "function_module" returning an expression defined with something like:

```
function assert(cond, str) =
      if(!cond)  error(str) halt()
      else if($children()) children() else undef ;
```
where children() is any expression following the function call. If we had something like "function_module" in the language an centralized error reporting could be done similar to the `assert`.  Hans had already mentioned in the forum the lack of something like that.

adrianVmariano, 2020-08-01:
It's true that the problem of not being able to write our own "function modules" makes centralizing error reporting awkward.  It could be done using dummy variables, but it's a mess:

     dummy1=report_error(....),
     dummy2=report_error(...),

But I think one thing that's not clear to me is what benefit derives from error centralization, other than possibly formatting uniformity and control.  

So should we close this issue?  Or should we just change all references to is_num into is_finite and *then* close the issue?

RonaldoCMP, 2020-08-01:
Regarding `is_finite`, it should be noted that it is better to use it in `asserts` to avoid `INF` values but in other places the call of `is_num` is lighter and may be preferable provided that a previous `assert` has excluded non finite values. 
The disadvantage of error centralization due to the effort of reviewing all code to change asserts can be dramatically reduced by using a regular syntax substitution depending on how the error reporting will be defined. But any way the change would have a somewhat messy look.
I don't think it is time now to do a change in error reporting. The proposal is not ripe enough. So I guess `errors.scad` should be changed avoiding HTML syntax in `echos` and excluded from `std.scad`. It is not used anyway. After that, this issue could be closed.

adrianVmariano, 2020-08-01:
I submitted  PR that deletes errors.scad completely, with appropriate changes to std.scad, polyhedra.scad and skin.scad.  So closing the issue.

closed 2020-08-01 by adrianVmariano