BelfrySCAD/BOSL2 #188: [BUG]Different criteria to deal with unsolvable tasks

RonaldoCMP, 2020-06-22:
A few weeks ago, I have reported a bug in BOSL2 polygon triangulation which produced an illegal operation warning when the polygon had self-intersections. I have advocated at that time the function should return an undef value as an indication of the issue. The calling code could then take the appropriate steps to deal with the issue. Thus the triangulation function would have the beneficial side effect of polygon self-intersection detection. 

I was told that undef is an unsafe response because it can propagate and its is hard to debug when it happens. I can't agree more when undef comes out without a precise meaning reported in the documentation. The triangulation was modified by issuing an error when the triangulation process fails.

I have found other places in BOSL2 code where undef is intentionally output when there is no solution to a function task and the function documentations explicitly advert that. After all there is no point in stopping a run with an error only because two lines have no intersection or a linear system has no solution. It is the user responsibility to consider those possibilities and take the appropriate way. 

In a library like BOSL2 it is important to have a consistent criteria to deal with exceptions. In the above cases , I still think that undef is the best answer to unsolvable problems. When appropriately documented, undef is a meaningful value as it happens in expressions like vector[-1]. An odd alternative to that might be to add one more argument, 'safe', to those critical functions so that an assert is called if safe=true and undef is returned otherwise. A less elegant solution would be a global variable $safe that the user can freely set to true if he/she wants to be in a safer mode without undefs. What the default value safe or $safe would have is still a concern.

adrianVmariano, 2020-06-23:
I think there are situations where returning undef is reasonable.  Note, however, that I think vector[-1] would better simply be an error---it's a programmer mistake that this happens.

revarbat, 2020-07-07:
I'm on the fence about returning `undef`.  To be honest, I had a lot of functions returning it early in BOSL2's evolution, but @adrianVmariano has been talking me out of that on various new things, so we're inconsistent now.

revarbat, 2020-07-07:
The `safe=` argument idea isn't bad, but the name bugs me.  I think something like `assert=false` or `err=false`, maybe.

adrianVmariano, 2020-07-07:
I think that the reasonable case for a function to return undef is that the function can fail and there's no easy way to detect that independently of running the function.  So for example, we have that the intersection of parallel lines returns undef.   But I do think that this should be minimized because undefs propagate through the code and then you can't tell where/why it failed.  

Having some kind of systematic option seems like a good idea for cases like this.  Hmmm.  How about `undef=true`?

RonaldoCMP, 2020-07-07:
You can't use `undef` as an argument name.

 As it is radical to issue an error in cases like that (line
intersections), it might be an option. But what to do as the default
response? I can't think of  any value that could not become an undef if not
properly dealt by the calling code. It is the same situation of accessing
an element of an array without checking (or be sure) it is a non void
array. In some cases, an empty list may be a reasonable output to no answer
but it would not avoid an undef in a careless code.

adrianVmariano, 2020-07-07:
Yeah, I see undef doesn't work as an argument, which I suppose makes sense.  Though so often openscad lets you do things that don't make sense, so I guess I've gotten accustomed to it.  

I don't think I understand your point.  I was thinking that invalid input should immediately generate an assertion.  But for valid input maybe returning undef is OK.  If you ask for the intersection of two lines and the intersection does not exist, then undef makes sense as a result.  It's not clear that returning an empty list is better because the careless code will index into the empty list, and the result will then become undef.  On the other hand, the advantage of the empty list is that since it's not undef, we know it resulted intentionally and not "by accident".   That is, we can tell that it is a calculated return from the function as opposed to a result of mishandled input.  Maybe switching returns to empty list does make sense for this reason.  At the moment, the line intersection functions don't check validity of input.  

Note that for circle_circle_tangents when the tangents don't exist you get the empty list back, not undef.

RonaldoCMP, 2020-07-07:
I agree that invalid inputs should produce an error. That is tacit.

For valid inputs, I can see the following scenarios taking a linear system
solver (LSS) as an example althogh not all of them will be meaningful for a
given function.
1) usual case (one solution is found );
2) unusual cases (more than one solution is found);
3) no solution;
4) ill conditioned input data (one, many or no solutions?)

The issue is what to do in each of the 3 last circumstances. For the LSS,
the usual output will be a list with the solution. An empty list might be
used to indicate scenario 3. The unusual cases would require a more complex
output for a LSS. The last scenario for a LSS might be merged with the
second and produce a similar output.

In the (rare) cases the function doesn't output a list for a found
solution, an empty list for no solution seems to be unnatural, artificial,
and undef should be output. Anyway, the function outcome for all those
exceptional scenarios should be fully documented explaining its semantics
and so clarifying the unusual outcome is not an accident.

adrianVmariano, 2020-07-07:
I've never used a language that had "undef" before, so maybe I have a different perspective.  But take the case of tangent lines to a pair of circles.  There are either 4, 2 or zero solutions.  It seems very natural to produce a list of 4, 2 or zero (empty list).  Producing undef instead of the empty list seems kind of unnatural to me.  The empty list elegantly captures the result: no solutions.  

The case of the linear solver is a little bit more complicated.   The one I implemented fails to get a solution when the input is rank deficient.  It doesn't mean there isn't a (unique) solution.  (In other words, a rank deficient problem that appears overdetermined might actually have a unique solution.)   I think this is a better case for returning undef.  

I guess after further thought on this I think that we should return empty list when a computation has "no solutions" as its result rather than returning undef.  If empty list is awkward or doesn't make sense then as a last resort, we return undef.  The linear solver case is an example where returning undef is a better option because there aren't zero solutions---we just haven't found the solution.  (Or maybe somebody should go fix my code to handle the rank deficient cases more intelligently.)    But the case of line intersection could be changed to return empty list instead of undef.

revarbat, 2020-07-09:
So the question becomes, which functions are candidates for re-assessing?

RonaldoCMP, 2020-07-09:
> I guess after further thought on this I think that we should return empty list when a computation has "no solutions" as its result rather than returning undef. If empty list is awkward or doesn't [...]

For the caller code, it is irrelevant why a function didn't  provide a solution. To be safe it needs to deal with the answer "no solution found". So I see no need to reply an undef for a rank deficient system and an empty list when the system has no solution.  In case the code of the linear solver is reviewed and rank deficiency is fully considered it still will have two possible answers: a solution and "no solution found".  An empty list would be an appropriate return for all inputs for which a solution can't be found avoiding the caller to check both undef or empty list as exceptions.

To simplify I would consider an empty list as an answer to "no solution found" for any function that normally would output a list and would use undef for those functions returning non list objects. That would be clearer as a standard for us and the users. For the eventual case of functions that may return either a list or something else ( I can't imagine none) the convention for "no solution found" might be undef.

In any case, that should be clearer in the function docs what an undef or empty list means.

RonaldoCMP, 2020-07-09:
> So the question becomes, which functions are candidates for re-assessing?

I would say all functions where an assert is issued although the input is valid and all functions that may has "no solution found" as a possible return. For instance, I think polygon triangulation should return an empty list for self-intersection polygons as it is hard to check that before calling the function. Returning an empty list will be a beneficial side effect as it could be used to check the existence of self-intersections. Errors should be issued only for invalid inputs.

There is another mater related to that. Many functions in the library does not check its input and so all of them should  be reviewed. However I think it will be an unreasonable overhead to check the inputs to basic function that are used extensively all over the library: select() for example. If the library code is written carefully, what is the point to check the inputs of each internal call to select ? It should be checked just when called by an user code.  Or when a new library function is being developed and/or in beta test. That means we would need two functions for each basic task: an API version with data check and an internal use version without it. All functions under std.scad are in principle candidates to that approach.

adrianVmariano, 2020-07-09:
The linear solver I wrote solves the least squares problem, so there is in principle always a solution---except that it fails to solve cases where the input is rank deficient.   Now it's not clear actually that in the rank deficient case you want to return the least squares solution, especially for square inputs.  I don't know.   But there is really no case of zero solutions, just one or many.  It's not clear that returning an empty list for "many solutions" (singular input) makes sense.   I mean, if we make it the convention it's OK.  I do think avoiding undef as much as possible is desirable so that the presence of undef means there's a bug in the code.

How many functions are there that return undef?

I am not sure about the notion that functions should not issue errors.  I guess it's a question of what is "valid" input.  I have already pushed for the unit() function to return an error when the input is the zero vector.  It should not return zero like it currently does, nor should it return undef or empty list.  If the caller wants to handle the zero case, it is easy to check before calling this function.  But it's a nuisance to make every use of unit() check its return value.   And for the internal application that currently needs unit() to return zero for zero inputs, write an internal function.  I think that when it's trivial to check the input then functions should issue errors when the input doesn't allow the problem to be solved.  It's only in cases where the error condition is nontrivial that returning an empty list (or undef) is appropriate.

RonaldoCMP, 2020-07-09:
There is a subtle difference between "no solution" and "no solution found": the later applies to your linear solver. So I agree that it returns an empty list for rank deficient systems. I think there are only three non boolean values that are converted to value false in boolean expressions: 0, undef and []. That makes any of them natural candidates to express an exception easy to check.

For the function unit(), it is weird to return 0 though. If the division was done without any check the returned value would be a list of nan's that is harder to detect than [] or undef.  In math, this function is undefined for zero vectors and the value undef of the language is provided with that meaning. But, if we adhere to the rule I have suggested, [] should be returned and that would be my choice. Are there any function that may return an empty list and have an exception to be discriminated from it? That would spoil my rule.

I have examined a few files included by std.scad for now and found 2 or 3 places where undef is a possible outcome but possibly due to missing input checks. There are many possibly returning nan or inf.

adrianVmariano, 2020-07-09:
Returning 0 is clearly a bad idea because it looks like a valid answer (a number).  I think either [] or undef are the only reasonable options to return.  And clearly [] makes sense for "no solutions".  Revar, do you have any thoughts?   Return [] systematically when there are no solutions?  Is there any situation where this would be problematic?  

The function unit() is a case where I think that returning an error result or any sort is annoying.  This function gets used all over the place, and it's really annoying to check its return value---I want to just compute with it.  So the possibility that it returns an unusable value is very annoying.  I think that it should issue an assert if the input is zero.  If you want to trap this case in your code it is easy to check before calling unit().  In general this is how I think it should work:  when it is trivial to check validity of the input before calling, the function should fail with an assert on such invalid input such as a zero vector for unit().  Only when this check is nontrivial should we return an error result such as [] or undef.  In the matrix example, checking that a matrix is singular is clearly nontrivial, and I think also that checking the state of triangulation is nontrivial.  You basically have to perform the computation to validate the input.  In this case it makes sense to return some kind of error result rather than using an assert.   But it seems reasonable to have a flag that causes an assert rather than an error return, like Revar suggested above, `error=true` or something. 

It is indeed weird to return zero from unit().  Revar does this for specialized reasons having to do with his attachment code.  I think he should write a specialized internal function for this case.  

There are still many functions that need input checks.  I believe I opened an issue about that a while back, but the work is definitely not done on that front.

revarbat, 2020-07-10:
I just changed `unit()` so it asserts an error for zero-length vectors, unless a `dflt=` value is given.

revarbat, 2020-07-10:
I first implemented it as `anchor_unit()`. I quickly realized on updating code to match, though, that what I needed was not an anchor specific variation, but a more generalized `unit()` with definable return on zero-length input.

RonaldoCMP, 2020-07-10:
Why not add 'dflt=value' to all functions with exception cases? That gives
the caller code the choice of the return it wants for exceptions. The
default value of 'dflt' might still be something reasonable ([] for no
solution found, for instance) or undef (for an assert).

adrianVmariano, 2020-07-10:
dlft isn't a very good name for that behavior.  Should it be error=?  Or zero=?

RonaldoCMP, 2020-07-10:
>
> dlft isn't a very good name for that behavior. Should it be error=? Or
> zero=?

Maybe not but 'dflt' has been used in BOSL2 to indicate what to return when
there is not anything better. That is how I see the parameter 'dflt' in
'unit()', 'get_radius()' and 'get_height' for instance.

Anyway, what matters is the semantics and the standard way we will
deal with the exception cases.

adrianVmariano, 2020-07-10:
Yes, I know you've been using dflt for cases like get_radius.  For a case like that, a default makes sense.  You return one of the user supplied values...or the default if no user input was given.  The default is..well, the default.   For the general category of how to handle error conditions, "default" doesn't make sense.  It's not a default.  It's an error state.  What's the default value of a unit vector?  It seems like that would make sense if you gave no input (which is impossible) not invalid input.   So basically what I'm saying is that you're trying to generalize the "dflt" mechanism into a domain where it's not appropriate, which is why I'm suggesting you use a different term.   Also one can imagine a function which has a default and *also* has an error return case, so if we're designing a generic mechanism we now have a problem.   Ideally something like 'error_return=...' would make very good sense...but it's kind of verbose.

revarbat, 2020-07-10:
I considered `zero=` but that one bugs my sense of aesthetics for some reason.  Do we use `err=` as an argument anywhere already?

adrianVmariano, 2020-07-10:
You think `zero=17` is funny looking?
 
It looks like "err" is used in 3 places, and never as an arg.

revarbat, 2020-07-11:
So, proposition stands thus:not declaring an `err=` argument should assert on error.  Declaring `err=...` returns the value of `err` on error.

revarbat, 2020-07-11:
So, proposition stands thus:
- Declaring `err=...` returns the value of `err` on error.
- Not declaring an `err=` argument should assert on error.
- Default value for `err=` that invokes assertion on error, should NOT be `undef`, because that's a common value we might want to return on error.

revarbat, 2020-07-11:
I wouldn't be against `error=`, if not using a full word argument bugs folks.

adrianVmariano, 2020-07-11:
It's not clear what you pick to indicate that errors should assert.    Also, what parameter should I use on my poly_roots function to indicate that it should return error bounds in addition to the roots?  I was thinking I should put that feature in, because I think it will make real roots more robust if I use those error bounds.

revarbat, 2020-07-11:
`bounds=true`?

revarbat, 2020-07-11:
I suppose that `error="AsSeRt"` or `error=9876543210` could be used with minimal chance of running into those values.  The number would be more efficient than the string.

adrianVmariano, 2020-07-11:
I would suggest `error=[[["assert"]]] or something like that.

RonaldoCMP, 2020-07-11:
I am still confused. What functions would be eligible to receive an err
argument? All functions for which would be hard to validate parameters
before the call? All functions that have exception cases for which no
solution can be found? All functions that are returning undef now?

I certainly agree with an argument that indicates what should be returned
instead of halting the run with an error. But it is not clear what
functions should include them. That mechanism would not be included for all
bad input data, for instance. It might be included when it is hard to
validate the input before the function call. And when there are exceptional
cases? Defining a semantic for an err argument is not enough to answer
those questions.

adrianVmariano, 2020-07-11:
How useful is it really to be able to control the return value in the error condition?  I know that for unit() with this special requirement there's a use case.  But it seems like the main reason to want an error return is so you can trap it and then do something different.

adrianVmariano, 2020-07-13:
The reason for my last question is that if you don't really need a return val you could just have `error=true` or `error=false`.  

Another idea.  You have `error=true` for an assert and `error=[value]` for a return value.  That lets you return any value.

adrianVmariano, 2021-02-06:
With the new handling of undef is this a non-issue now?  We should just return undef for unsolvable tasks?

revarbat, 2021-05-27:
Yeah, I'm of a mind to just return undef for unsolvable.  Except in the case where multiple results might be returned, in which case, returning a null list sounds appropriate.

adrianVmariano, 2021-05-28:
Are there any cases where we currently return something else and should change it (back?) to undef?

revarbat, 2021-05-28:
I vaguely recall a couple, but hell if I recall where.

adrianVmariano, 2021-05-28:
I guess we can assert that the standard is to return undef for undefined results and if we find places where it's wrong...fix it.

RonaldoCMP, 2021-05-28:
I think some functions in geometry.scad return [] instead of undef for
unsolvable tasks.

adrianVmariano, 2021-05-31:
It seems reasonable that a function that returns various results can return zero results with the empty list.  I don't think it's necessarily desirable to return undef for zero results and a list for >0 results.

Can we close this issue?

revarbat, 2021-05-31:
Yeet.

closed 2021-05-31 by revarbat